realsenseai/librealsense #13676: Android has been brushing this exception, has it affected, the preview screen is normal

1programmer2dfg3, 2025-01-12:
W/librs: control_transfer returned error, index: 768, error: Broken pipe, number: 32

MartyG-RealSense, 2025-01-12:
Hi @1programmer2dfg3  An Android user who experienced the error that you described posted their Java solution for it at https://github.com/IntelRealSense/librealsense/issues/10242#issuecomment-1079947049

Another Android case that had this error can be found at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4407594297491/comments/4407653205011

1programmer2dfg3, 2025-01-12:
Thank you, I looked at the url link you provided, and the comment did not give a specific solution

MartyG-RealSense, 2025-01-12:
Apart from the two links above, the only other Android reference for that highly rare error message that was available, unfortunately, is https://github.com/IntelRealSense/librealsense/issues/7691

**control_transfer returned** messages, if they generate multiple times, can indicate that there is a communication problem between the camera and the computer, most often related to a problem with the USB cable or a problem on the computer's USB port (such as not being able to cope with the volume of data arriving from the camera at a high resolution).

closed 2025-01-26 by MartyG-RealSense