realsenseai/librealsense #13678: [Question] Inquiry about Realsense Depth Camera Series

soribido, 2025-01-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   | 
| Firmware Version                   |  | 
| Operating System & Version |   Ubuntu22.04 LTS or Windows11  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  opencv/python }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

I have some questions regarding the Realsense Depth Camera Series. I am planning to install a depth camera on the ceiling to detect depth information of objects and people. My intention is to treat the camera and PC as separate entities, receiving real-time video streams from the camera via remote communication.

I am specifically interested in obtaining only the depth image (map) and RGB image. This process will be carried out using Python and OpenCV, avoiding the use of other languages, GUIs, or additional software. More specifically, I aim to use the depth information corresponding to the same location as the RGB image input for a detection model. The distance between the ceiling and the ground is expected to be approximately 8 to 10 meters.

Given this context, I have a few questions. While I am not an expert in cameras, I understand that the answers might differ depending on whether the camera uses stereo or ToF technology. I appreciate your understanding in this regard.

1. Is it possible to install the camera on the ceiling and receive video streams on a remote PC?
2. Does the camera support the RTSP protocol?
3. What is the expected margin of error when detecting a 1.8-meter-tall person from a height of 10 meters?
4. Can the camera provide both RGB images and depth maps simultaneously?
5. Does the depth map represent relative distances, or can it be converted into absolute distances through specific processing? If so, could you explain how this is achieved?
6. In a scenario where the camera is mounted on a 10-meter ceiling, how is depth calculated for objects at different angles? For example, for a 2-meter-tall object directly below the camera and another 2-meter-tall object at a 45-degree angle from the camera, will the depth values differ? If so, will the difference be approximately sqrt(2), or will it be calculated differently?
7. I’ve observed in some cases that depth information might not be captured (e.g., appearing black) in dark areas or other specific conditions. What causes this issue, and are there any solutions?
8. According to [this issue](https://github.com/IntelRealSense/librealsense/issues/13308), it seems that RGB and depth images might not align properly. Does this issue still persist?

Lastly, I’d like to know if there are any ToF-based Realsense cameras available. If so, could you recommend a suitable camera for this setup?

soribido, 2025-01-15:
@MartyG-RealSense 
Thank you for your kind and detailed response.

I have a follow-up question based on your response to my first inquiry. As I understood, it seems that devices like a PoE Hub or NVR cannot be used to connect the RealSense camera to the network directly, and instead, a Raspberry Pi or Jetson Nano is required. Is that correct? Could you please clarify why these devices are necessary, and if a PoE Hub or NVR is not suitable in this case?

MartyG-RealSense, 2025-01-15:
RealSense 400 Series cameras do not have a power-over-ethernet connector (PoE) and instead draw their power and communicate via a USB cable.  

Although the majority of users of the **rs-server* RealSense ethernet tool at the link below use a Raspberry Pi with it, as far as I am aware it is possible to use another board or a PC computer with it.  The important thing is that the computer / computing board that the camera is plugged into runs the Linux OS, as the rs-server networking program can only be compiled on Linux.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

An alternative method of networking RealSense cameras is to use the Python-based **EtherSense** project, whose documentation shows the use of PoE with an **Up Board** instead of a Pi.

https://github.com/krejov100/EtherSense

https://dev.intelrealsense.com/docs/depth-camera-over-ethernet-whitepaper

soribido, 2025-01-16:
@MartyG-RealSense 
Thank you for your response.
I will conclude my questions here and open a new issue if I have further inquiries in the future.

closed 2025-01-16 by soribido

MartyG-RealSense, 2025-01-16:
@soribido  You are very welcome.  Please do feel free to open a new issue later if you have additional questions.

MartyG-RealSense, 2025-05-23:
Hi again.  As you referenced PoE in this January 2025 discussion, I thought I would let you know about the new-generation RealSense **D555** camera that is coming soon that is equipped with a PoE interface.

https://www.intelrealsense.com/depth-camera-d555/