realsenseai/librealsense #13685: RealSense error calling rs2_pipeline_start( ):xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error.

kummerrrrrr, 2025-01-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435i } | 
| Firmware Version                   | (2.55.1) | 
| Operating System & Version |   {VM (Ubuntu 20.04) | 
| Kernel Version (Linux Only)    |  (Linux vision 5.15.0-130-generic)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {C++/opencv/pcl/ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I compiled librealsense using source code, located the executable file, and executed ./rs-pointcloud and ./realsense-viewer respectively.
The former reported an error: RealSense error calling rs2_pipeline_start(pipe:0x5560af0439b0):xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error.
The latter can run normally. (Please see the attachment)
May I ask why this result occurs?
[image]
[image]

kummerrrrrr, 2025-01-16:
By the way, it seems that all errors point to one point (as shown in the figure below), pipe.start(cfg); The pipeline is opened incorrectly because two streams are enabled at the same time.
[image]
I don't know if you still remember that I came here a few weeks ago and discussed this issue. At that time, I ruled out some possible problems, such as I have changed the USB to 3.2, and the firmware and SDK are kept to the latest version. Other than that, I really don't know what the problem is.
Thank you!

MartyG-RealSense, 2025-01-16:
Hi @kummerrrrrr  Can you confirm please that realsense-viewer is not streaming depth at the time that you launch rs-pointcloud?  Only one program at a time can access a particular camera stream, so if depth was enabled in realsense-viewer first and then rs-pointcloud was launched, rs-pointcloud would have a blank black screen because realsense-viewer has already claimed exclusive use of the depth stream until it is stopped or realsense-viewer is closed.

kummerrrrrr, 2025-01-16:
Yes, I always close the current program before launching another one.

kummerrrrrr, 2025-01-16:
It's the same

[image]

MartyG-RealSense, 2025-01-16:
If realsense-viewer is able to work correctly then it indicates that the problem is not related to the ability to run RealSense programs that use graphics.

A difference between your realsense-viewer program and rs-pointcloud is that in realsense-viewer you do not have RGB color enabled, whilst rs-pointcloud automatically uses both depth and the RGB color stream.

If you enable first depth and then RGB in the 3D pointcloud mode of realsense-viewer, is a pointcloud with both depth and color detail on it able to be displayed?

kummerrrrrr, 2025-01-16:
Yes，I can get the pointcloud of color.

[image]

MartyG-RealSense, 2025-01-16:
Thank you.  Next, please run **rs-capture**.  If it works and can display both depth and color streams then that would indicate that the input-output error is a problem that is specific to the rs-pointcloud example program on your computer.

kummerrrrrr, 2025-01-16:
Still reports the error 
(ros-foxy) vision@vision:~/librealsense-master/build/Release$ ./rs-capture 
 16/01 20:43:57,762 ERROR [139951433836288] (librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 16/01 20:43:57,762 ERROR [139952346539840] (librealsense-exception.h:52) xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error
 16/01 20:43:57,763 ERROR [139951433836288] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Connection timed out
 16/01 20:43:57,867 ERROR [139952346539840] (rs.cpp:237) [rs2_pipeline_start( pipe:0x55b9699c8a90 ) Backend] xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error
 16/01 20:43:57,868 ERROR [139952346539840] (rs.cpp:237) [rs2_delete_pipeline_profile( profile:nullptr ) UNKNOWN] null pointer passed for argument "profile"
RealSense error calling rs2_pipeline_start(pipe:0x55b9699c8a90):
    xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error

MartyG-RealSense, 2025-01-16:
It is not clear why realsense-viewer would work but the example programs would not, as the the input/output error would suggest that the problem is related to USB (the USB port or the USB cable).  The USB cable is detected by realsense-viewer as 3.2, suggesting that it is in good working order.

Which method did you use to install librealsense from source code?

kummerrrrrr, 2025-01-18:
Hello, I think my problem is solved. I used VMware Workstation to create a Linux virtual machine under Windows 11. I accidentally saw the realsense developer's note that it is not recommended to do development under a virtual machine because there is a problem with the conversion of USB devices from Win->VM. I tried to install the Linux system for testing and found that the problem has been solved!
I hope this post will not be deleted so that more developers can be prompted.
Thanks.

MartyG-RealSense, 2025-01-18:
Thanks so much @kummerrrrrr for sharing the experience of your success with the RealSense community!

Posts may be closed but they are not deleted and remain accessible.

MartyG-RealSense, 2025-01-24:
Case closed due to no further comments received.  It will remain accessible in searches.

closed 2025-01-24 by MartyG-RealSense