realsenseai/librealsense #13686: failed to set power mode when running realsense d4xx on the jetson orin nano

stevedanomodolor, 2025-01-16:
I am connected to a realsense d4xx camera and i am trying to run the depth sensor and the color sensor at the same time. I noticed that after setting both camera and opening it, i get the following exception using the pyrealsense2 python library “failed to set power state”

I am using a power supply that is 5v and 2.3 A. It seems to happens only when i try to set both camera sensor, It happens always on the “device = device_list[selected_ind]” line of the following code of pyrealsense. Is this a power issue?.

i am running the latest jetpack 6.1 super on max mode and I also did the jetson_clock.

context = rs2.context()
device_list = context.query_devices()
if len(device_list) > 0:
selected_ind = 0
device = device_list[selected_ind]
usb_descriptor = device.get_info(rs2.camera_info.usb_type_descriptor)
if “3.2” in usb_descriptor:
return 2
else:
return 1
else:
return 0

MartyG-RealSense, 2025-01-17:
Hi @stevedanomodolor  For the Jetson Nano board, Intel strongly recommend enabling the **barrel jack** power connector for extra power by following instructions at the link below.  I assume that the same power recommendation is applicable to the Orin Nano model.

https://jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

stevedanomodolor, 2025-01-17:
> Hi @stevedanomodolor  For the Jetson Nano board, Intel strongly recommend enabling the **barrel jack** power connector for extra power by following instructions at the link below.  I assume that the same power recommendation is applicable to the Orin Nano model's barrel jack.
> 
> https://jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

I am using the Jetson orin nano see https://developer.nvidia.com/embedded/learn/get-started-jetson-orin-nano-devkit, would it be enough just to get a power supply with 5v 4A?. In this case the step to enable the barrel connector isn't necessary right?

MartyG-RealSense, 2025-01-17:
In theory it sounds reasonable, though I do not know if it would work in practice.

You could try setting your Orin Nano to **MAXN** power mode, where power supply is unrestricted, to see whether it enables both sensors to be enabled simultanously.

https://www.jetson-ai-lab.com/initial_setup_jon.html#switch-to-maxn-mode

stevedanomodolor, 2025-01-17:
> In theory it sounds reasonable, though I do not know if it would work in practice.
>
> You could try setting your Orin Nano to **MAXN** power mode, where power supply is unrestricted, to see whether it [...]

I am actually on MAXN. Adding more logs for more context. This is the dmesg from when I connect the camera and when it crashes.

```
[  347.246934] usb 2-1.4: new SuperSpeed USB device number 4 using tegra-xusb
[  347.279199] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  347.279663] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  347.295980] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input12
[  347.298191] uvcvideo 2-1.4:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[  347.298209] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  347.320157] hid-generic 0003:8086:0B3A.0004: hidraw1: USB HID v1.10 Device [Intel(R) RealSense(TM) Depth Camera 435i Intel(R) RealSense(TM) Depth Camera 435i] on usb-3610000.usb-1.4/input5
[  560.811555] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  560.812181] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  560.829406] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input13
[  560.974014] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  560.974430] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  560.990816] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input14
[  561.060153] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  561.060652] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  561.077496] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input15
[  563.813051] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  563.813545] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  563.830541] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input16
[  563.941774] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  563.942168] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  563.959399] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input17
[  564.031274] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  564.031676] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  564.049284] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input18
[  567.698778] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  567.699165] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  567.716766] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input19
[  567.843760] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  567.844163] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  567.860330] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input20
[  567.934283] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  567.934822] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  567.952437] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input21
[  569.032748] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  569.033295] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  569.049808] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input22
[  569.186128] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  569.186666] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  569.203154] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input23
[  569.276448] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  569.276954] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  569.293557] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input24
[  569.365256] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  569.365867] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  569.382433] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input25
[  569.945036] usb 2-1.4: Disable of device-initiated U1 failed.
[... cut ...]
I dont know if this is the ideal way of using the pyrealsense2 library but i call the function often to know the state of the camera. To see if the camera is connected and when I setup the the color and depth sensor is when I start getting this runtime exceptions. Is this function meant to be called that often?

```
def get_status()
    context = rs2.context()
    device_list = context.query_devices()
    if len(device_list) > 0:
    selected_ind = 0
    device = device_list[selected_ind]
    usb_descriptor = device.get_info(rs2.camera_info.usb_type_descriptor)
    if “3.2” in usb_descriptor:
    return 2
    else:
    return 1
    else:
    return 0
```

MartyG-RealSense, 2025-01-17:
Calling a function frequently increases the risk of something going wrong in the program if the camera does not respond in time.

A 'right' way to test the connection would be to use an instruction called **set_devices_changed_callback** to listen for camera connection and disconnection events.  Code for doing this in pyrealsense2 is discussed at https://github.com/IntelRealSense/librealsense/issues/4212#issuecomment-586262375

stevedanomodolor, 2025-01-17:
> Calling a function frequently increases the risk of something going wrong in the program if the camera does not respond in time.
> 
> A 'right' way to test the connection would be to use an instruction called **set_devices_changed_callback** to listen for camera connection and disconnection events. Code for doing this in pyrealsense2 is discussed at [#4212](https://github.com/IntelRealSense/librealsense/issues/4212) with a script provided at [#4212 (comment)](https://github.com/IntelRealSense/librealsense/issues/4212#issuecomment-586262375)

Since i dont call the context get query constantly and use the callback you mentioned, I have not gotten that error. Now I noticed that I get another error when I try to configure the visual preset on the depth sensor. I read the visual present range and set it to the maximum. 

```
    self._depthSensor.set_option(rs.option.visual_preset, self.preset)
RuntimeError: get_xu(ctrl=1) failed! Last Error: Success

```

This is the the dmesg error 
```

Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 4647.336239] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 4647.353129] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input145
[ 4647.353823] uvcvideo 2-1.4:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[ 4647.353840] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 4649.336633] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 4649.337115] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 4649.354105] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input146
[ 4649.420453] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 4649.420848] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 4649.437240] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input147
[ 4649.534237] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 4649.535770] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 4649.555103] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input148
[ 4649.626428] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 4649.626870] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 4649.644130] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input149
[ 4649.734184] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 4649.734606] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 4649.751870] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input150
[ 4649.776888] uvcvideo 2-1.4:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[ 4649.776918] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 4650.629026] usb 2-1.4: Process 17882 (pt_main_thread) called USBDEVFS_CLEAR_HALT for active endpoint 0x82
[ 4651.346324] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 4651.346743] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 4651.364444] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input151
[ 4651.487930] uvcvideo 2-1.4:1.1: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 4651.488361] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 4651.505295] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.4/2-1.4:1.0/input/input152
[ 4651.608392] uvcvideo 2-1.4:1.4: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[ 4651.608433] usb 2-1.4: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)

```

MartyG-RealSense, 2025-01-17:
https://github.com/IntelRealSense/librealsense/issues/10014 has a demonstration of configuring a visual preset on Python that you can compare to your own code.

self.profile = self.pipeline.start(config)

```
HIGH_ACCURACY = 3
HIGH_DENSITY = 4
MEDIUM_DENSITY = 5
self.profile.get_device().sensors[0].set_option(rs.option.visual_preset, HIGH_DENSITY)
```

stevedanomodolor, 2025-01-17:
It works but sometimes the setting fails. Not sure why is happens.

stevedanomodolor, 2025-01-17:
> Did you build librealsense from source code with the **-DFORCE_RSUSB_BACKEND=true** flag, please? If you did, then a RSUSB backend build of the SDK will sometimes not be ready to respond when a request is made to change a setting with SET_OPTION. [#6952](https://github.com/IntelRealSense/librealsense/issues/6952) is an example of this issue.
> 
> If librealsense is built with the 'native' method (RSUSB = false and a patch script applied to the kernel) then the SDK responds faster and more reliably to option setting requests.

Thank you again for the fast response. I really appreciate it. 
Yes, I built with the RSUSB option set to true. That might answer why this behavior is inconsistent. When using docker, how would the kernel patch work. Would I still need to do it on the host device ?

MartyG-RealSense, 2025-01-17:
If you have an Orin model of Jetson with JetPack 6 then the 'native' kernel-patching method of librealsense installation will not work according to the official instructions for a native build with kernel patching on Jetson.  Instead, the instructions recommend installing using the RealSense **MIPI driver** on an Orin with JP6, a link to which is in those instructions.

https://github.com/intelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

stevedanomodolor, 2025-01-17:
Thank you for the response so far. Really insightfull

closed 2025-01-17 by stevedanomodolor