realsenseai/librealsense #13692: Intel RealSense Viewer not detecting camera after fresh install on Linux based OS

bohumilk, 2025-01-22:
Hello, I have Intel D455 camera using together with Raspberry Pi 4. Now I switched to Raspberry Pi 5 with new installation of 64bit version of Bookworm Raspberry Pi OS. I followed the guides for Linux from: [https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide](url) and everything was fine, but after running the RealSense Viewer, the camera is not detected by it after connection on raspberry. I am not very skilled in Linux, so I am unable to resolve the problem. To me it seems the camera is detected properly by linux, but not by Intel software. I red a lot of other threads, but without any luck. Also, the camera is used on Windows machine where it works perfectly and it is detected on Raspberry Pi 4, where it is recording, so the problem is only with this fresh install...

This is result of: dmesg
[image]

Results of commands shown in header:
[image]

[image]

[image]

[image]

Result of rs-enumerate-devices:
[image]

Messages after start of Intel RealSense Viewer:
[image]

Thank you for any help!

MartyG-RealSense, 2025-01-22:
Hi @bohumilk  Does it make a difference if you launch programs in **sudo** admin powers mode, such as `sudo realsense-viewer`

bohumilk, 2025-01-22:
@MartyG-RealSense I also tried the sudo as I red it in recent issues with Jetson here, and there is no difference in my case.

MartyG-RealSense, 2025-01-25:
There was a previous case at https://github.com/IntelRealSense/librealsense/issues/12574 of the camera working on Pi 4 but not on Pi 5.  There was not a solution found in that particular case.

A Pi 5 and Bookworm user at https://github.com/IntelRealSense/librealsense/issues/13373 also had problems.

A guide for installing librealsense on a Pi 5 with Ubuntu instead of Bookworm was posted at the link below.

https://www.robotexchange.io/t/how-to-setup-the-intel-realsense-software-and-pyrealsense2-library-in-ubuntu-on-a-raspberryi-pi-5/3414

MartyG-RealSense, 2025-02-06:
Hi @bohumilk  Do you require further assistance with this case, please?  Thanks!

bohumilk, 2025-02-12:
Ubuntu seems to be working fine on Raspberry Pi 5 with Intel D455 camera so thanks a lot! I have also Raspberry Pi 4 where the software is running, but with significant framedrops. My original idea was to have one microSD card which I can switch between both, but with Ubuntu it seems to be too much on Pi 4. Any suggestion on how to make the system running with camera at 60 FPS with 848x480 on both, RGB and Depth? Thank you!

MartyG-RealSense, 2025-02-12:
Is Ubuntu on the microSD card as well or just the librealsense installation?

If the OS was separate from the microSD card then you could consider using Ubuntu MATE (a less demanding flavor of Ubuntu) on the Pi 4.

https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md

If the OS needs to be on the microSD card with librealsense then perhaps you could try MATE with the Pi 5 too to see what the camera's performance is like with MATE.

bohumilk, 2025-02-12:
Yes, everything is installed on the microSD card. It is 128 GB Sandisk Extreme A2 U3 V30. The idea was to have the possibility to swap the microSD card (with whole installation) between both Raspberry devices. After installation of Ubuntu on Pi 5 I just put the card into Pi 4 to test it. The result was significant framedrops.

Basically, I do not need to have live stream. I need to save the recording at this resolution and framerate for future evaluation but the framedrops I mentioned, they are in the recorded file.  

Thank you for the suggestion, I will try MATE with both to see if it will do the job!

closed 2025-02-28 by MartyG-RealSense