realsenseai/librealsense #13695: having trouble trying to build java_example (or any other example) from android wrapper

Taim974-H, 2025-01-22:
Epecifically when trying to build java example, I am getting this error:

> Task :prepareKotlinBuildScriptModel UP-TO-DATE
Download https://egiintel.jfrog.io/artifactory/librealsense/com/intel/realsense/librealsense/2.18.0/librealsense-2.18.0.pom.sha1, took 232 ms
Download https://egiintel.jfrog.io/artifactory/librealsense/com/intel/realsense/librealsense/2.18.0/librealsense-2.18.0.pom, took 232 ms
[Fatal Error] librealsense-2.18.0.pom:2:10: Already seen doctype.

and when trying to build aar myself i get errors, i notice librealsense:buildCMakeRelWithDebInfo[arm64-v8a] failing. 
> Task :librealsense:buildCMakeRelWithDebInfo[arm64-v8a] FAILED
FAILURE: Build failed with an exception.

how do i fix this and how do i run the app on my android phone (samsung S23 ultra)?

MartyG-RealSense, 2025-01-25:
Hi @Taim974-H  The RealSense Android wrapper no longer supports JFrog and has not done so for a while.  The wrapper has not been updated for over two years at the time of writing this, except for a couple of minor changes such as adding support for Android 12.  

The end of JFrog support - which affects the java_example and native_example projects - is discussed at https://github.com/IntelRealSense/librealsense/issues/12850

Regarding the AAR, are you building it with Android Studio?  If you are, which Android Studio version are you using please?  If it is the 2024 version then librealsense version **2.56.3** should be used as Android Studio 2024 support was first added in that SDK, as described at https://github.com/IntelRealSense/librealsense/pull/13432

Taim974-H, 2025-01-25:
Thank you for your reply. I noticed that JFrog support has ended, and the latest update is two years old. I am using Android Studio Ladybug 2024.2.1 with the latest SDK (2.56.3).

**I was able to resolve the issue** by downgrading to JDK 17 and adjusting the Gradle versions (AGP 8.1.4 and Gradle 8.9). I also encountered another error related to long path lengths (which I found a related issue for). To address this, I moved my folder to C:/librealsense, cleaned and rebuilt the project, and despite numerous warnings, the build was successful.

I then generated the AAR file, moved it to the libs directory in the java_example, updated the module build.gradle file (added implementation files('../libs/librealsense-release.aar') in dependencies and mavenCentral in repositories), cleaned the build again, invalidated caches, and rebuilt the project. It worked after that.

closed 2025-01-25 by Taim974-H

MartyG-RealSense, 2025-01-25:
It's great to hear that you were successful.  Thanks so much for sharing the details of the procedure that worked for you!