realsenseai/librealsense #13699: realsenser-viewer do not work when my pc screen resolution is 240Hz

WhyxyxIsNotAnAvailableName, 2025-01-23:
OS ：ubuntu 20.04

MartyG-RealSense, 2025-01-25:
Hi @WhyxyxIsNotAnAvailableName This issue has not been previously reported.

Are you able to correct the resolution without adjusting it if you use the F8 key on the keyboard to toggle realsense-viewer between full-screen and windowed display mode, please?

WhyxyxIsNotAnAvailableName, 2025-01-26:
> Hi [@WhyxyxIsNotAnAvailableName](https://github.com/WhyxyxIsNotAnAvailableName) This issue has not been previously reported.
>
> Are you able to correct the resolution without adjusting it if you use [...]

Thanks for your reply! I've already gone home because of the winter break, and I'll try it as you say when school starts, and I'll probably get back to you in two weeks.

MartyG-RealSense, 2025-01-26:
Okay, that's no problem at all.  I look forward to your next report in two weeks.  Thanks very much for the update!

WhyxyxIsNotAnAvailableName, 2025-02-08:
> Okay, that's no problem at all. I look forward to your next report in two weeks. Thanks very much for the update!

Sorry, F8 key does not work. Could this issue be related to my GLFW driver? Because the following error occurs when using the 'realsense-viewer' command:    
```
08/02 11:42:05,667 INFO [140173010294656] (context.cpp:128) Found 2 RealSense devices (0xff requested & 0xff from device-mask in settings)
GLFW Driver Error: The GLFW library is not initialized
GLFW Driver Error: The GLFW library is not initialized
GLFW Driver Error: The GLFW library is not initialized
GLFW Driver Error: The GLFW library is not initialized
GLFW Driver Error: The GLFW library is not initialized
GLFW Driver Error: The GLFW library is not initialized
GLFW Driver Error: The GLFW library is not initialized
GLFW Driver Error: The GLFW library is not initialized  
```
 Here's the strange thing: even though I only had one camera connected to my PC, the INFO message still says two found.The phenomenon is still that when my computer is at 60Hz refresh rate, it is normal to use, 240Hz is not working.  
Please let me know if fixing this issue requires me to try more of the other things, thanks for the help.

WhyxyxIsNotAnAvailableName, 2025-02-09:
> I am not aware of a past case where a RealSense user has used a 240 Hz monitor refresh rate with the RealSense SDK / realsense-viewer software, so there is no previous reference to consult, unfortunately.
> 
> Does it still occur if you update your computer's video drivers?
> 
> I note your earlier statement that it works if you set 60 Hz first and then change to 240 Hz afterwards.
> 
> What happens if you input the command below into the Ubuntu terminal whilst realsense-viewer is closed and then launch realsense-viewer whilst in 240 hz mode?
> 
> `rm ~/.realsense-config.json`
> 
> What this command does is remove the configuration file of realsense-viewer and so force it to apply default values when it opens.

Sorry，`rm ~/.realsense-config.json` commend does not work either.   
The driver of my NVIDIA graphics card is relatively new and suitable, as I use the GPU to run other deep learning tasks normally. My GPU is an RTX 4060 and the driver version is 550.142

MartyG-RealSense, 2025-02-09:
Thanks for testing the rm command.

After the GLFW warning disappeared, does the F8 fullscreen toggle key work now?

WhyxyxIsNotAnAvailableName, 2025-02-10:
> Thanks for testing the rm command.
> 
> After the GLFW warning disappeared, does the F8 fullscreen toggle key work now?

sorry, F8 fullscreen toggle key does not work.  
But I was able to read the camera's RGB image and display it on the screen through the following simple python test program (it works fine whether my screen refresh rate is 60Hz or 240Hz).  
```
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
pipeline.start(config)
try:
    while True:
        frames = pipeline.wait_for_frames()
        color_frame = frames.get_color_frame()
        if not color_frame:
            continue
        color_image = np.asanyarray(color_frame.get_data())
        cv2.imshow('RealSense RGB', color_image)
        if cv2.waitKey(1) & 0xFF == ord('q'):
            break
finally:
    pipeline.stop()
    cv2.destroyAllWindows()
```  
My program running environment is as follows：  
```
python Version：3.10.0  
pyrealsense2 Version: 2.55.1.6486    
opencv-python Version: 4.10.0.84
```
Based on this phenomenon, I think this problem may not be related to my NVIDIA graphics card driver or video driver?

MartyG-RealSense, 2025-02-10:
Thanks very much for the update.

A difference between realsense-viewer and your Python program will be that whilst your program and the stream display panels of the Viewer will both use GLFW to display the image, the realsense-viewer also additionally uses a graphical user interface (GUI) called **ImGui** to display the contents of the options side-panel, of which the enlarged 'Add Source' button is a part.

WhyxyxIsNotAnAvailableName, 2025-02-10:
> Thanks very much for the update.
>
> A difference between realsense-viewer and your Python program will be that whilst your program and the stream display panels of the Viewer will both use GLFW to [...]

Will the question appear on **ImGui**? How should I check this?

MartyG-RealSense, 2025-02-10:
If you are using Python code then there is no way to check for a possible issue with ImGui in your own script as realsense-viewer is based on C++ code and there is not an equivalent implementation of imGuI in pyrealsense2 (Python supports GUI interfaces such as Pyglet and TKinter instead).  

You could post a question on the ImGUI GitHub's issues support page regarding whether there are previous known issues with ImGui at 240 Hz.

https://github.com/ocornut/imgui/issues

If someone in the RealSense community can implement a solution in the code of librealsense that resolves your 240 Hz problem then they would be very welcome to submit it as a Pull Request (PR) so that the RealSense development team can consider it for merging officially into the SDK.

https://github.com/IntelRealSense/librealsense/pulls

MartyG-RealSense, 2025-02-21:
Hi @WhyxyxIsNotAnAvailableName  Do you require further assistance with this case, please?  Thanks!

WhyxyxIsNotAnAvailableName, 2025-02-23:
> Hi [@WhyxyxIsNotAnAvailableName](https://github.com/WhyxyxIsNotAnAvailableName) Do you require further assistance with this case, please? Thanks!

No more for now, thanks for your help!

MartyG-RealSense, 2025-02-23:
You are very welcome.  As you do not need further help at this time, I will close this issue.  Thanks again!

closed 2025-02-23 by MartyG-RealSense