realsenseai/librealsense #13706: no attribute 'pipeline"

Oscar33391, 2025-01-23:
when i am working with jetson nano, i using the pyrealsense2 with Intel D4350i ,  and facing this error

Initialization error: module 'pyrealsense2' has no attribute 'pipeline'
Traceback (most recent call last):
  File "/home/jetson/jetracer/notebooks/test123.py", line 6, in <module>
    dc = DepthCamera()
  File "/home/jetson/jetracer/notebooks/realsense_depth.py", line 11, in __init__
    self.pipeline = rs.pipeline()
AttributeError: module 'pyrealsense2' has no attribute 'pipeline'

MartyG-RealSense, 2025-01-25:
Hi @Oscar33391  Does it make a difference if you change **import pyrealsense2 as rs** to the line below, please?

`import pyrealsense2.pyrealsense2 as rs`

Oscar33391, 2025-01-31:
hi @MartyG-RealSense , When i import pyrealsense2.pyrealsense2, it output ModuleNotFoundError: No module named 'pyrealsense2.pyrealsense2'

MartyG-RealSense, 2025-01-31:
Thanks very much for the information.  Please assist me with the questions below.

1.  I note that you refer to your camera above as **D4350i**.  Can you confirm whether its name is shown as D435i or D430i please?

2.  Which method did you use to install the pyrealsense2 wrapper on your Jetson Nano?

3.  Does pyrealsense2 work if you try the test script at the link below, or does it still say `AttributeError: module 'pyrealsense2' has no attribute 'pipeline'` when the script is run?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py

Oscar33391, 2025-01-31:
hi @MartyG-RealSense 

1, my camera is D435i

2. i dowload it by running these commands
  2.1 git clone https://github.com/IntelRealSense/librealsense.git
        sudo apt-get update && sudo apt-get upgrade
        sudo apt-get install python3 python3-dev -y
        sudo apt-get install libssl-dev
        cd librealsense
        mkdir build
        cd build
        cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true  -DPYTHON_EXECUTABLE=/usr/bin/python3.6
        make -j$(nproc)
        sudo make install
and then i edit my bashrc 
export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.6/pyrealsense2

source ~/.bashrc

3. it still say AttributeError: module 'pyrealsense2' has no attribute 'pipeline'

MartyG-RealSense, 2025-01-31:
The Python version referenced in the DBUILD_PYTHON_BINDINGS and PYTHONPATH lines has to match the Python 3 version that is actually installed on your computer or pyrealsense2 will not be able to locate Python.  So it would not work, for example, if you were setting the Python version as **python3.6** but actually had Python 3.9 installed on your computer. 

You can find the correct PYTHONPATH to use with the command `which python3`

You should not need to use the DPYTHON_EXECUTABLE parameter though unless you have multiple Python versions installed on the same computer.

********

If you do have Python 3.6 installed on your computer then it will not work with librealsense version 2.54.1 and newer as Python 3.6 support was removed in 2.54.1.

Oscar33391, 2025-01-31:
hi @MartyG-RealSense 

i changed my python version to python3.8 and when i run pipeline = rs.pipeline(), i still got the same error

Oscar33391, 2025-01-31:
and is it normal that only have 3 file in the pyrealsense folder?

[image]

MartyG-RealSense, 2025-01-31:
When the pyrealsense2 wrapper has been built from source code like you did, the important location to look in for wrapper files is the librealsense source code folder.

`librealsense > build > wrappers > python`

What files do you have in the 'python' folder at this location, please?

Oscar33391, 2025-01-31:
hi @MartyG-RealSense it have 5 file in this folder

CMakeFiles cmake_install.cmake Makefile pyrealsense2Config.cmake pyrealsense2ConfigVersion.cmake

MartyG-RealSense, 2025-02-01:
When pyrealsense2 is successfully built from source code, it should generate '.so' files, **librealsense2.so** and **pyrealsense2.so**.

When building the wrapper for Python 3, pyrealsense2.so will have an auto-generated complex filename with a number in it that depends on the Python 3 version being used.,  For example, using Python 9 may result in a filename such as  `pyrealsense2.cpython-39m-arm-linux-gnueabihf.so` 

[image]

The pyrealsense2.so file will typically be placed by the installation in the `build > wrappers > python` folder, whilst librealsense2.so will be elsewhere.

*********

Do you still get the pipeline attribute error if you input the line below into your terminal before you launch your Python script?

`export PYTHONPATH=$PYTHONPATH:/usr/local/lib`

Oscar33391, 2025-02-03:
hi @MartyG-RealSense 
due to some unknow reason, i cannot boot my jetson nano, so i redowloaded my image and real sense libary and i changed the version of librealsense to 2.49.0 and i forgot that some detail that i forgot to mention, after i successfully i changed to location of pyrealsense2 folder by running
 sudo mv /usr/local/liv/python3.6/pyrealsense2/ /usr/local/lib/python3.6/dist-packages/   
if i didn't do this step, it will output no module named pyrealsense2

Also i tired to run 
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
before launching the python script
import pyrealsense2
print(pyrealsense2.__version__)
it output attributeError: module 'pyrealsense2' has no attribute '__version__'

Oscar33391, 2025-02-03:
update i fixed everything by following the instructions on https://github.com/IntelRealSense/librealsense/issues/11506#issuecomment-1599168850

MartyG-RealSense, 2025-02-04:
IT's great to hear that you achieved a solution, @Oscar33391 - thanks very much for the update!

MartyG-RealSense, 2025-02-13:
Hi @Oscar33391  Do you require further assistance with this case, please?  Thanks!

Oscar33391, 2025-02-13:
No thank

closed 2025-02-13 by Oscar33391