realsenseai/librealsense #13707: pipeline.start() RuntimeError: No device connected

ashrafyahya, 2025-01-24:
Hi,

-I am trying to use the Intel RealSense Depth Camera D455 on a Jetson Orin Nano Dev Kit with JetPack 6.0. The camera is detected using the following methods:
    v4l2-ctl --list-devices
    lsusb
    rs-enumerate-devices
    realsense-viewer

- It works with realsense-viewer without any issues. Here is how I installed the camera:
sudo apt update
sudo apt upgrade -y
sudo apt install -y git cmake libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev
sudo apt install -y libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev
git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
mkdir build && cd build
cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true
make -j4
sudo make install
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && sudo udevadm trigger
cd ~/librealsense/build/examples/capture
./rs-capture
sudo apt install -y python3-dev python3-pip
pip3 install pyrealsense2

- My code to use the camera is as simple as this:
import pyrealsense2 as rs

# Initialize the pipeline
pipeline = rs.pipeline()
config = rs.config()

# Configure streams
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

# Start the pipeline
pipeline.start(config)
print("Camera successfully started!")

- However, I get this error:
pipeline.start() RuntimeError: No device connected

- What could be the reason?

MartyG-RealSense, 2025-01-25:
Hi @ashrafyahya  Can you first confirm whether launching your pyrealsense2 script in sudo admin powers mode makes a difference to camera detection, please?  For example:

`sudo python3 test.py`

ashrafyahya, 2025-01-25:
Hi @MartyG-RealSense thanks for your answer.

I'm using a python virtual environment so when I use 
  sudo python3 test.py 
I get "No module named pyrealsense2".

MartyG-RealSense, 2025-01-26:
What happens if you change the import instruction of your pyrealsense2 script to the line below and then try a sudo launch of the script again, please?

`import pyrealsense2.pyrealsense2 as rs`

ashrafyahya, 2025-01-26:
Hi again,

i tried it, it doesn't make any difference. I get the same error: 
    ModuleNotFoundError: No Module named 'pyrealsense2'.

ashrafyahya, 2025-01-26:
I suspect it has something to do with Python, because the camera is also recognized by C++ OpenCV.  
But I have no idea how to solve this problem with Python.

MartyG-RealSense, 2025-01-27:
Looking at your **cmake ...** CMake build instruction, I note that it does not include the flag `-DBUILD_PYTHON_BINDINGS:bool=true`

The pyrealsense2 Python wrapper will therefore have not been installed when you built the librealsense SDK, which could account for why the module cannot be found.

To build the wrapper without installing the entire SDK again, please try navigating to the **build** directory of your librealsense source code folder and then inputting the build command below.

`cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true`

ashrafyahya, 2025-01-27:
As you said, I ran the command   
      cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true  
 But I still get " No device connected"  

[image]  

[image]

MartyG-RealSense, 2025-01-27:
Now that the wrapper has been installed, please try a sudo launch again.

`sudo python3 testCam.py`

ashrafyahya, 2025-01-27:
I tried it but I get "No module named pyrealsense2".
So changed  the import row to 
       import pyrealsense2.pyrealsense2 as rs
I get the same error again:  
        No module named pyrealsense2

ashrafyahya, 2025-01-27:
I even recompiled again with:  
 ``sudo make uninstall && make clean && make && sudo make install``
Nothing has changed. I still get "No device connected"

MartyG-RealSense, 2025-01-27:
Could you provide an image of the files in the `librealsense > build > wrappers > Python` directory of your source code folder, please?

ashrafyahya, 2025-01-27:
Here is the image of ```ls librealsense/build/wrappers/python/ ``` again.

[image]

MartyG-RealSense, 2025-01-27:
It looks as though most of the files that I would expect to be in the **wrappers > Python** folder after a source code build of the wrapper are absent.

[image]

ashrafyahya, 2025-01-27:
So could you please write the right way to install realsense d455 on jetson orin nano dev kit, please? we may get it.

ashrafyahya, 2025-01-27:
I do not see any .so file. could you show me how you have installed realsense by you, please? we may get it.

MartyG-RealSense, 2025-01-27:
I do not see any .so files either.  I recall that there have been occasions where those files have been found in a location other than the build/wrappers/python directory that they are expected to be in.

If you have a Python 3 version that is between 3.7 and 3.10 (but not newer than 3.10) then it is possible to install pyrealsense2 on Arm-based computing devices such as Nvidia Jetson simply by using the command **pip install pyrealsense2** to install from PyPi packages instead of building it from source code.  The pip install method of wrapper installation does not require the .so files but it doesn't always work on Jetson boards though.

 **********

If installing with the pip method is not possible (if, for example, your Python version is 3.11 or 3.12) then building from source code is the next best installation method.  There are occasions where there are much fewer files in the /python folder than there should be after installation has completed though, like you experienced yourself.

[image]

When this happens, there is not much that can be done to correct it.  This is why the ideal situation is to be able to install from pip packages with **pip install pyrealsense2** instead of source code.

Does the **No device connected** message still appear if you input into the terminal the command below before running your script?

`export PYTHONPATH=$PYTHONPATH:/usr/local/lib`

ashrafyahya, 2025-01-27:
yes i use python 3.10.12 and yes i still see the message ``No device connected`` even when i input ``export PYTHONPATH=$PYTHONPATH:/usr/local/lib``

MartyG-RealSense, 2025-01-27:
As you have Python 3.10 there is a chance that the **pip install pyrealsense2** installation method will work so long as your librealsense version is 2.53.1 or newer as 2.53.1 is when pip install support for Python was introduced.  Can you confirm which librealsense version you are currently using, please?

ashrafyahya, 2025-01-27:
I am using the version 2.55.1.6486 as shown below:

[image]

MartyG-RealSense, 2025-01-27:
That message makes me think that the 2.55.1 version of pyrealsense2 has previously been installed on your Jetson using the 'pip install pyrealsense2' method.

I went back to the very first comment in this discussion and at the bottom of your installation script is `pip3 install pyrealsense2`.  So it looks like the wrapper was already earlier installed by the pip method because of that line in the install script.

Since the camera was already not being detected before the source-code install of the wrapper was performed with CMake, it is unlikely that building the wrapper from source has had a negative effect on the pip-installed wrapper.

ashrafyahya, 2025-01-27:
so should i remove the source installation and use only ``pip3 install pyrealsense2`` ?

MartyG-RealSense, 2025-01-27:
It sounds as though the wrapper was able to install successfully on your Jetson with the `pip install pyrealsense2` method, except for the camera non-detection problem. 

The pyrealsense2 wrapper needs the librealsense source installation to function, so removing the source-code installation of librealsense would break pyrealsense2.

Sometimes just wiping the computer's storage completely and installing everything from the beginning, including Ubuntu, can suddenly cause an unsolvable problem to be fixed because the wipe removed a hidden glitch that was causing problems.

Since `pip3 install pyrealsense2` is in your installation script, you should not need to repeat the pip install command after the installation script has completed.

MartyG-RealSense, 2025-02-06:
Hi @ashrafyahya  Do you require further assistance with this case, please?  Thanks!

ashrafyahya, 2025-02-08:
Hi @MartyG-RealSense,

In summary, I can say that Intel Realsense with Python 10 is unusable on Jetson Orin Nano with Ubuntu 22.
If your question is about closing the issue, yes you can close it.

Thanks for your help.

MartyG-RealSense, 2025-02-08:
Yes, I was asking whether the issue could be closed or whether you wanted to continue it.  As you said to close it, I will do so.  Thanks very much for the update!

closed 2025-02-08 by MartyG-RealSense