realsenseai/librealsense #13708: Building librealsense v2.51.1 from source for Jetson NX with Jetpack 5.1.1?

Atharva-05, 2025-01-24:
I want to build librealsense from source for my Nvidia **Jetson Orin NX**. Here are configuration details:
Jetpack version: 5.1.1 (L4T 35.3.1)
Kernel: 5.10.104-tegra
Ubuntu Version: 20.04
Required use with **ROS 2 Foxy**

I would like to know if this version of librealsense is compatible with my Jetpack installation.

According to this comment, librealsense v2.51.1 is the last supported version for ROS foxy.
> The compressed topics plugin for a specific ROS version should ideally be installed.  For Foxy that would be:
> 
> 
> 
> `sudo apt install ros-foxy-compressed-image-transport`
> 
> 
> 
> However, Foxy is only supported by the RealSense ROS2 wrapper up until version 4.51.1, which should be used with librealsense 2.51.1. 
> 
> 
> 
> An explanation of how image_transport works can be found at the link below, which states that "image_transport itself provides only "raw" transport so as not to impose unnecessary dependencies on client packages".
> 
> 
> 
> https://wiki.ros.org/image_transport
> 
> 
> 
> It is possible to **unsubscribe** from specific topics during runtime after launch has completed so that they are not included in the topic list.  Usually though the raw topics are active when publishing compressed topics.  
> 
> 
> 
> https://answers.ros.org/question/409553/how-can-i-remove-active-topic/ 

 _Originally posted by @MartyG-RealSense in [#13249](https://github.com/IntelRealSense/librealsense/issues/13249#issuecomment-2280014864)_

**However it is unclear whether this is suitable for Jetpack 5.**
___

Relevant comments that cause the doubt:

> That is an important detail.
> 
> 
> 
> And this is correct, JP5 (Based on U20) first official release release was 2.54.1.
> 
> Could be that 2.53.1 you see is some version we did to test the Debians integration.
> 
> 
> 
> I think this explains it. 

 _Originally posted by @Nir-Az in [#10526](https://github.com/IntelRealSense/librealsense/issues/10526#issuecomment-1717386150)_

> I think the only solution for you can be running a JP4 docker container on your Jetson NX JP5. 

 _Originally posted by @Nir-Az in [#10526](https://github.com/IntelRealSense/librealsense/issues/10526#issuecomment-1717480318)_

Is it recommended to build librealsense v2.51.1 from source for the given Jetpack configuration?

Thanks,
Atharva

MartyG-RealSense, 2025-01-25:
Hi @Atharva-05  JetPack 5 support was introduced later than librealsense 2.51.1, in version 2.54.1.

[image]

Before that, the newest JetPack version supported was 4.6.

Atharva-05, 2025-01-27:
@MartyG-RealSense Thanks for your reply!

How should I proceed then with the configuration I have?
Jetpack version: 5.1.1 with ROS 2 Foxy?

I tried with v2.54.1 but there is no data in the compressed RGB image. I am able to get the RGB and depth uncompressed image, and the compressed depth image.

MartyG-RealSense, 2025-01-27:
If you are using ROS2 Foxy, have you installed the compressed image transport module for Foxy using the command below?

`sudo apt install ros-foxy-image-transport`

This should publish the compressed color topic `/camera/camera/color/image_raw/compressed`

Atharva-05, 2025-01-27:
@MartyG-RealSense 
I have installed the image transport modules and can see the compressed color image topic, and am receiving data of the type sensor_msg/CompressedImage. The "data" field in it seems corrupted since it is blank.

Data on the similar topic for the compressed depth image is okay.

MartyG-RealSense, 2025-01-27:
My understanding is that the problem with the compressed color topic on Foxy is in the image transport plugin rather than the RealSense ROS wrapper.  So it is not something that the RealSense ROS development team could correct.

Would it be possible for your project to use a newer ROS2 version than Foxy, please?  Or are you constrained to using Foxy because you are using Ubuntu 20.04

Atharva-05, 2025-01-28:
@MartyG-RealSense Thanks for the information! I am constrained to a Ubuntu 20.04 + ROS 2 Foxy configuration for this project. My objective is to record compressed image data (both color and depth).

Could you suggest any other alternatives for this?

Also, I get the following error in the realsense node log when I try to view the compressed color image:
`
[realsense2_camera_node-1] [ERROR] [1738060749.788401996] [CompressedPublisher]: OpenCV(4.5.4) /home/ubuntu/build_opencv/opencv/modules/core/src/alloc.cpp:73: error: (-4:Insufficient memory) Failed to allocate 382617800244720 bytes in function 'OutOfMemoryError'
`

And the output of the ros2 topic echo command is:
```
header:
  stamp:
    sec: 1738060749
    nanosec: 716224365
  frame_id: camera_color_optical_frame
format: rgb8; jpeg compressed bgr8
data: []
```

The data on the compressed depth topic is okay. What could be the reason behind this inconsitency?

MartyG-RealSense, 2025-01-28:
If the problem is in Foxy specifically and you need to use Ubuntu 20.04 then the ROS2 version **Galactic** that came directly after Foxy is 20.04 compatible.  However, you mention that you are also unable to change from ROS2 Foxy unfortunately.

The `Failed to allocate ... bytes in function 'OutOfMemoryError'` problem and failure to publish `camera/camera/color/image_raw` was previously reported to have occurred on a Jetson with wrapper 4.54.1 at https://github.com/IntelRealSense/realsense-ros/issues/3105

In that discussion, a user at https://github.com/IntelRealSense/realsense-ros/issues/3105#issuecomment-2456909507 shared a workaround that resolved the issue in their particular situation.

Atharva-05, 2025-01-29:
@MartyG-RealSense Thank you for pointing to this! The suggested workaround worked for me as well! :)

MartyG-RealSense, 2025-01-29:
It's great to hear that the linked-to workaround worked for you too, @Atharva-05 - thanks very much for the update!

MartyG-RealSense, 2025-02-06:
Hi @Atharva-05  Do you require further assistance with this case, please?  Thanks!

Atharva-05, 2025-02-07:
Hello @MartyG-RealSense
Thanks for the assistance! I was able to resolve the issue of getting the compressed image as discussed. Closing the issue as no further assistance is needed.

closed 2025-02-07 by Atharva-05

MartyG-RealSense, 2025-02-07:
Hi @Atharva-05  It's great to hear that you achieved a solution.  Thanks very much for the update!