realsenseai/librealsense #13714: D435i does not boot up at all dead on arrival

ojingo, 2025-01-27:
Got a brand new d435i - nada no work tried different usb cables and different machines ... this unit is doa. How do we replace or refund!

ojingo, 2025-01-28:
> Hi @ojingo  If your D435i is brand new and was purchased from a retailer then it should be within the first 30 days of its sales warranty.  I would therefore recommend contacting the retailer that you bought it from to query about an 'RMA return' of the camera.

It’s from your online store You are the retailer Sent from my iPhone

MartyG-RealSense, 2025-01-28:
Official guidance about communicating with the RealSense Store about a return can be found at the link below.

https://support.intelrealsense.com/hc/en-us/articles/360051623953-How-to-return-a-faulty-Intel-RealSense-Camera

closed 2025-02-13 by MartyG-RealSense