realsenseai/librealsense #13717: L515 with rs-server throws error 404

Ali-AliAli-Ali, 2025-01-28:
| Required Info| |
|--------------|----------------|
| Camera Model | L515 |
| Platform (client) | PC |
| Operating System & Version (client) | Ubuntu 22.04 |
| Kernel Version (client) | 6.8.0-52-generic |
| Platform (server) | Raspberry Pi 4 |
| Operating System & Version (server) | Debian 12 |
| Kernel Version (server) | 6.6.51+rpt-rpi-v8 |
| SDK Version | 2.47.0 |
| Language | C++ |	

**The main question:** Does rs-server support L515? If yes, how to connect them?

**Context:**
I need to stream realtime RGBD-data from L515 connected to Raspberry Pi 4 over the Wi-Fi, connecting to RPi via ssh to create a SLAM algorithm. I tried both:
- RealSense Viewer, as written in the guide https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras,
- Python Ethernet example https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server.

Python example worked out but failed to transmit depth data as frames, making it a 2D depth map, from which I found it impossible to recalculate an exact 3D view even knowing camera intrinsics (if it is possible, I would really appreciate if you told me how!). 

As I found `rs-server`, I switched to C++, since `rs-server` isn't enabled for Python wrapper. Now RealSense Viewer and librealsense examples throw error 404 when` rs-server` is enabled on RPi 4 (and 107 when it isn't):
```
terminate called after throwing an instance of 'rs2::error'
  what():  [getStreams] error: 404 File Not Found, Or In Incorrect Format - 404
```
The error appears when creating `net_device` with RPi's static IP in local network. The exact location of error is a string constructor `basic_string.h`, row 456:
```
basic_string(const basic_string& __str)
      : _M_dataplus(_M_local_data(),
        _Alloc_traits::_S_select_on_copy(__str._M_get_allocator()))
      { _M_construct(__str._M_data(), __str._M_data() + __str.length()); }
```

If the camera is connected directly to pc, everything works well, but I need to use it as a peripheral device with RPi.

Thanks in advance!

Ali-AliAli-Ali, 2025-01-28:
Thanks @MartyG-RealSense for [responce in another issue]( https://github.com/IntelRealSense/librealsense/issues/9014#issuecomment-2618946679). Updated the issue. Still need help, please.

Ali-AliAli-Ali, 2025-01-28:
> It is also worth highlighting that librealsense version 2.47 does not support kernel 6.6 or 6.8. 6.6 is not supported and 6.8 support was not added until SDK 2.56.2 Beta, which does not support the rs-server tool (which was removed in 2.54.1 onwards).

Could I ask more questions based on your responce, please?
1. What are the consequences of the lack of kernel support? There have been no problems with librealsense on PC so far.
2. Does SDK 2.56.3 Beta support L515? In according [Release note](https://github.com/IntelRealSense/librealsense/releases/tag/v2.56.3) L515 is mentioned in Supported devices section.

MartyG-RealSense, 2025-01-28:
Hi @Ali-AliAli-Ali  Instead of using networking, you could consider generating a 3D pointcloud on a Pi with an L515 attached to it, and control the Pi directly over wi-fi from another computer or a tablet using a remote access tool such as Google's **Chrome Remote Desktop** - https://remotedesktop.google.com/

1.  It is possible to use librealsense without support for a particular kernel, but there may be unforeseen consequences in regards to stability unless librealsense has been built from source code in 'libuvc backend' or 'RSUSB backend' mode so that the kernel is bypassed, helping to avoid kernel related problems.

2.  The final librealsense version to officially support L515 was 2.50.0.  Using it with recent librealsense versions can result in the L515 being undetected by RealSense programs.

MartyG-RealSense, 2025-01-28:
No more updates supporting L515, including networking between computers, will be released in the future unfortunately. 

Potentially you could communicate between computers using ROS1 Noetic and the RealSense ROS1 wrapper, using guidance in the multi-computer pointcloud tutorial at the link below.

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

Ali-AliAli-Ali, 2025-01-29:
Hello @MartyG-RealSense! Thank you for the guidance and advices. 

If you don't mind I would ask to leave the issue open for 2 weeks in order to respond with feedback when I finish implementing data processing and transferring between RPi and PC. I may need further assistance and would aprecciate it if you could provide it.

MartyG-RealSense, 2025-01-29:
You are very welcome.  It's no problem at all to leave the issue open for 2 weeks.  Good luck!

closed 2025-03-08 by MartyG-RealSense