realsenseai/librealsense #13723: Pyrealsense no device connected

Aterox-os, 2025-01-29:
Hello,
I recently started using my D435 with my M2 Macbook. I had a lot of issues with the RealSense viewer ( #13640) but got it working through parallels which works for now since I actually just need pyrealsense. I tried running the default python test:
```
import pyrealsense2 as rs
pipe = rs.pipeline()
profile = pipe.start()
try:
  for i in range(0, 100):
    frames = pipe.wait_for_frames()
    for f in frames:
      print(f.profile)
finally:
    pipe.stop()
```
But after some time I get the following error:
```
Traceback (most recent call last):
  File "/Users/finn/Desktop/pyrealsense/main.py", line 3, in <module>
    profile = pipe.start()
RuntimeError: No device connected
```
I tried running `rs-enumerate-devices` but it says that there is no camera connected. If I run `system_profiler SPUSBDataType` the output is:
```
2025-01-29 21:30:10.581 system_profiler[21985:135905] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
USB:

    USB 3.1 Bus:

      Host Controller Driver: AppleT8112USBXHCI

    USB 3.1 Bus:

      Host Controller Driver: AppleT8112USBXHCI

        Intel(R) RealSense(TM) Depth Camera 435 :

          Product ID: 0x0b07
          Vendor ID: 0x8086  (Intel Corporation)
          Version: 50.cb
          Manufacturer: Intel(R) RealSense(TM) Depth Camera 435 
          Location ID: 0x00100000
```
So the camera is connected and recognised by macOS. 
I also tried running my script with sudo but then I get the following error:
```
Traceback (most recent call last):
  File "/Users/finn/Desktop/pyrealsense/main.py", line 1, in <module>
    import pyrealsense2 as rs
  File "/Users/finn/Desktop/pyrealsense/.venv/lib/python3.13/site-packages/pyrealsense2/__init__.py", line 2, in <module>
    from .pyrealsense2 import *
ImportError: dlopen(/Users/finn/Desktop/pyrealsense/.venv/lib/python3.13/site-packages/pyrealsense2/pyrealsense2.cpython-313-darwin.so, 0x0002): Library not loaded: @rpath/librealsense2.2.56.dylib
  Referenced from: <0B5AC2F3-B89A-37B4-BEF2-DB8017891ED4> /Users/finn/Desktop/pyrealsense/.venv/lib/python3.13/site-packages/pyrealsense2/pyrealsense2.2.56.3.cpython-313-darwin.so
  Reason: no LC_RPATH's found
```

I am running version 2.56.3 and build it manually. I think it's a problem with the permission but I could find anything online on how to resolve the issue. Has someone solved the problem?

MartyG-RealSense, 2025-01-30:
Hi @Aterox-os  It sounds as though the camera works at first and then later disconnects and becomes unavailable.  Is that what is happening, please?

Are you using a USB C to C cable on your Mac?  If you are then these can have more problems with RealSense cameras than a USB A to C cable like the official 1 meter one supplied with the camera?

Aterox-os, 2025-01-30:
I am not sure if the camera disconnects. I can see that macOS recognizes it but the librealsense tools can't. If I keep running `system_profiler SPUSBDataType` while the program tries to connect to the camera its still visible. So I would say that it is disconnecting. I am using a 3rd party USB c to USB c cable since my mac doesnt have a USB a port. I currently dont have a USB c to USB a cable lying around since I bought the camera second hand without a cable. I can buy a USB a cable and try using it with a adapter.

MartyG-RealSense, 2025-01-30:
Thanks very much for the update.  

If the camera is initially detected when the script starts and then disconnects, that could also be a symptom of insufficient power being provided to the camera by the USB system to enable the streams. 

I look forward to your next report.  Good luck!

Aterox-os, 2025-01-30:
Ok so I actually found a cable and I am clueless now. If I use the USB C to USB A cable with an adapter I get a segmentation fault??? I also tried using a different adapter but it's the same result. Any ideas?

MartyG-RealSense, 2025-01-30:
What method did you use to install pyrealsense2, please?

There is a pre-built package of pyrealsense2 for MacOS created by a RealSense community member that can be found at the link below.

https://github.com/cansik/pyrealsense2-macosx

Aterox-os, 2025-01-30:
I tried to build it manually with the correct make flag and install it in my venv with pip. Both method didn't work. I installed the prebuild package in venv. Is it correct that the import is still `pyrealsense2`? Because if it is then it doesn't work for me. I still get the error that no device was detected.

MartyG-RealSense, 2025-01-30:
Yes, usually the line is `import pyrealsense2 as rs`

However, on some computers it is necessary to use `import pyrealsense2.pyrealsense2 as rs`

Aterox-os, 2025-01-30:
I tried it still no device connected.

MartyG-RealSense, 2025-01-31:
Do you get the same message about no device connected if you try running other pyrealsense2 example scripts, such as the tutorial one here:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py

MartyG-RealSense, 2025-01-31:
Hi @Nyanziba  Some Mac users have found that the realsense-viewer will sometimes be able to eventually start if the `sudo realsense-viewer` launch command is repeated multiple times (four or more).

The current theory for the problems with using RealSense cameras on Mac is that it was caused by a change by Apple in their USB policy.  It is discussed at https://github.com/IntelRealSense/librealsense/issues/11815#issuecomment-1557960723

My Intel RealSense colleagues welcome RealSense Mac users investigating the problems, discussing it amongst themselves and providing feedback that may help towards improving Mac compatibility.

MartyG-RealSense, 2025-02-21:
Hi @Aterox-os  Have you been able to make any progress with your issue, please?  Thanks!

closed 2025-03-08 by MartyG-RealSense