realsenseai/librealsense #13724: v2.56.3 build fails from psl undefined references

sebastianserna30, 2025-01-30:
**Description**
Build error when compiling librealsense with CHECK_FOR_UPDATES=ON. The error occurs during linking of the realsense-viewer executable due to missing PSL (Public Suffix List) symbols that are required by the static libcurl library.

**Error Message**
```
[ 79%] Linking CXX executable ../../Release/realsense-viewer
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(psl.c.o): in function `Curl_psl_destroy':
psl.c:(.text+0x29): undefined reference to `psl_free'
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(psl.c.o): in function `Curl_psl_use':
psl.c:(.text+0xbb): undefined reference to `psl_latest'
/usr/bin/ld: psl.c:(.text+0x187): undefined reference to `psl_builtin'
/usr/bin/ld: psl.c:(.text+0x1a1): undefined reference to `psl_free'
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(cookie.c.o): in function `Curl_cookie_add':
cookie.c:(.text+0xca2): undefined reference to `psl_is_cookie_domain_acceptable'
collect2: error: ld returned 1 exit status
```

**System**
- Ubuntu 22.04.1 LTS
- Kernel: 6.8.0-51-generic
- x86_64
- libpsl-dev/jammy 0.21.0-1.2build2 amd64 [installed]
- libpsl5/jammy 0.21.0-1.2build2 amd64 [installed]

**Steps to Reproduce**
1. Clone librealsense repository
2. In a build dir configure with CHECK_FOR_UPDATES enabled:
```bash
cmake .. -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DCHECK_FOR_UPDATES=ON
```
4. Build:
```bash
make
```

**Additional Information**
- libpsl is properly installed on the system
- nm analysis of libcurl.a shows undefined PSL references
- Multiple attempts to fix link order (-lpsl before/after libcurl.a) did not resolve the issue

sebastianserna30, 2025-01-30:
Vieweng the symbol table with nm:
`
nm ./libcurl/libcurl_install/lib/libcurl.a | grep psl_`

gives:

```
                 U Curl_psl_release
                 U Curl_psl_use
                 U psl_is_cookie_domain_acceptable
                 U Curl_psl_destroy
0000000000000000 T Curl_psl_destroy
00000000000001b0 T Curl_psl_release
0000000000000040 T Curl_psl_use
                 U psl_builtin
                 U psl_free
                 U psl_latest
                 U Curl_psl_destroy
                 U psl_check_version_number
```

So apparently a lot of undefined symbols.

MartyG-RealSense, 2025-01-30:
Hi @sebastianserna30  You likely do not need to include **-DCHECK_FOR_UPDATES=ON** as this flag is usually not used, except in instructions for installing librealsense on an old Apple Mac computer with an Intel CPU chip instead of on a PC.

sebastianserna30, 2025-01-30:
I find interesting that make says:

`make[2]: *** No rule to make target 'tools/realsense-viewer/libcurl/libcurl_install/lib/libcurl.a', needed by 'Release/realsense-viewer'.  Stop.
`

but actually libcurl is being installed at 
`libcurl/libcurl_install/lib/libcurl.a`

from external_libcurl.cmake:
`DCMAKE_INSTALL_PREFIX=${CMAKE_CURRENT_BINARY_DIR}/libcurl/libcurl_install`

So maybe realsense-viewer is looking in the current dir and not in the global dir

Nir-Az, 2025-01-30:
@sebastianserna30  this looks like a local issue.
We build according to the command you shared on a fresh VM's and Ubuntu 20/22/24 LTS with CHECK_FOR_UPDATES=ON and no issues.
However, If you can find a suitable fix that will help your configuration and will not break the native installation, we welcome you to open a PR and we will review and merge if it works.

sebastianserna30, 2025-01-30:
> Hi [@sebastianserna30](https://github.com/sebastianserna30) You likely do not need to include **-DCHECK_FOR_UPDATES=ON** as this flag is usually not used, except in instructions for installing librealsense on an old Apple Mac computer with an Intel CPU chip instead of on a PC.

True, I could leave that out. I have it because if I configure cmake with CHECK_FOR_UPDATE=false, then this error gets bypassed. (As then all the libcurl stuff is skipped)

sebastianserna30, 2025-01-30:
> [@sebastianserna30](https://github.com/sebastianserna30) this looks like a local issue. We build according to the command you shared on a fresh VM's and Ubuntu 20/22/24 LTS with CHECK_FOR_UPDATES=ON [...]

@Nir-Az, thank you for testing in an isolated VM. That helps!  I will post here if I make the PR.

MartyG-RealSense, 2025-02-06:
Thanks so much!   :)

MartyG-RealSense, 2025-03-08:
Hi @sebastianserna30  Have you created your PR yet, please?  Thanks!

MartyG-RealSense, 2025-03-15:
Thanks very much for creating the PR.  I have added an **Enhancement** label to this issue to ensure that it is kept open whilst your PR is active.  Thanks again!

ztony478, 2025-04-15:
build with ./configure have the same error 

Ubuntu 24.04.1 LTS
x86_64
libpsl-dev/jammy 0.21.0-1.2build2 amd64 [installed]
libpsl5/jammy 0.21.0-1.2build2 amd64 [installed]

nm libcurl.a | grep psl_
                 U Curl_psl_release
                 U Curl_psl_use
                 U psl_is_cookie_domain_acceptable
                 U Curl_psl_destroy
0000000000000000 T Curl_psl_destroy
0000000000000190 T Curl_psl_release
0000000000000040 T Curl_psl_use
                 U psl_builtin
                 U psl_free
                 U psl_latest
                 U Curl_psl_destroy
                 U psl_check_version_number

kaidokert, 2025-09-28:
I just ran into the same issue and #13852 worked around it without changing anything else ( building from trunk )

Nir-Az, 2025-09-29:
@sebastianserna30 / @kaidokert 
Thanks for reporting, I just created #14315 which should solve your issues.
I will run some testing and update when merged