realsenseai/librealsense #13725: Found 0 RealSense devices - Jetson Orin Nano & D435i

c3hodgins, 2025-01-30:
Hello,
I am trying to setup my Jetson Orin Nano to work with the Intel RealSense D435i Depth Camera, and the RealSense Viewer application is not recognizing any RealSense devices.

I was able to connect the camera to the Jetson Orin Nano in the fall, but since then I had to reflash the operating system and thus lost the working version of the RealSense viewer. 

Model: Jetson Orin Nano
Kernel: 5.15.148-tegra
OS: Ubuntu 22.04.5 Jammy

I have followed the installation guide under the libuvc_installation.sh script, changing the line 46 to add the -DFORCE_RSUSB_BACKEND tag. The software builds after about 20-30 minutes, and functions on the software side quite well, launching realsense-viewer. I believe it is not a hardware or usb driver issue because lsusb does correctly display the camera as a realsense device, as well as no issues in the dmesg console, showing a pretty normal usb connection protocol.

Additionally, when launching the realsense-viewer app it enumerates the video files present in /sys/class/../video4linux/video* but does not actually find the video feeds to be a realsense device. Here is the output log of running realsense-viewer:

 30/01 16:14:31,436 INFO [281472906827840] (backend-v4l2.cpp:753) Enumerating UVC video0 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video0
 30/01 16:14:31,642 INFO [281472906827840] (backend-v4l2.cpp:753) Enumerating UVC video1 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video1
 30/01 16:14:31,644 INFO [281472906827840] (backend-v4l2.cpp:753) Enumerating UVC video2 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video2
 30/01 16:14:31,645 INFO [281472906827840] (backend-v4l2.cpp:753) Enumerating UVC video3 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video3
 30/01 16:14:31,646 INFO [281472906827840] (backend-v4l2.cpp:753) Enumerating UVC video4 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.3/video4linux/video4
 30/01 16:14:31,647 INFO [281472906827840] (backend-v4l2.cpp:753) Enumerating UVC video5 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.3/video4linux/video5
 30/01 16:14:32,299 INFO [281472906827840] (backend-v4l2.cpp:753) Enumerating UVC video0 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video0
 30/01 16:14:32,300 INFO [281472906827840] (backend-v4l2.cpp:753) Enumerating UVC video1 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video1
 30/01 16:14:32,301 INFO [281472906827840] (backend-v4l2.cpp:753) Enumerating UVC video2 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video2
 30/01 16:14:32,302 INFO [281472906827840] (backend-v4l2.cpp:753) Enumerating UVC video3 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.0/video4linux/video3
 30/01 16:14:32,303 INFO [281472906827840] (backend-v4l2.cpp:753) Enumerating UVC video4 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.3/video4linux/video4
 30/01 16:14:32,304 INFO [281472906827840] (backend-v4l2.cpp:753) Enumerating UVC video5 realpath=/sys/devices/platform/bus@0/3610000.usb/usb2/2-1/2-1.2/2-1.2:1.3/video4linux/video5
 30/01 16:14:32,315 INFO [281472906827840] (context.cpp:128) Found 0 RealSense devices (0xff requested & 0xff from device-mask in settings)

Like I mentioned earlier, at some point I was able to get the realsense to work, and I am currently unsure of how to reproduce the environment I was when successfully launching the app.

MartyG-RealSense, 2025-01-31:
Hi @c3hodgins  On line 46 of the libuvc_installation.sh build script, please try adding the flags `-DBUILD_EXAMPLES=TRUE -DBUILD_GRAPHICAL_EXAMPLES=TRUE` and then rebuilding librealsense with the script.

The **-DBUILD_GRAPHICAL_EXAMPLES=TRUE** flag causes support for graphics to be included in the SDK build.

c3hodgins, 2025-02-04:
@MartyG-RealSense Yes my line 46 in the libuvc_installation.sh script looks like this, yet it is still not detecting the realsense device. Is it possible that I can detect the realsense device in the pyrealsense2 library without the realsense viewer software detecting the device because the jetson recognizes the realsense device?

(line 46)
cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true

I got this from a different thread https://github.com/IntelRealSense/realsense-ros/issues/2481#issuecomment-1253662954 .

c3hodgins, 2025-02-04:
Also notable I cannot open /dev/video2 (supposed to be the rgb channel on the realsense d435i) using software like cheese as it returns the error: 

JPEG parameter struct mismatch: library thinks size is 584, caller expects 728

c3hodgins, 2025-02-04:
Update again:
Confirmed video stream using opencv on the different video inputs in /dev though both pyrealsense and realsense ros wrapper unable to detect the realsense device. As suspected the issue is with the connecting the video streams to the software recognizing the realsense device.

MartyG-RealSense, 2025-02-05:
Your line 46 looks fine.

'/video2' is the default channel number for RGB on the D435i, but /video channel numbers can change depending on what attached devices other than the RealSense camera that the computer detects.  At https://github.com/IntelRealSense/librealsense/issues/11553#issuecomment-1463442865 the subject of setting fixed addresses for devices with **symbolic links** via udev device-handling rules is discussed.

Were you using Ubuntu 22.04 Jammy on your previous setup before you reflashed the OS, please?

A scenario that sometimes occurs is that realsense-viewer can detect the camera but pyrealsense2 cannot.  So if realsense-viewer cannot detect the camera then it is unlikely that pyrealsense2 will detect it either.

Have you tried launching realsense-viewer in sudo admin powers mode with the command `sudo realsense-viewer`

c3hodgins, 2025-02-12:
@MartyG-RealSense Yes, we have tried using the root user to run the RealSense-viewer application. We have surmised that this issue is holding us back from our use case and to work around it by just implementing OpenCV to get the RGB camera stream, instead of going through the RealSense-SDK in the meantime. Hoping to return to this issue further down the line to address the issue as we may need it to improve functionality for our system.

MartyG-RealSense, 2025-02-13:
Would you like to close this issue for now if you are not working on it right now and re-open it at a future date or create a new issue?  Thanks!

MartyG-RealSense, 2025-02-28:
Case closed for now.

closed 2025-02-28 by MartyG-RealSense

MartyG-RealSense, 2025-03-25:
Hi @ssharma0704  Is the camera detectable if you quit out of the launch with Ctrl-C, unplug and re-insert the camera and then try the launch again?

ssharma0704, 2025-03-26:
@MartyG-RealSense it's the same output "No RealSese devices were found!"

MartyG-RealSense, 2025-03-27:
And the camera non-detection also occurs if you launch with ros2 run instead of ros2 launch?

`ros2 run realsense2_camera realsense2_camera_node`

Normally I would not recommend having realsense-viewer open and streaming during a ROS session, but is the camera detected if you enable the depth stream in realsense-viewer first and then perform the ROS launch?

ssharma0704, 2025-04-03:
[

](url)

So here I ran the ros2 run realsense2_camera realsense2_camera_node first, it didn’t detect the camera, I ran the realsense-viewer I was getting the depth image. I ran the ros2 run realsense2_camera realsense2_camera_node after the getting depth image in the realsense-viewer as well but still it did not detect the camera

MartyG-RealSense, 2025-04-04:
@ssharma0704  Your ROS launch log and your RealSense Viewer image show that you have firmware driver version **5.13.0.50** installed in your camera and you are currently using version **2.56.3** of the Viewer, and ROS2 wrapper version **4.55.1**.

[image]

If your Viewer version is 2.56.3 then your librealsense SDK version is also likely 2.56.3, rather than 2.55.1.

Ideally, ROS wrapper version **4.56.3** should be used with librealsense 2.56.3, whilst you are currently using the older 4.55.1 wrapper that was designed for librealsense 2.55.1.

Most importantly, it is recommended that firmware **5.16.0.1** is used with 2.56.3, and also with wrappers 4.55.1 and 4.56.3.

When the **Firmware update recommended** box pops up in the top corner of the Viewer when it is launched, would it be possible for you to click the **Install** button to update the camera to firmware 5.16.0.1 to see if this makes the camera detectable in your ROs launch, please?

ssharma0704, 2025-04-04:
ros2 launch realsense2_camera rs_launch.py 
[INFO] [launch]: All log files can be found below /home/rover/.ros/log/2025-04-04-14-01-20-567375-rover-desktop-59331
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [59333]
[realsense2_camera_node-1] [INFO] [1743793281.127832463] [camera.camera]: RealSense ROS v4.56.3
[realsense2_camera_node-1] [INFO] [1743793281.128263673] [camera.camera]: Built with LibRealSense v2.56.3
[realsense2_camera_node-1] [INFO] [1743793281.128319291] [camera.camera]: Running with LibRealSense v2.56.3
[realsense2_camera_node-1]  04/04 14:01:21,163 WARNING [281472396683488] (ds-motion-common.cpp:455) No HID info provided, IMU is disabled
[realsense2_camera_node-1]  04/04 14:01:21,166 ERROR [281472396683488] (rs.cpp:256) [rs2_create_device( info_list:0xffff4c01ebe0, index:0 ) UNKNOWN] bad optional access
[realsense2_camera_node-1]  04/04 14:01:21,166 ERROR [281472396683488] (rs.cpp:256) [rs2_delete_device( device:nullptr ) UNKNOWN] null pointer passed for argument "device"
[realsense2_camera_node-1]  04/04 14:01:21,166 WARNING [281472396683488] (rs.cpp:392) null pointer passed for argument "device"
[realsense2_camera_node-1] [WARN] [1743793281.166536256] [camera.camera]: Device 1/1 failed with exception: bad optional access
[realsense2_camera_node-1] [ERROR] [1743793281.166789638] [camera.camera]: The requested device with  is NOT found. Will Try again.

MartyG-RealSense, 2025-04-05:
@ssharma0704  RealSense cameras that are equipped with an IMU, like the D435i that you have, may be undetected on Jetsons that are using JetPack 6. This is because an instruction called **hiddraw** that RealSense IMU camera models rely upon was removed in JetPack 6.

Workarounds include:

1.  Building the librealsense SDK from source code with the **libuvc backend** installation method, using the pre-made `libuvc_installation.sh` build script.  This installation method bypasses the kernel.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh
 
Or

2.  Build the RealSense MIPI driver at the link below to add support for IMU-equipped cameras on JetPack 6.  This is a complicated installation process though.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

ssharma0704, 2025-04-08:
Hey @MartyG-RealSense  Thank You for helping me out! The last instruction you sent helped me a lot! and it is also working for D455F and D435i. Thanks for the replying promptly. I believe that there are a lot people who are trying to figure this out and struggling to make it work.

MartyG-RealSense, 2025-04-08:
Hi @ssharma0704  It's great to hear that you were successful.  The MIPI driver is likely the ideal solution to this JP6 issue for RealSense users with a Jetson Xavier or Orin, as it is purposely designed to address the problem for those Jetson models.  

For those who have a different model of Jetson or are unable to perform the MIPI driver's installation procedure, a more general approach is recommended of building librealsense from source code in libuvc backend or RSUSB backend mode, and additionally downgrading the firmware driver to 5.13.0.50 if the camera is still not detected.  

This firmware downgrade cannot be performed with D455F though as the minimum firmware this model supports is 5.15.0.2 and installing 5.13.0.50 on a D455F would render the camera inoperable.

ssharma0704, 2025-04-08:
@MartyG-RealSense I used the libuvc backend installation method, and i tried it on Nvidia Orins and Seeed Orin Nxs it works on both with JP6.2 and i used the D455F on both the computers and it worked just fine

MartyG-RealSense, 2025-04-08:
Thanks very much again for the confirmation of the method you used for a successful outcome.  For some Jetson users the libuvc backend is sufficient to resolve the JP6 issue without having to take the further step of a firmware change.

lijin255, 2025-06-24:
> [@MartyG-RealSense](https://github.com/MartyG-RealSense) I used the libuvc backend installation method, and i tried it on Nvidia Orins and Seeed Orin Nxs it works on both with JP6.2 and i used the D455F on both the computers and it worked just fine我使用了 libuvc 后端安装方法，并在 Nvidia Orins 和 Seeed Orin Nxs 上尝试过，它与 JP6.2 兼容，并且我在两台计算机上都使用了 D455F，并且运行良好
I am currently experiencing similar problems with Jetson Orin. Can I add your email address so I can communicate with you?
best wishes！