realsenseai/librealsense #13728: Found 0 device on Windows 11

TtopsBo, 2025-01-31:
Hello,
I downloaded Realsense Viewer v2.55.1 on Windows 11, but my computer could not find it using a USB 3.2 Gen1 / Gen2 port. My mainboard is Gigabyte Z890, and the chip is Ultra 9 285K, but no Graphics card is available. However, the camera works on Ubuntu 20.04 on Jetson Nano. Is it caused by the compatibility of my PC?

MartyG-RealSense, 2025-01-31:
Hi @TtopsBo  I recommend first checking the **camera privacy** settings on Windows 11 to confirm whether or not permission to access your RealSense camera is being blocked.

1.  Type **privacy** into the search box at the bottom of your Windows screen to find and open the Privacy Settings interface.

2.  In the search box of the 'Privacy & Security' panel, type **camera** to find and access the camera privacy settings.

[image]

3.  Finally, ensure that 'Camera access' and 'Let apps access your camera' are both set to **On**.

[image]

TtopsBo, 2025-01-31:
Thanks for your quick response. I checked the access permissions, and they are all set to on. When searching for a camera, no camera was found. Even when I used Win+X to search for equipment, there was also nothing.

MartyG-RealSense, 2025-02-01:
Thanks very much for the further information.

The Ultra 9 285K CPU uses integrated **Intel Arc** graphics technology and so can provide graphics without the need for a separate graphics card.

If the camera works on Ubuntu then it sounds as though there is not a problem with the camera hardware, its firmware driver or the USB cable.

Does the camera remain undetected if you unplug the micro-sized end of the USB cable from the camera, turn the connector around the other way and re-insert it into the camera?  USB-C cables are two-way insertion at the micro-sized end, and a RealSense camera will sometimes perform better or only be detected if the micro-sized connector is in one particular orientation of the two insertion directions available.

Can you also confirm which RealSense camera model you are using, please.  Thanks!

TtopsBo, 2025-02-01:
The model is D435. Actually, I have already tried what you mentioned. The camera firmware is 2.56.0.1. Thanks for your advice.

MartyG-RealSense, 2025-02-01:
On Windows you can enable frame metadata in the RealSense Viewer simply by clicking the 'Enable' button on the pop-up in the top corner of the Viewer window when it appears.

TtopsBo, 2025-02-01:
I must admit that it works well on my laptop with USB 3.0. The compatibility of my desktop USB 3.2 is not suitable because the port is always detected as USB 2.1. Could this be the reason that the detection is not so successful as it is USB 2.1?

I also find that the driver is provided by Microsoft, date 2006/6/21, version 10.0.26100.2894. Is it appropriate, or can I get the correct driver?

MartyG-RealSense, 2025-02-01:
Although USB is a standard, not all USB ports are made equal and on some computers the camera may be repeatedly detected as USB 2.1 even if the port is USB 3.  

Microsoft drivers often have old dates like 2006, so that is normal and not a cause of the mis-detection problem.

There have been a very small number of past cases where a RealSense camera had problems on a USB 3.2 port but worked fine with USB 3.1 Gen 1. Using a USB 3.1 hub instead of attaching the camera directly to the computer's USB 3.2 ports might make a difference.  A hub has its own USB controller separate from the controller that handles the computer's ports.

TtopsBo, 2025-02-01:
Many thanks for your suggestions.

MartyG-RealSense, 2025-02-06:
Hi @TtopsBo  Do you require further assistance with this case, please?  Thanks!

TtopsBo, 2025-02-06:
HI, @MartyG-RealSense .Thanks for your support, and you can close it.

closed 2025-02-06 by MartyG-RealSense