realsenseai/librealsense #13730: Suggestions for Resolving Frame Drop Issues and hardware sync with RealSense Camera at 60Hz

Akumar201, 2025-02-02:
I am experiencing frame drop issues while using a RealSense 405 camera at 60Hz. My primary requirement is to capture data consistently at 60Hz or higher without dropped frames. I understand that multiple factors can affect performance, including the camera model, cable quality, and system configuration. However, I wanted to seek guidance from experts who may have tested and validated solutions for this issue. Specifically, I would appreciate recommendations on:
1. The best RealSense camera model for achieving stable 60Hz performance.
2. Any specific hardware or cables that could help minimize frame drops.
3. Whether hardware synchronization ([Intel's hardware sync guide](https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration)) can help in this scenario. From my understanding, hardware sync may have limitations in terms of achievable frame rates, but I would love any insights on this.

Ideally it would be great if a camera can do 60HZ with no frame drop and hardware sync option. Such that I can attach multiple camera but get the data synchronized to my computer. 

Thank You

MartyG-RealSense, 2025-02-02:
Hi @Akumar201  The D405 camera model is not equipped with a set of connector pins necessary for hardware sync that other RealSense 400 Series camera models have.  

Hardware sync is not likely to be of benefit for the goal of reducing frame drops, but is very helpful if you will be using multiple cameras and want to try to sync them to the timing of a master signal.

If a USB cable longer than 2 meters is used then as you mentioned, cable quality will be important .  It should also be a cable that is designed for data transfer and not an inexpensive charging / device sync cable.

For 5 meters length or longer, a cable with 'active repeater' components inside it to boost the signal is recommendable.

A programming measure that can reduce the risk of frame drops is to add latency by increasing the **frame queue size** so that the pipeline can hold a greater number of frames simultaneously.  This will increase the amount of computer memory consumed though, so a computer with good memory capacity will be beneficial when using this technique.

https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0#latency-vs-performance

MartyG-RealSense, 2025-02-13:
Hi @Akumar201  Do you require further assistance with this case, please?  Thanks!

Akumar201, 2025-02-17:
@MartyG-RealSense, Thank You for your help , I am closing this issue for now.

Akumar201, 2025-03-26:
Hello @MartyG-RealSense , I have a follow-up question. I’m wondering whether it’s better to use the realsense-ros2 wrapper or work directly with librealsense to get the best performance. Also, is librealsense available in C++, and would there be any performance difference between the two approaches?

MartyG-RealSense, 2025-03-26:
Yes C++ is fully supported.  It is the librealsense SDK's default language and so does not require the installation of a wrapper to program in it.

Typically, working directly with librealsense scripting will give better performance, as the ROS wrapper operates as a layer on top of librealsense.

Akumar201, 2025-03-27:
Thank You for your reply. I would like to know do you have measured the difference in performance between pyrealsense vs librealsense.

MartyG-RealSense, 2025-03-27:
I recommend disregarding the Frame Drops per Second graph in the realsense-viewer tool as the information on it may not be relevant to actual performance.  For example, you can have a situation where the streams are performing smoothly at 30 FPS with no obvious lag whilst the Frame Drops graph has huge peaks on it.

Clicking on the 'i' information button on the top of a stream panel to overlay statistics such as actual real-time FPS at the top of the panel is a better guide to performance.

I'm not aware of a case where pyrealsense2's performance has been formally benchmarked against C++ but if there is a difference then it is not significant enough to be concerned about.  Some types of programming function are easier to implement in C++ than they are in pyrealsense2 though, as in Python you have the complication of storing data in arrays.

closed 2025-04-10 by Akumar201