realsenseai/librealsense #13732: disable width stream causes rgb frame to be darker

rxwen, 2025-02-03:
Hello, 

I'm using D405 camera. One thing I noticed is that when a depth stream is disabled, the captured rgb frame will be darker. How will the darkness be related to the depth stream? 

The sample code is like:
```
pipeline = rs.pipeline() 
config = rs.config()
config.enable_device(serial_number)
config.enable_stream(
    rs.stream.color,
    640,
    480,
    rs.format.bgr8,
    60
)
config.enable_stream(
    rs.stream.depth,  # type: ignore
    640,
    480,
    rs.format.z16,
    60
) # if this config is disabled, the captured rgb frame will be darker.

frames = pipeline.wait_for_frames(timeout_ms=2000)
 color_frame = frames.get_color_frame() # it will be darker

```

Thanks in advance.

MartyG-RealSense, 2025-02-03:
Hi @rxwen  The D405 camera model does not have a dedicated RGB sensor or RGB settings and so generates its RGB image by obtaining it the depth sensor and then passing it through an Image Signal Processor (ISP) chip to produce the final RGB image.

Because the RGB image comes from the depth sensor, the RGB image is affected by changing the depth settings, such as the exposure value.

I was not able to replicate the issue you experienced of the RGB image becoming darker when depth is disabled.  However, you can brighten the D405's RGB by setting a higher depth exposure value (its default exposure value is 33000). Setting a custom exposure value will automatically disable auto-exposure though.

On the D405 model you can also obtain an alternative RGB image from its 'depth from left infrared sensor' mode.  This is done by defining config.enable code for an **infrared** stream but setting its format to BGR8 instead of the default infrared format of Y8.

MartyG-RealSense, 2025-02-13:
Hi @rxwen  Do you require further assistance with this case, please?  Thanks!

rxwen, 2025-02-16:
It's ok. Thanks.

closed 2025-02-16 by rxwen