realsenseai/librealsense #13736: Issue Installing RealSense Packages on WSL2 (Kernel 5.15.167.4-1)

mganglb, 2025-02-03:
Hi all,

I am experiencing issues installing the RealSense packages on WSL2. Based on similar reports from others, this seems to be a recurring problem. I am wondering if this could be related to the latest kernel update: Kernel version: 5.15.167.4-1 ([Microsoft WSL2 Kernel Releases](https://github.com/microsoft/WSL2-Linux-Kernel/releases)).
I was following this Guidelines: https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide

## System Information

### Windows 11

```
PS C:\Users\▮▮▮> wsl --version
WSL version: 2.3.26.0
Kernel version: 5.15.167.4-1
WSLg version: 1.0.65
MSRDC version: 1.2.5620
Direct3D version: 1.611.1-81528511
DXCore version: 10.0.26100.1-240331-1435.ge-release
Windows version: 10.0.22631.4751
```

### WSL Environment

```
~/▮▮▮r$ hostnamectl
 Static hostname: ▮▮▮
       Icon name: computer-container
         Chassis: container
      Machine ID: ▮▮▮▮
         Boot ID: ▮▮▮▮
  Virtualization: wsl
Operating System: Ubuntu 22.04.5 LTS                      
          Kernel: Linux 5.15.167.4-microsoft-standard-WSL2
    Architecture: x86-64
```

### Failing Command
```
sudo apt-get install librealsense2-dkms
sudo apt purge librealsense2-dkms 
sudo apt remove librealsense2-dkms
```

### Output

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
librealsense2-dkms is already the newest version (1.3.28-0ubuntu1).
The following packages were automatically installed and are no longer required:
  docker-scan-plugin libsecret-1-0 libsecret-common libu2f-udev
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y

Setting up librealsense2-dkms (1.3.28-0ubuntu1) ...
Removing old librealsense2-dkms-1.3.28 DKMS files...
Deleting module librealsense2-dkms-1.3.28 completely from the DKMS tree.
Loading new librealsense2-dkms-1.3.28 DKMS files...
/usr/src/librealsense2-dkms-1.3.28/dkms.conf: line 39: [: 5.15: integer expression expected
/usr/src/librealsense2-dkms-1.3.28/dkms.conf: line 101: [: 5.15: integer expression expected
Building for 5.15.167.4-microsoft-standard-WSL2
Module build for kernel 5.15.167.4-microsoft-standard-WSL2 was skipped since the
kernel headers for this kernel does not seem to be installed.

Loading the modified modules into kernel...modprobe: FATAL: Module uvcvideo not found.
dpkg: error processing package librealsense2-dkms (--configure):
 installed librealsense2-dkms package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 librealsense2-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

### Additional Information
tes system ended up in a weired state. librealsense2-dkms was not installed succesfully nor is it possible to delete it.

```
~▮▮▮▮$ sudo apt purge librealsense2-dkms || sudo apt remove librealsense2-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  dctrl-tools dkms docker-scan-plugin libsecret-1-0 libsecret-common libu2f-udev linux-headers-5.15.0-131 linux-headers-5.15.0-131-generic linux-headers-generic
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  librealsense2-dkms*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 13.2 MB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 147328 files and directories currently installed.)
Removing librealsense2-dkms (1.3.28-0ubuntu1) ...
Deleting module librealsense2-dkms-1.3.28 completely from the DKMS tree.
Reloading original modules into kernel...modprobe: FATAL: Module uvcvideo not found.
dpkg: error processing package librealsense2-dkms (--remove):
 installed librealsense2-dkms package post-removal script subprocess returned error exit status 1
dpkg: too many errors, stopping
Errors were encountered while processing:
 librealsense2-dkms
Processing was halted because there were too many errors.
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

### Possible Cause

It appears that the kernel headers for the latest WSL2 kernel (5.15.167.4-1) are missing or not properly installed, preventing the module from building successfully.

### Steps to Reproduce

- Install WSL2 with Ubuntu 22.04.
- Run sudo apt-get install librealsense2-dkms.
- Observe the error message related to kernel headers and module loading failure.

### Expected Behavior
The installation should complete without errors, and the required RealSense modules should be available in the kernel.

If anyone has successfully installed librealsense2-dkms on this kernel, please share any required workarounds.

Would appreciate any insights or fixes for this issue. Thanks!

MartyG-RealSense, 2025-02-03:
Hi @mganglb  A discussion at https://github.com/IntelRealSense/librealsense/issues/13506 highlights how other RealSense users of WLS2 have had problems installing librealsense from dkms packages on it but have succeeded when building from source code.

mganglb, 2025-02-03:
I am aware of similar problems, and tried out most of the approaches already.
- https://github.com/IntelRealSense/librealsense/issues/12810#issuecomment-2047281195
- https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

### libuvc
I have tried that approach beforehand, but it am not able to remove the "broken" version of  librealsense2-dkms. Out of that reason the script fails as well.

```
~/RAS_AIPRISM/2025/ai_prism_docker$ sudo  ./libuvc_installation.sh
+ exec
+ exec
+ '[' 0 -ne 0 ']'
+ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.5 LTS
Release:        22.04
Codename:       jammy
++ uname -r
+ echo 'Kernel version 5.15.167.4-microsoft-standard-WSL2'
Kernel version 5.15.167.4-microsoft-standard-WSL2
+ sudo apt-get update
Hit:1 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:2 https://download.docker.com/linux/ubuntu jammy InRelease                                                                                                                                
Hit:3 https://dl.google.com/linux/chrome/deb stable InRelease                                                                                                                                 
Hit:4 https://nvidia.github.io/libnvidia-container/stable/deb/amd64  InRelease                                                                               
Hit:5 https://librealsense.intel.com/Debian/apt-repo jammy InRelease                                                                   
Hit:6 http://archive.ubuntu.com/ubuntu jammy InRelease                                                           
Hit:7 http://archive.ubuntu.com/ubuntu jammy-updates InRelease                             
Hit:8 https://packages.microsoft.com/repos/code stable InRelease                           
Hit:9 http://archive.ubuntu.com/ubuntu jammy-backports InRelease     
Hit:10 https://packagecloud.io/dirk-thomas/vcstool/ubuntu jammy InRelease
Reading package lists... Done
+ cd /root/
+ sudo rm -rf ./librealsense_build
+ mkdir librealsense_build
+ cd librealsense_build
++ sudo swapon --show
++ wc -l
+ '[' 2 -eq 0 ']'
+ echo Installing Librealsense-required dev packages
Installing Librealsense-required dev packages
+ sudo apt-get install git cmake libssl-dev freeglut3-dev libusb-1.0-0-dev pkg-config libgtk-3-dev unzip -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
pkg-config is already the newest version (0.29.2-1ubuntu3).
freeglut3-dev is already the newest version (2.8.1-6).
cmake is already the newest version (3.22.1-1ubuntu1.22.04.2).
git is already the newest version (1:2.34.1-1ubuntu1.12).
libgtk-3-dev is already the newest version (3.24.33-1ubuntu2.2).
libssl-dev is already the newest version (3.0.2-0ubuntu1.18).
libusb-1.0-0-dev is already the newest version (2:1.0.25-1ubuntu2).
unzip is already the newest version (6.0-26ubuntu3.2).
The following packages were automatically installed and are no longer required:
  dctrl-tools dkms docker-scan-plugin libsecret-1-0 libsecret-common libu2f-udev linux-headers-5.15.0-131 linux-headers-5.15.0-131-generic linux-headers-generic
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  librealsense2-dkms
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 13.2 MB disk space will be freed.
(Reading database ... 145614 files and directories currently installed.)
Removing librealsense2-dkms (1.3.28-0ubuntu1) ...
Reloading original modules into kernel...modprobe: FATAL: Module uvcvideo not found.
dpkg: error processing package librealsense2-dkms (--remove):
 installed librealsense2-dkms package post-removal script subprocess returned error exit status 1
dpkg: too many errors, stopping
Errors were encountered while processing:
 librealsense2-dkms
Processing was halted because there were too many errors.
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

### Additional Information
sudo dpkg --remove --force-remove-reinstreq librealsense2-dkms ||
sudo apt-get purge librealsense2-dkms ||
sudo apt-get remove librealsense2-dkms ||

``` 
The following packages were automatically installed and are no longer required:
  dctrl-tools dkms docker-scan-plugin libsecret-1-0 libsecret-common libu2f-udev linux-headers-5.15.0-131 linux-headers-5.15.0-131-generic linux-headers-generic
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  librealsense2-dkms
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 13.2 MB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 145614 files and directories currently installed.)
Removing librealsense2-dkms (1.3.28-0ubuntu1) ...
Reloading original modules into kernel...modprobe: FATAL: Module uvcvideo not found.
dpkg: error processing package librealsense2-dkms (--remove):
 installed librealsense2-dkms package post-removal script subprocess returned error exit status 1
dpkg: too many errors, stopping
Errors were encountered while processing:
 librealsense2-dkms
Processing was halted because there were too many errors.
E: Sub-process /usr/bin/dpkg returned an error code (1)```

mganglb, 2025-02-04:
### Update

- we were able to remove the "broken" pkg, by following this guidlines: https://phoenixnap.com/kb/fix-sub-process-usr-bin-dpkg-returned-error-code-1
- Running https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md those commands after, worked for us.
- Now only one small issue remains
[image]

Ty so much
Markus

MartyG-RealSense, 2025-02-04:
It's great news that you were successful!

The multiple udev rules warning is easy to fix.  It usually occurs if librealsense has been installed from both packages and source code on the same computer.  If you delete the **/etc** rule from your computer then the warning should disappear.  

The /lib rule should not be deleted, as doing so can break linkages between packages.

mganglb, 2025-02-04:
Tank you for the hint.

As we attempt to utilize the Python package, we've run into another problem — the camera isn’t being recognized. I’d be grateful for any assistance.

## Steps to reproduce
1) pip install pyrealsense2
2) run following script
```
import pyrealsense2 as rs

print(f"Fetch context")
ctx = rs.context()
print(f"number of devices found: {len(ctx.devices)}")
for idx in range(0, len(ctx.devices)):
    d = ctx.devices[idx]
    print(f"d: {d}")
    serial_number = d.get_info(rs.camera_info.serial_number)
    print(f"serial_number: {serial_number}")
    name = d.get_info(rs.camera_info.name)
    print(f"name: {name}")

# output:
# Fetch context
# number of devices found: 0
```

## Expected behaviour
- The connected realsense is listed

## addtional Information
- Realsense-Viewer is detecting camera in WSL2
- `lsusb` (in WSL2) lists the Realsense camera as well
- to attach the USB device, we followed the following Guidelines:  https://learn.microsoft.com/en-us/windows/wsl/connect-usb#attach-a-usb-device.

br
Markus

MartyG-RealSense, 2025-02-05:
Have you tried launching your Python script in sudo mode?  For example: `sudo python3 test.py`