realsenseai/librealsense #13739: Is there a way to get a translation from color to depth rather than *the* color translated to depth.

timprepscius, 2025-02-03:
align gives back an aligned image.

I want an image where each pixel is a tuple of u16 x, u16 y

where x,y is the depth image pixel for the given color pixel.

Is this possible without going in and ripping up align_neon::align_other_to_z

MartyG-RealSense, 2025-02-04:
Hi @timprepscius  The function **rs2_project_color_pixel_to_depth_pixel()** might meet your needs.  It converts a color pixel to a depth pixel without having to perform alignment.

https://intelrealsense.github.io/librealsense/doxygen/rsutil_8h.html#af04b7b1ba50b1d9e03089d1e5186d2f8

A C++ example of use of this instruction is at https://github.com/IntelRealSense/librealsense/issues/2982

timprepscius, 2025-02-04:
It doesn't work

timprepscius, 2025-02-04:
I think the depth image is too noisy.

I've read the algorithm over and over and it seems like it should work, but not if the depth is too noisy, then it's a random probability.

I'm guessing the align uses some algorithm to rule out pixels which have already been placed or something?

MartyG-RealSense, 2025-02-04:
I'm not aware of such a filtering algorithm in the alignment system.

Alignment is regulated by a component of the SDK called the **align processing block**.  What this mainly does is automatically take account of differences in resolution between the depth and color streams, and resizes the field of view to the FOV size of the target stream.  

For example, on a D435 camera which has a smaller FOV size on its RGB sensor than on its depth sensor, performing **depth to color** alignment resizes the depth FOV to that of the color FOV, causing some of the detail at the outer edges of the aligned image to be cut off.  

If **color to depth** alignment is performed then the color FOV size is resized to that of the depth FOV.  If the depth FOV is larger than color then the aligned image is stretched out to match the larger depth FOV size.

timprepscius, 2025-02-04:
Ok -

So - what I’m seeing with regard to the color to depth pixel is that:

If I run alignment - then the aligned color does  match up to the
depth/infrared 0

If I do color to depth - pixel by pixel - only need 20 or so pixels- the
depth pixel is often slightly off.  It almost seems like a skew issue but
I’ve checked that the intrinsics and extrinsics are correct.

I believe I’m preparing all of the data correctly -

What would you guess is happening?

Am I  correct that I should deconstruct that align function and see if that
can provide me with reliable depth pixels?

What would you do?

I must have a mapping of color to depth - not an aligned color.

MartyG-RealSense, 2025-02-04:
If you have a D405, D415 or D455 type camera (D455, D455f, D456, D457) then it supports a feature called 'color from left infrared sensor' where an RGB image is obtained from the left infrared sensor instead of the RGB sensor.  In this mode, the color image is perfectly aligned with the depth map, but is not a depth-color aligned image.  More information about this mode if you have a compatible camera can be found at the link below.  The D435 type cameras do not support this mode, unfortunately.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#use-the-left-color-camera

A drawback with this mode is that on all the supporting cameras except D405 the camera's infrared emitter component will overlay a semi-random pattern of dots over the RGB image, meaning that the emitter has to be turned off in order to remove the dot pattern if you do not want the pattern.  This can reduce the quality of the depth image and result in more holes unless the camera is used in a well illuminated location.

Regarding intrinsics, if the image is not using alignment then the depth intrinsics will be used for the depth image and color intrinsics for the color image.  

After depth to color alignment the color intrinsics are used for the aligned image because the origin of depth changes from the center of the left IR sensor to the center of the RGB sensor.

If the incorrect type of intrinsics are used (such as depth intrinsics instead of color intrinsics) then this can result in incorrect values in the aligned image.

closed 2025-02-21 by MartyG-RealSense