realsenseai/librealsense #13741: API Suggestion: Improve rs2::frame::get_sensor()

axbycc-mark, 2025-02-05:
In my `frame_callback(rs2::frame)` I was looking for a way to retrieve the sensor that  created the frame. I only found `rs2_sensor* rs2::frame::get_sensor()` which confusingly returns a C API object from a CPP API funcion. By chance, I found the other fuction `rs2::sensor_from_frame` which seems to be what I want.

https://github.com/IntelRealSense/librealsense/blob/e1688cc318457f7dd57abcdbedd3398062db3009/include/librealsense2/hpp/rs_sensor.hpp#L357-L361

My suggestion is to fold this function into the `rs2::frame::get_sensor()` where useres will likely be looking for it, or at least add documentation in `rs2::frame::get_sensor()` which points to this function.

Also, I believe this function has one too many shared ptrs. The `rs2::sensor` itself already contains a shared ptr. This should do.

```
 inline sensor sensor_from_frame(frame f) 
 { 
     std::shared_ptr<rs2_sensor> psens(f.get_sensor(), rs2_delete_sensor); 
     return  sensor(psens); 
 } 
```

MartyG-RealSense, 2025-02-05:
Hi @axbycc-mark  Thanks so much for your code suggestion!  We welcome the submission of code changes from RealSense users in the form of a **Pull Request (PR)** so that the RealSense development team can consider merging it into the librealsense SDK.  Pulls can be submitted at the page at the link below.

https://github.com/IntelRealSense/librealsense/pulls

MartyG-RealSense, 2025-02-13:
Hi @axbycc-mark  Do you require further assistance with this case, please?  Thanks!

axbycc-mark, 2025-02-13:
No assistance required, thank you. I'll submit a PR if I find the time.

closed 2025-02-13 by axbycc-mark