realsenseai/librealsense #13746: Unable to calibrate IMU for D435i

sushantgov, 2025-02-05:
Console keeps outputting warnings about the camera moving even though it's in a stable position, can someone please help?
 
Device 943222072764 successfully updated to FW: 5.16.0.1
root@hal9000:~# python rs-imu-calibration.py
python: can't open file 'rs-imu-calibration.py': [Errno 2] No such file or directory
root@hal9000:~# cd librealsense/tools/rs-imu-calibration/
root@hal9000:~/librealsense/tools/rs-imu-calibration# python rs-imu-calibration.py
waiting for realsense device...
  Device PID:  0B3A
  Device name:  Intel RealSense D435I
  Serial number:  943222072764
  Product Line:  D400
  Firmware version:  05.16.00.01
Start interactive mode:
FOUND GYRO with fps=200
FOUND ACCEL with fps=63
-------------------------
*** Press ESC to Quit ***
-------------------------

Align to direction:  [ 0. -1.  0.]   Mounting screw pointing down, device facing out
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING                                                       
 Status.collect_dataWARNING: MOVING

MartyG-RealSense, 2025-02-06:
Hi @sushantgov  If the camera is on a completely flat surface and stationary and the IMU calibration tool still consistently acts as though it is still detecting movement then there is not a fix for this unfortunately and I would recommend to stop using the calibration tool.

As Accel is being detected as having a value of '63', this indicates that your D435i was manufactured some years ago, before mid 2022, as the default Accel value became '100' on cameras manufactured after mid 2022 because the IMU hardware component inside the camera was changed to a different but similar component.

May I confirm that you have taken your hand away from the camera once it has been rotated to the target angles of [ 0. -1. 0.] please?  Once the camera has been positioned, it needs to be left stationary for a short time before the calibration process begins, indicated by a series of full-stop dots that increase in number until there is a row of around twenty of them.

If you are concerned about the accuracy of your IMU then you can ensure that a feature called Motion Correction is enabled, which 'fixes' raw IMU data to make it more accurate.  In the RealSense Viewer tool this IMU data auto-correction feature is enabled by default.

joomamma, 2025-02-06:
I am having the same issue. Left the camera running for a good 15 minutes, but it still says "moving"

When I run rs-motion, it shows the camera rotating erratically for a few seconds before it stabilizes.

When you say, don't use the calibration tool, is there another tool we can use? or do we trust that the 435i camera is calibrated without the tool?

MartyG-RealSense, 2025-02-06:
Hi @joomamma  There is not an alternative IMU calibration tool.  However, in the majority of results of a completed calibration that users of this tool have shared on this support forum, the result of the existing calibration stored in the camera is close to the ideal target value and it should therefore not require a re-calibration.

sushantgov, 2025-02-06:
I can confirm that my hand was off the sensor and stable for more than 25 of those status messages. I tried using Kalibr as well but I received an optimization error with respect to spline coefficient buffer being too large, not sure if it's something to do with the IMU, but the VINS-Fusion application only seems to run well in stereo mode and drifts extremely when IMU is enabled. Is it wise to move to the newer realsense D435i models if IMU calibration and accuracy is more important for my application?

MartyG-RealSense, 2025-02-07:
If you run the librealsense SDK's C++ example program **rs-motion** then this will give you an indication about how stable your IMU is.  It displays a graphical representation of a camera which rotates as the IMU values change.  If the IMU is unstable then the on-screen image of the camera will typically spin wildly on its own when the program launches without you touching the camera.  

If it doesn't spin then it may still be over-sensitive, shifting a lot with only a small movement of the camera.  rs-motion demonstrates how code can be used to stabilize the camera by adjusting an 'alpha' value to make it less sensitive.

https://github.com/IntelRealSense/librealsense/tree/master/examples/motion

If you installed the librealsense SDK with tools and example programs included then you should be able to find a pre-built executable version of rs-motion in Ubuntu's **usr/local/bin** directory.

**********

Another way of testing IMU stability is to enable the **Motion Module** in the realsense-viewer tool so that you can observe how stable the displayed IMU values for Accel and Gyro are.

*********

If the IMU of your old camera manufactured in early 2022 or older consistently demonstrates wild fluctuations without being moved then you could certainly consider replacing it with a modern model.

closed 2025-02-28 by MartyG-RealSense