realsenseai/librealsense #13748: m

DanielBaldrigue, 2025-02-06:
| | |
|---|---|
|**librealsense**|2.56.3 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2025-02-06:
Hi @DanielBaldrigue  This error usually occurs when the **Add Source** button at the top of the RealSense Viewer's options side-panel has been used and a camera, such as the built-in RGB webcam in a laptop, has been unable to be accessed.  Some RealSense users also use this button if their RealSense camera has not been automatically detected.

Can you confirm whether this error occurred when the Add Source button was used, please?

Also, which RealSense camera model do you have?

DanielBaldrigue, 2025-02-25:
> Hi @DanielBaldrigue <https://github.com/DanielBaldrigue> Do you require
> further assistance with this case, please? Thanks!

I hope this email finds you well
The problem has been solved!
Thank you so much for the help!
You can delete my case from GitHub!
Best regards,
Daniel Baldrigue

MartyG-RealSense, 2025-02-25:
You are very welcome.  I'm pleased that I could help.  As you have found a solution, I will close the case.  Thanks again!