realsenseai/librealsense #13749: pyrealsense gyro

Oscar33391, 2025-02-07:
when i run my script

import pyrealsense2 as rs

# Create a pipeline
pipeline = rs.pipeline()

# Create a config and enable the gyro stream
config = rs.config()
config.enable_stream(rs.stream.gyro)

# Start the pipeline
pipeline.start(config)

try:
    while True:
        # Wait for a coherent set of frames
        frames = pipeline.wait_for_frames()

        # Get gyro frame
        gyro_frame = frames.first_or_default(rs.stream.gyro)

        if gyro_frame:
            # Get gyro data
            gyro_data = gyro_frame.as_motion_frame().get_motion_data()
            print(f"Gyro X: {gyro_data.x}, Gyro Y: {gyro_data.y}, Gyro Z: {gyro_data.z}")

finally:
    # Stop the pipeline
    pipeline.stop()

it output
 Traceback (most recent call last):
  File "/home/jetson/jetracer/notebooks/gyro_test.py", line 11, in <module>
    pipeline.start(config)
RuntimeError: Failed to set frequency 2. device path: /sys/devices/70090000.xusb/usb1/1-2/1-2.4/1-2.4:1.5/0003:8086:0B3A.0001/HID-SENSOR-200076.2.auto/iio:device1/in_anglvel_sampling_frequency Last Error: Permission denied
then i ran 
sudo chmod 666 /sys/devices/70090000.xusb/usb1/1-2/1-2.4/1-2.4:1.5/0003:8086:0B3A.0001/HID-SENSOR-200076.2.auto/iio:device1/in_anglvel_sampling_frequency
then it output 
python3: /home/jetson/librealsense/src/linux/backend-hid.cpp:826: uint32_t librealsense::platform::iio_hid_sensor::get_channel_size() const: Assertion `!_channels.empty()' failed.
Aborted (core dumped)

how can i fix it?

MartyG-RealSense, 2025-02-07:
Hi @Oscar33391  I note that you have an Nvidia Jetson.  Are you using JetPack 6 on it, please?  If you are then RealSense cameras equipped with an IMU may not provide IMU data or sometimes not be detected at all,  This is because JetPack 6 removed an instruction called **hiddraw** that IMU-equipped cameras (known as HID devices) relied upon.

Downgrading the JetPack version from 6 to 5.1.3 may enable the IMU to work.

**********

If it is not possible for you to downgrade from JetPack 6 then building the librealsense SDK from source code with the **libuvc backend** installation procedure at the link below may work.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The libuvc method provides a pre-made build script called **libuvc_installation.sh**

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

For some RealSense users, the IMU still does not function after installing librealsense with the libuvc backend procedure.  Some users got it working after taking the further step of downgrading their camera's firmware driver to an earlier version.

I would not recommend downgrading to an older firmware than 5.13.0.50.  For the **D455f** camera model the minimum firmware used should be **5.15.0.2** and not older than that, otherwise the D455f will be unable to operate.

*******

Another method of adding IMU support for JetPack 6 is to install the RealSense **MIPI driver** at the link below.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

Oscar33391, 2025-02-12:
I am currently using jetpack 4.5.1, is there any method i can enable this function with this vesion？

MartyG-RealSense, 2025-02-12:
Although JetPack 5.1.3 was the most recent JP version that the IMU worked with before JP6, it should be able to work with JP 4.5.1 too. 

JetPack 4 is meant to be used with Ubuntu 18.04.  JetPack 5 was designed for use with Ubuntu 20.04.  Which Ubuntu version are you using JP 4.5.1 with, please?

Oscar33391, 2025-02-13:
hi @MartyG-RealSense i am using the Ubuntu 18.04.5

Oscar33391, 2025-02-13:
i fixed the problem by changing the dictionary location, i put the pyrealsense2.so file in the home page which make it not working. in order to fix it, i use command sudo mv pyrealsense2.so (file path to the dist-packages file in python3.6)

MartyG-RealSense, 2025-02-13:
It's great to hear that you achieved a solution.  Thanks very much for the update and for sharing the details of your method!

closed 2025-02-21 by MartyG-RealSense