realsenseai/librealsense #13751: Android. Receiving depth data

HappyElon, 2025-02-09:
Hello, 

I'm trying to make a project. It's goal to measure the linear dimensions of an object. I need to obtain a depth image simultaneously with the RGB image to achieve my goal. But there's a problem: I can't obtain data either directly from the FrameSet or through getDistance().

My first try was receiving image from FrameSet like this:

```Java
public void onCaptureButtonClick(View view) {
        if (!mIsStreaming) {
            Toast.makeText(this, "Streaming is not active", Toast.LENGTH_SHORT).show();
            return;
        }

        try {
            // Frame receiving
            try (FrameSet frames = mPipeline.waitForFrames()) {
                try (FrameSet processed = frames.applyFilter(mColorizer)) {
                    // RGB saver
                    saveRgbFrame(processed);

                    // Depth saver
                    saveDepthFrame(processed);

                    Log.d(TAG, "Captured RGB and Depth frames");
                    Toast.makeText(this, "Frame captured successfully", Toast.LENGTH_SHORT).show();
                }
            }
        } catch (Exception e) {
            Log.e(TAG, "Error capturing frame: " + e.getMessage());
            Toast.makeText(this, "Failed to capture frame", Toast.LENGTH_SHORT).show();
        }
    }

private void saveDepthFrame(FrameSet frames) {
        try (Frame f = frames.first(StreamType.DEPTH)) {
            if (f == null) {
                Log.e(TAG, "Invalid or missing DepthFrame in the FrameSet");
                return;
            }

            DepthFrame depthFrame = f.as(Extension.DEPTH_FRAME);
            int width = depthFrame.getWidth();
            int height = depthFrame.getHeight();

            byte[] bytearray = new byte[width * height * 2];

            if (bytearray.length != width * height * 2) {
                Log.e(TAG, "Incorrect bytearray size for depth data");
                return;
            }

            depthFrame.getData(bytearray);

            int[] depthData = new int[width * height];
            for (int i = 0; i < depthData.length; i++) {
                int low = ((Byte.toUnsignedInt(bytearray[i * 2])));
                int high = (Byte.toUnsignedInt(bytearray[i * 2 + 1])) << 8;
                Log.e("DATA CHECK", String.valueOf(bytearray[i*2]) + "+" + bytearray[i*2+1]);
                depthData[i] = high | low; // Assembling 16-bit numeric
            }

            // Saving file
            String filename = "depth_frame_" + System.currentTimeMillis() + ".txt";
            File file = new File(getExternalFilesDir(null), filename);
            try (FileWriter writer = new FileWriter(file)) {
                for (int value : depthData) {
                    writer.write(value + "\n");
                }
                Log.d(TAG, "Raw depth data saved to " + file.getAbsolutePath());
            }
        } catch (Exception e) {
            Log.e(TAG, "Failed to save raw depth data", e);
        }
    }
```

In onCaptureButtonClick() I handle button, which is used for image shot. In the code above you can see how I call my saveDepthFrame() after receiving FrameSet and applying Colorizer. In saveDepthFrame() I receive depth image from FrameSet as it was shown in examples and then copy it to the buffer by .getData() method. And after that I don't understand what do I have to do with received data because it's definetely broken. I used my python code to restore image from data (I used this code before when I was experimenting with camera and it's doing its job correctly) and I saw that image has a lot of noise in it, values are changing from 0 to 100000.

 Clearly data was damaged while receiveing it through code.

I also tried to run examples from main librealsense branch. Java example doesn't connect camera streams. I give permission to app to connect camera but I still receive "Wait for camera" message on screen. Capture example is working correctly, but it doen't show any depth frame data usage. So I modified it like this just to receive some data. (I also commented this mBackGroundText.setVisibility(state ? View.VISIBLE : View.GONE); to make textView show me result)

```Java

Runnable mStreaming = new Runnable() {
        @Override
        public void run() {
            try {
                try(FrameSet frames = mPipeline.waitForFrames()) {
                    try(FrameSet processed = frames.applyFilter(mColorizer)) {
                        mGLSurfaceView.upload(processed);
                        try (Frame f = processed.first(StreamType.DEPTH)){
                            DepthFrame depth = f.as(Extension.DEPTH_FRAME);
                            try {
                                mBackGroundText.setText(String.valueOf(depth.getDistance(depth.getWidth() / 2, depth.getHeight() / 2)));
                            } catch (Exception e){
                                mBackGroundText.setText("GET DISTANCE ERROR " + e);
                            }
                            Log.e("DEPTH VALUE", String.valueOf(depth.getDistance(depth.getWidth() / 2, depth.getHeight() / 2)));
                        } catch (Exception e) {
                            Log.e("Streaming error", "Streaming error");
                        }
                    }
                }
                mHandler.post(mStreaming);
            }
            catch (Exception e) {
                Log.e(TAG, "streaming, error: " + e.getMessage());
            }
        }
    };

```

But now I receive this error message

[image]

How do I receive depth data in Z16 format? Are there any way to do it without byte[] array?

Thanks in advance for your help.

MartyG-RealSense, 2025-02-10:
Hi @HappyElon  https://github.com/IntelRealSense/librealsense/issues/4498 has Java code that a RealSense Android user shared for converting Z16 data to ARGB 8888 format and displaying it.

MartyG-RealSense, 2025-02-21:
Hi @HappyElon  Do you require further assistance with this case, please?  Thanks!

HappyElon, 2025-02-24:
Hi, 

Thanks for you help. Unfortunately I moved to the another location for work and have no access to my realsense camera for now. So I can't test solutions you provided, but they seem to be solving my problem. I close this issue, if I will have some questions I'll ask them in another issue.

Thanks again for giving me the fast answer.