realsenseai/librealsense #13755: Frame freeze

egleorl1, 2025-02-10:
Hello! I am using an Intel RealSense D415 camera in my setup, where I need precise alignment between RGB and depth streams. My objective is to map keypoints extracted from YOLO-Pose from the 2D plane (RGB) to the 3D plane (depth). The pose estimation model runs on saved RGB frames, and the depth alignment is critical for ensuring the correct Z-coordinate assignment to keypoints.

However, I am encountering an issue where both color and depth streams occasionally freeze and then resume, resulting in:- 
- Frame loss
- Irregular playback speed, where frames seem to "jump" to catch up after a freeze

This problem occurs both in Intel RealSense Viewer and when running my Python script in Anaconda.

This is the script I'm currently using: 

```
import os
os.environ["CUDA_VISIBLE_DEVICES"] = "0"  # Force NVIDIA GPU

import pyrealsense2 as rs
import numpy as np
import cv2
import torch

# ✅ Force PyTorch to use GPU
device = torch.device("cuda" if torch.cuda.is_available() else "cpu")
print(f"🔹 PyTorch using: {device}")

# ✅ Check OpenCV CUDA support
if "CUDA" in cv2.getBuildInformation():
    print("✅ OpenCV has CUDA support.")
else:
    print("⚠️ OpenCV is NOT using CUDA.")

# RealSense Configuration
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.color, 848, 480, rs.format.bgr8, 15)
config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 15)

# Start Streaming
profile = pipeline.start(config)

# Align depth to RGB
align = rs.align(rs.stream.color)

# Output directory
output_dir = "S4"
os.makedirs(output_dir, exist_ok=True)

frame_count = 0
print("Recording frames... Press 'q' to stop.")

try:
    while True:
        frames = pipeline.wait_for_frames()
        aligned_frames = align.process(frames)

        color_frame = aligned_frames.get_color_frame()
        depth_frame = aligned_frames.get_depth_frame()

        if not color_frame or not depth_frame:
            continue

        # Convert frames to numpy arrays
        color_image = np.asanyarray(color_frame.get_data())
        depth_image = np.asanyarray(depth_frame.get_data())

        # Save RGB & Depth frames
        rgb_filename = os.path.join(output_dir, f"rgb_frame_{frame_count:04d}.png")
        depth_npy_filename = os.path.join(output_dir, f"depth_frame_{frame_count:04d}.npy")

        cv2.imwrite(rgb_filename, color_image)
        np.save(depth_npy_filename, depth_image)  # Save aligned depth map

        # Display frames
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        cv2.imshow('RGB Video', color_image)
        cv2.imshow('Depth Video', depth_colormap)

        frame_count += 1

        # Stop recording with 'q'
        if cv2.waitKey(1) & 0xFF == ord('q'):
            break

finally:
    pipeline.stop()
    cv2.destroyAllWindows()

print(f"Recording complete. Frames saved in: {output_dir}")

```

I already tried:
- Lowering the frame rate from 30 to 15 hz → Slight improvement, but the issue persists.
- Forcing GPU computation → Improved performance, but the freezes still happen.
- Checked related issues, especially #13730, which suggests increasing the Frame Queue Size.

Before applying Increase Frame Queue Size, I would like to check if there is a more suitable solution for my specific issue :)

**Hardware & Software Details**
Camera Model: RealSense D415 (single camera)
SDK Version: v2.56.3
Firmware Version: 5.16.0.1
Python Version: 3.11.11
Operating System: Windows 11 24H2 (Build 26100.2894)
Processor: Intel Core i5-12450H (12th Gen, 8 cores, 12 threads)
USB: 3.0

MartyG-RealSense, 2025-02-10:
Hi @egleorl1  You can further reduce the processing burden on the CPU when using rs-align to perform alignment by installing librealsense with CUDA support enabled.  This can be done by compiling librealsense from source code with CMake and including the flag **-DBUILD_WITH_CUDA=ON** in the CMake build command.

When librealsense's CUDA support is enabled, the work of processing depth-color alignment will be offloaded from the CPU onto the Nvidia GPU, which can result in a considerable reduction in CPU usage percentage.  However, the CUDA support is almost always used with Nvidia Jetson computing boards with an integrated Nvidia GPU, so using it on a Windows PC with an Nvidia GPU may have less predictable results.

*********

As you are using a USB cable that is 2 meters in length, it is important that it is a high quality ('premium grade') cable, as performance can reduce and the risk of camera connection drops increase as cable length increases unless the cable is a high quality one designed for data transfer, and not an inexpensive one meant only for device charging or syncing.

**********

A code-based method that can improve performance is to use the SDK's **Keep()** instruction to store frames in memory instead of writing them to the computer's storage.  It requires a computer with a large amount of memory capacity though, as the Keep() method progressively consumes memory over time as it is filled with camera frames.  On a computer with a good amount of memory, there may be capacity to hold around 30 seconds worth of frames unless some frames are periodically released to free up memory,  

https://github.com/IntelRealSense/librealsense/issues/6146 has a simple example of using Keep() in pyrealsense2.

egleorl1, 2025-02-13:
@MartyG-RealSense Thanks for your answer!

Regarding CUDA:
I have re-installed RealSense with CUDA support, following these steps:

1) Cloned https://github.com/IntelRealSense/librealsense.git and built librealsense with the following command:
```
cmake .. -DBUILD_PYTHON_BINDINGS=ON -DBUILD_WITH_CUDA=ON -DCMAKE_BUILD_TYPE=Release
cmake --build . --config Release --parallel
```
2) Copied the necessary files (pyrealsense2.pyd and realsense2.dll) from librealsense\build\Release to my Anaconda environment, where I run the RealSense frame acquisition script

However, I am not entirely sure if RealSense is actually using CUDA for depth-color alignment. I monitored system usage while running the script, and here are the performance metrics I observed in Task Manager (attached screenshot). Could you confirm if this indicates proper CUDA usage?

[image]

Regarding the USB Cable Details:
For my setup, I am using a USB Type-C to USB Type-A cable (USB 3.0 or higher). The cable is approximately 2 meters long, and I have ensured it is connected directly to a USB 3.0 port

Regarding keep():
I don't think I can use Keep() since it seems to work (if I understood correctly) only for short-term storage (around 30 seconds), while I need to record for about 5 minutes.
To reduce computational load, I implemented a resolution reduction + upscaling technique to maintain the expected image size while decreasing the processing burden:

```
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.color, 640, 360, rs.format.bgr8, 15)  # Lower resolution to reduce processing load
config.enable_stream(rs.stream.depth, 640, 360, rs.format.z16, 15)

# ✅ Start streaming
profile = pipeline.start(config)

# ✅ Increase the frame queue size to prevent freezes and frame loss
pipeline.get_active_profile().get_device().first_depth_sensor().set_option(rs.option.frames_queue_size, 32)

# ✅ Align depth to RGB
align = rs.align(rs.stream.color)

# ✅ Output directory for saving frames
output_dir = "S4"
os.makedirs(output_dir, exist_ok=True)

frame_count = 0
print("🎥 Recording frames... Press 'q' to stop.")

try:
    while True:
        frames = pipeline.wait_for_frames()
        aligned_frames = align.process(frames)

        color_frame = aligned_frames.get_color_frame()
        depth_frame = aligned_frames.get_depth_frame()

        if not color_frame or not depth_frame:
            continue

        # ✅ Convert frames to numpy arrays
        color_image = np.asanyarray(color_frame.get_data())
        depth_image = np.asanyarray(depth_frame.get_data())

        # ✅ UPSCALING with OpenCV CUDA (if available)
        target_size = (848, 480)  # Maintain final YOLO input size

        if use_cuda:
            gpu_color = cv2.cuda_GpuMat()
            gpu_depth = cv2.cuda_GpuMat()

            gpu_color.upload(color_image)
            gpu_depth.upload(depth_image)

            color_image = cv2.cuda.resize(gpu_color, target_size, interpolation=cv2.INTER_LINEAR).download()
            depth_image = cv2.cuda.resize(gpu_depth, target_size, interpolation=cv2.INTER_NEAREST).download()
        else:
            color_image = cv2.resize(color_image, target_size, interpolation=cv2.INTER_LINEAR)
            depth_image = cv2.resize(depth_image, target_size, interpolation=cv2.INTER_NEAREST)

        # ✅ SAVE RGB & Depth Frames
        rgb_filename = os.path.join(output_dir, f"rgb_frame_{frame_count:04d}.png")
        depth_npy_filename = os.path.join(output_dir, f"depth_frame_{frame_count:04d}.npy")

        cv2.imwrite(rgb_filename, color_image)
        np.save(depth_npy_filename, depth_image)  # Save aligned depth map

        # ✅ Display images
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        cv2.imshow('RGB Video', color_image)
        cv2.imshow('Depth Video', depth_colormap)

        frame_count += 1

        # Stop recording with 'q'
        if cv2.waitKey(1) & 0xFF == ord('q'):
            break

finally:
    # Stop streaming and release resources
    pipeline.stop()
    cv2.destroyAllWindows()

print(f"✅ Recording complete. Frames saved in: {output_dir}")
```

My main concern is whether this compromises RGB-Depth alignment, as maintaining alignment is crucial for my application. Would this approach introduce any misalignment issues?
Are there any other potential optimizations to reduce CPU load and/or frame freezing?

Thanks :)

MartyG-RealSense, 2025-02-14:
When using the RealSense Viewer, depth and RGB color are not aligned, and can only be mapped together in the Viewer's 3D pointcloud mode.  This pointcloud alignment uses a different process from align_to and so would not benefit from librealsense's CUDA acceleration.  However, the librealsense CUDA support does also accelerate pointclouds, so there could still be a speed benefit from CUDA when in the Viewer's 3D pointcloud mode even if acceleration is not being applied to alignment.

When RGB is not aligned, it is offset horizontally from the depth image due to the physical position of the RGB sensor on the front of the camera.

The D415 camera supports an RGB mode called 'color from left infrared sensor' where you can obtain an RGB image from the left infrared sensor instead of the RGB sensor.  The RGB image from the left-side infrared sensor is by default aligned perfectly with the depth image without having to perform depth-color alignment, because the depth image also originates from the left sensor.  Information about this mode can be found at the link below.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#use-the-left-color-camera

Depth-color alignment is a processing-intensive operation, so using the RGB from infrared image and not performing alignment could significantly reduce processing demands.

To obtain the RGB from left infrared sensor image, you can just change your cfg color stream instruction from rs.stream.color to rs.stream.infrared

`config.enable_stream(rs.stream.infrared, 640, 360, rs.format.bgr8, 15)`

************

Writing data to file formats such as .png and .npy will also be processing-inefficient.  The most efficient form of data writing in the librealsense SDK is the **.bag** file format.

**********

Depth and RGB frames should sync automatically when the depth and RGB FPS values are the same.  You can force the two streams to try to maintain the same FPS and so increase the chances of the frame auto-syncing occurring by disabling an RGB function called **auto-exposure priority**.

https://github.com/IntelRealSense/librealsense/issues/11246 provides a pyrealsense2 example of defining "depth_sensor" and "color_sensor" parameters with `depth_sensor, color_sensor, *_ = profile.get_device().query_sensors()` and then you can set auto_exposure priority to '0' (disabled):

`color_sensor.set_option(rs.option.auto_exposure_priority, 0)`

egleorl1, 2025-02-14:
Thanks a lot, that was really helpful!

So, if I understood correctly, aside from the 3D point cloud mode (which I don’t think fits my needs), the only alternative to obtain aligned depth and RGB images without the computational cost of rs.align() is to use the IR camera instead of the RGB sensor, right?

I’ve tested this solution, and it seems to work well. However, I’m not sure whether the RGB images I get from the IR sensor are optimal for my machine learning model for pose estimation.

If for this reason I decide to keep using the standard RGB sensor, would the solution I previously proposed work fine? That is:

1. Recording from the Viewer in .bag format, making sure that:
- The RGB and stereo cameras share the same FPS.
- The Auto Exposure Priority option is disabled in RGB.
2. Aligning depth to RGB in post-processing using rs.align().
3. Extracting the RGB frames as .png and the depth information as .npy, since this is the most compatible format for my machine learning pipeline.
I don’t really mind the computational load during post-processing, since my main concern was avoiding frame drops during recording. If alignment is done after recording, this shouldn’t be a problem.

Would you say this is a good approach? Let me know if you see any potential issues with it!

Thanks again for your support!

MartyG-RealSense, 2025-02-14:
A .bag file, once imported and played back, is treated as though it is a live camera.  So yes, once the bag is accessed then you can apply align_to in real-time to the depth and color stream data of the bag and then extract the frames as .png and .npy as your script currently does with live-camera data.

So your plan sounds like a very reasonable and achievable one.