realsenseai/librealsense #13759: D455 is not detected by Windows and Ubuntu

manic1841, 2025-02-13:
Required Info |  
-- | --
Camera Model | D455
OS | Windows, Ubuntu20.04

### Issue Description
Hello,
I was working with D455 few months ago, and everything was working fine. After that, I packed it up and stored it away. This week, I tried to connect it again to my Ubuntu device, but it wasn't detected. I also tried it on my Windows PC, but no luck there either.

When I plug in the USB. nothing happens. However, the D455 does start to warm up after a while, so it seems like it's getting power, but there's no connection signal.

Is there anything I can do to fix this?

Thanck you so much.

MartyG-RealSense, 2025-02-13:
Hi @manic1841  There is a rarely occurring phenomenon where if a RealSense camera is stored for weeks or months in certain kinds of enclosed locations such as a cupboard, suitcase or drawer then it may not function correctly when it is used for the first time after being taken out of storage.

There is a small chance that the camera might recover itself it is warmed repeatedly over a period of time (such as the powering from the USB that you describe).  In the majority of cases it is a permanent failure of the hardware though.

Can you provide information about the two questions below, please?

1.  Did you purchase your D455 within the previous 12 months?

2.  What country are you located in?

manic1841, 2025-02-14:
Hi @MartyG-RealSense 
Thanks for your response. It's unfortunate to hear that the issue might be due to hardware failure. I didn't store the camera in a closed place. It was kept on a shelf in the lab. 

I will keep trying to power on the camera and hope that it starts working.

Here are the answer to your questions:
1. No, I purchased the D455 more then a year ago.
2. I'm located in Taiwan.

MartyG-RealSense, 2025-02-14:
Thanks very much for the information.  I will discuss your case with my Intel RealSense colleagues.

MartyG-RealSense, 2025-03-03:
Hi @manic1841  Please send me an email at the address below so that I have your email address and we can discuss the next step in supporting your D455 camera problem.

martyx.grover@intel.com

MartyG-RealSense, 2025-03-08:
Hi @manic1841  As your issue was resolved over email, I will close the case.  Thanks again!

closed 2025-03-08 by MartyG-RealSense