realsenseai/librealsense #13760: RGBD Pointcloud not displayed on Rviz

shreeya0411, 2025-02-13:
Hi, I need some help with viewing my rgbd pointcloud on rviz. I input `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud` Next I go to rviz, add pointcloud2, update the topic to /camera/depth/color/points and for the global options I update the fixed frame to camera_link. I do not receive an image. 

Note that for rviz when I go to the fixed frame or the topic and click on the drop down I do not see any options to select. 

Also the rostopic list includes the /camera/depth/color/points however when I do rostopic echo /camera/depth/color/points I do not receive any data. Conversely when I input rostopic echo /camera/color/camera_info I do receive data. 

I am using a d405 camera 

I would really appreciate some guidance 

Thank you!

shreeya0411, 2025-02-13:
Just to add to my original comment. Following is the error I received when I input realsense-viewer 

[image]

MartyG-RealSense, 2025-02-13:
Hi @shreeya0411  D405 users of the RealSense ROS1 wrapper usually find that installing a "forked" version of the ROS1 2.3.2 wrapper at the link below that is customized to add D405 support instead of using the official 2.3.2 wrapper works well for them.

https://github.com/rjwb1/realsense-ros/tree/development

In regard to your realsense-viewer log, it looks as though you have librealsense installed in a ROS workspace directory.  Is this correct, please?  If you do have librealsense in the ROS workspace then it should be installed at a separate location outside of the workspace, otherwise problems can occur if librealsense and ROS data are mixed together in the same directory.

*********

Before changing your installation though, it would be worth first checking whether the alternative **RGBD** method of publishing a pointcloud in the RealSense ROS1 wrapper will work for you.

To use this method, support for an RGBD launch of the wrapper must be first installed manually using one of the instructions below, depending on which ROS1 branch you are using. 

**Kinetic**
`sudo apt-get install ros-kinetic-rgbd-launch`

**Melodic**
`sudo apt-get install ros-melodic-rgbd-launch`

**Noetic**
`sudo apt-get install ros-noetic-rgbd-launch`

Once RGBD support has been installed, use the **rs_rgbd.launch** launch file instead of the rs_camera.launch one.

`roslaunch realsense2_camera rs_rgbd.launch`

You do not need to define a pointcloud filter in the roslaunch instruction with this method.

rs_rgbd.launch should by default generate a point cloud that is published at **depth_image_proc**

shreeya0411, 2025-02-17:
Hi apologies for the late response. I reinstalled the forked version which seems to work. I am now able to view the pointcloud on rviz. As for the realsense-viewer issue you are correct that I had installed librealsense in the ros workspace which was causing the issue. Now I have undone this and realsense-viewer is streaming an image. Thank you for your helpful response.

closed 2025-02-17 by shreeya0411

MartyG-RealSense, 2025-02-17:
You are very welcome, @shreeya0411 - I'm pleased that I could help.  Thanks very much for the update!