realsenseai/librealsense #13763: No stream match for pointcloud chosen texture Process - Color

lina11116666, 2025-02-14:
Hello, I run the d435i camera on the Yushu robot dog go2, the system is ros2 foxy, but the output point cloud has a problem No stream match for pointcloud chosen texture Process - Color. Below is the log output after running the startup file. I searched for many related issues, but none of them were solved, including 'enable_color': True, # Enable color image stream
'enable_depth': True, # Enable depth stream
'align_depth': True, # Align depth stream to color image stream
'enable_pointcloud': True, # Enable point cloud generation
, but the problem is still not solved. Can you please help me take a look and help me solve it?
unitree@ubuntu:~/Desktop$ [INFO] [launch]: All log files can be found below /home/unitree/.ros/log/1970-01-04-12-13-23-782548-ubuntu-3471
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [3473]
[realsense2_camera_node-1] [INFO] [0000274404.366620128] [camera.camera]: RealSense ROS v4.51.1
[realsense2_camera_node-1] [INFO] [0000274404.366886848] [camera.camera]: Built with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [0000274404.366973120] [camera.camera]: Running with LibRealSense v2.51.1
[realsense2_camera_node-1] [INFO] [0000274404.418165728] [camera.camera]: Device with serial number 141722071482 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [0000274404.418369568] [camera.camera]: Device with physical ID /sys/devices/platform/3610000.xhci/usb1/1-3/1-3.1/1-3.1.3/1-3.1.3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [0000274404.418411840] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [0000274404.418852576] [camera.camera]: Device with port number 1-3.1.3 was found.
[realsense2_camera_node-1] [INFO] [0000274404.418904352] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [0000274404.418931616] [camera.camera]: Device 141722071482 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [0000274404.418962656] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [0000274410.529353440] [camera.camera]: Device with serial number 141722071482 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [0000274410.529530208] [camera.camera]: Device with physical ID /sys/devices/platform/3610000.xhci/usb1/1-3/1-3.1/1-3.1.3/1-3.1.3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [0000274410.529566080] [camera.camera]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [0000274410.529975040] [camera.camera]: Device with port number 1-3.1.3 was found.
[realsense2_camera_node-1] [INFO] [0000274410.530011840] [camera.camera]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [0000274410.530038144] [camera.camera]: Device 141722071482 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1] [INFO] [0000274410.536049312] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [0000274410.537925760] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [0000274410.537993472] [camera.camera]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [0000274410.538027680] [camera.camera]: Device Serial No: 141722071482
[realsense2_camera_node-1] [INFO] [0000274410.538054144] [camera.camera]: Device physical port: /sys/devices/platform/3610000.xhci/usb1/1-3/1-3.1/1-3.1.3/1-3.1.3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [0000274410.538076800] [camera.camera]: Device FW version: 05.15.01.00
[realsense2_camera_node-1] [INFO] [0000274410.538101536] [camera.camera]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [0000274410.538184800] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [ERROR] [0000274410.657436192] [camera.camera]: An error has occurred while calling sensor for: depth_module.auto_exposure_limit:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[realsense2_camera_node-1] [ERROR] [0000274410.659505184] [camera.camera]: An error has occurred while calling sensor for: depth_module.auto_gain_limit:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[realsense2_camera_node-1] [ERROR] [0000274410.661507584] [camera.camera]: An error has occurred while calling sensor for: depth_module.auto_exposure_limit_toggle:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[realsense2_camera_node-1] [ERROR] [0000274410.664372416] [camera.camera]: An error has occurred while calling sensor for: depth_module.auto_gain_limit_toggle:hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[realsense2_camera_node-1] [INFO] [0000274410.670925728] [camera.camera]: Set ROS param depth_module.profile to default: 640x480x15
[realsense2_camera_node-1] [WARN] [0000274410.738842848] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [0000274410.744596032] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x15
[realsense2_camera_node-1] [INFO] [0000274410.747061280] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [0000274410.748195808] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [0000274410.782606720] [camera.camera]: Stopping Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [0000274410.790873920] [camera.camera]: Starting Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [0000274410.872723008] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [0000274410.873078304] [camera.camera]: Stopping Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [0000274410.877562368] [camera.camera]: Starting Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [0000274410.891225696] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 15
[realsense2_camera_node-1] [INFO] [0000274410.892381056] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [0000274411.046916352] [camera.camera]: 
[realsense2_camera_node-1] [WARN] [0000274416.766247808] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1] [WARN] [0000274517.280838176] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1] [WARN] [0000274593.454987616] [camera.camera]: No stream match for pointcloud chosen texture Process - Color
[realsense2_camera_node-1] [WARN] [0000274605.210782688] [camera.camera]: No stream match for pointcloud chosen texture Process - Color

MartyG-RealSense, 2025-02-14:
Hi @lina11116666  The warning `No stream match for pointcloud chosen texture Process - Color` indicates that some of the color frames are missing, perhaps due to dropped frames.

If you do not need the aligned topics to be published then you could set align_depth to false, because if the pointcloud is enabled and depth and color are also enabled. depth and color will be mapped together automatically to generate a 'textured pointcloud' without having to have align_depth enabled.

I note that the USB connection speed is being detected as slow 2.1 instead of fast 3.2.  If you have the camera inserted into a USB 3 port on the computer, does it make a difference if you unplug the micro-sized end of the USB cable from the side of the camera, turn the connector around the other way and re-insert it into the camera (USB-C cables are two-way insertion at the micro-sized end).

Are you using the official 1 meter USB cable supplied with the camera or a cable of your own choice?  Sometimes when using one's own cable, its performance may be negatively affected by being low quality, or an inexpensive cable designed only for device charging rather than data transfer, or be a USB 2.1 cable that will limit the performance of a USB 3 device attached to it to USB 2.1 speeds.

lina11116666, 2025-02-15:
Thanks for your comment.
1. Whether align_depth is set to true or false, frame loss will occur.
2. Because the cable connecting the camera and the robot dog unitree go2 is a dual type-c cable provided by unitree, it may not be replaced.
3. After searching online, I modified these lines of code: #{'name': 'depth_module.profile', 'default': '0,0,0', 'description': 'depth module profile'}, to {'name': 'depth_module.profile', 'default': '640,480,15', 'description': 'depth module profile'}; #{'name': 'rgb_camera.profile', 'default': '0,0,0', 'description': 'color image width'}, to {'name': 'rgb_camera.profile', 'default': '640,480,15', 'description': 'color image width'}; #{'name': 'pointcloud.stream_filter', 'default': '2', 'description': 'texture stream for pointcloud'}, changed to {'name': 'pointcloud.stream_filter', 'default': 'RS2_STREAM_COLOR', 'description': 'texture stream for pointcloud'},
Then the frame loss is reduced, but there is still drift, that is, when returning to the origin again, it will not overlap. Is there any way to reduce this drift? Or should I consult elsewhere.

MartyG-RealSense, 2025-02-15:
If a RealSense camera is built into a product such as a robot dog then the manufacturer should be contacted for support, as companies producing RealSense-equipped products not manufactured by Intel are responsible for providing technical support of that product.

Can you confirm please what you mean by drift when returning to the origin and there not being overlap.  Thanks!

lina11116666, 2025-02-15:
Thank you for your reply. The main reason for zero drift does not seem to be caused by the camera. The frame loss phenomenon has been significantly reduced. Thank you for your help.

MartyG-RealSense, 2025-02-15:
You are very welcome!  :)