realsenseai/librealsense #13764: Realsense 435i connection issue to NVIDIA Jetson Orin Nano with Jetpack 6.0

mohandeepkapur, 2025-02-14:
Camera Model: D435i
SDK Version: 2.56
Platform: Jetson Orin Nano, JetPack 6.0
OS Version: 22.04.5 LTS
Kernel: 5.15.136-tegra

Hi, I've been following the Realsense SDK installation instructions for my Jetson Orin Nano from https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md. I opted for the RSUSB backend as it suits my use-case, so after installing the librealsense2 dev and utils packages I ran libuvc_installation.sh (with the default flag -DFORCE_LIBUCV=true being modified to -DFORCE_RSUSB_BACKEND=ON as recommended by the installation docs for SDKs >v2.30).

However, after the build finishes, my Jetson Orin Nano detects but still cannot connect to the Realsense (running rs-enumerate-devices and realsense-viewer outputs no realsense connected). I've attached a text file containing select pieces of the build script's log that indicate failed tests and missing header files.

Any help with debugging the issue would be appreciated.

(And just to confirm, there's no need to re-build the Jetson's kernel to enable HID support if using RSUSB backend?)

[libuvc_script_select_logs.txt](https://github.com/user-attachments/files/18803887/libuvc_script_select_logs.txt)

MartyG-RealSense, 2025-02-15:
Hi @mohandeepkapur  A couple of RealSense users with JetPack 6 who built librealsense with the RSUSB backend method and still had problems found that a solution was to downgrade the camera's firmware driver to an older version.  I recommend trying a downgrade to firmware version **5.13.0.50**

There is no need to apply a kernel patch script when using the RSUSB backend.

mohandeepkapur, 2025-02-18:
Hi Marty,

Thanks for your response! I had actually made a mistake and installed a copy of the SDK through sudo apt before building from source, and the executables my computer was using were of the former which lacked RSUSB support.

I do have another question, and it has to do with the ros-realsense wrapper. When running `ros2 launch realsense2_camera rs_launch.py`, I receive a stream of:

`[realsense2_camera_node-1]  18/02 10:30:11,588 WARNING [281472342550752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`.

I also built the ros2 wrapper from source and skipped installing librealsense2 again as a dependency, so it shouldn't be a "using incorrect SDK" issue.

Any help with debugging the issue would be appreciated.

MartyG-RealSense, 2025-02-18:
If the ROS2 topics are displaying correctly in RViz and at the speed that they are supposed to then it is safe to ignore the **control_transfer returned** warnings in the log.

The 'Resource temporarily unavailable' form of this warning can indicate that there is an issue related to the USB system (the USB port or the cable).  It could also indicate that the Jetson's CPU is having difficulty keeping up with processing new camera frames at the rate that they are arriving from the camera at the Jetson's USB port.

Reducing the resolution or FPS speed can reduce the number of warnings being generated in the log, though it is unlikely that they will be eliminated completely.  If you are unable to reduce resolution or FPS thought then just ignore the messages if the topics are performing correctly.

mohandeepkapur, 2025-02-21:
It was indeed a non-issue. Thanks!

closed 2025-02-21 by mohandeepkapur