realsenseai/librealsense #13768: Previously stable realsense SDK failed to find realsense device now

AMLIFY, 2025-02-17:
Camera Model: D435i&D455
SDK Version: 2.55.1
Platform: Jetson Orin Nano, JetPack 6.1
OS Version: 22.04.5 LTS
Kernel: 5.15.148-tegra

I have been successfully running realsense cameras since Dec. 2024. However, when I launch realsense viewer or its ros2 SDK, the output was like"No realSense devices were found!","Please connect a Realsense device!". I was not sure the reason why my two cameras could't work properly now, maybe it had something to do with the Software Upgrader? Because I always agreed to update when the Software Upgrader popped up, and maybe the latest version of certain libraries realsense SDK relied on no longer supports it? Looking forward to your suggestions.

MartyG-RealSense, 2025-02-17:
Hi @AMLIFY   If you have not changed your librealsense installation then a likely cause of a camera not being detected after a software upgrade on your Jetson may be if your JetPack version has been updated.

RealSense cameras equipped with an IMU can have problems with being detected on Jetsons with JetPack 6.  This is because JP6 removed a function called **hiddraw** that RealSense IMU cameras - known as HID devices - depended upon.

Do you know whether your JetPack version has changed since December 2024, please?

AMLIFY, 2025-02-18:
Hi @MartyG-RealSense I checked the jetpack version and it has been 6.1 since I started to use it.

MartyG-RealSense, 2025-02-18:
Was the Software Upgrader that you refer to a pop-up box in realsense-viewer?

If it was in realsense-viewer, do you recall what the pop-up message said?  For example, if it was for a camera firmware version update.

AMLIFY, 2025-02-18:
No, it's the software of Ubuntu.

MartyG-RealSense, 2025-02-18:
Thank you.  Is your Ubuntu kernel version still the same?

AMLIFY, 2025-02-18:
Yes, and I reran the script libuvc_installation.sh, but it didn't make a difference.

MartyG-RealSense, 2025-02-18:
If you have installed librealsense using the libuvc_installation.sh script then librealsense should not be dependent on a particular Ubuntu version or kernel version (as a libuvc build of the SDK bypasses the kernel), so changes to Ubuntu should not have a significant effect on librealsense.

Do you have the option to wipe the storage of your Jetson and install everything again from the beginning, including Ubuntu 22.04?  Sometimes doing so will cause an unsolvable problem to suddenly be resolved, because wiping the storage removes a hidden problem that was preventing the camera from working.

AMLIFY, 2025-02-18:
Hi @MartyG-RealSense I found the package upgrade log, and I decided to check the most recent log before the breakdown of realsense SDK. Reinstalling everything would be the solution with the lowest priority because of the time cost. I would update 没有discovery ASAP.

MartyG-RealSense, 2025-02-18:
Okay, thank you for the update.  I look forward to your next report.

AMLIFY, 2025-02-20:
Hi @MartyG-RealSense I guessed I found the problem. When I installed librealsense for the first time, the same problem occurred.  I solved the problem by uninstalling librealsense and running the libuvc_installation.sh. so the camera could work. I installed rtabmap recently due to my project. However, it had a dependency on librealsense, so librealsense was installed automatically. I could't uninstall librealsense now because of the dependency of rtabmap, so I'm wondering if there is a solution?

AMLIFY, 2025-02-20:
I just uninstalled rtabmap and librealsense, then my camera can be detected by realsense viewer now. I'm gonna try installing rtabmap from the source code.

MartyG-RealSense, 2025-02-20:
I'm pleased to hear that you made significant progress.  Please do keep me updated about your tests.  Good luck!

AMLIFY, 2025-02-22:
I built rtabmap from the source code and solved several cmake errors. Still librealsense were installed automatically, but this time I could uninstall it without an influence on rtabmap dependencies. I guessed the issued was resolved, and thanks for your reply!

MartyG-RealSense, 2025-02-22:
It's excellent to hear that your issue has been resolved.  Thanks very much for the update.  As the problem is resolved, I will close this issue.  Thanks again!

closed 2025-02-22 by MartyG-RealSense