realsenseai/librealsense #13771: Realsense Viewer error

yijhenHuang, 2025-02-17:
| | |
|---|---|
|**librealsense**|2.56.3 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:572

```
I use D455f intel realsense camera. please help., i only change the font size and font and clicked "apply" button then it shows this.

MartyG-RealSense, 2025-02-17:
Hi @yijhenHuang  If you have Windows 11 then please try using the instructions at https://github.com/IntelRealSense/librealsense/issues/13457#issuecomment-2432900837 to delete the **realsense-config.json** configuration file to reset the Viewer's configuration to its default settings.

yijhenHuang, 2025-02-17:
hi, i have followed the suggestion you have given me, but i can't find realsense-config.json in roaming 

[image]

yijhenHuang, 2025-02-17:
i forgot to say that i delete the my  Intel RealSense SDK 2.0 folder , then i download  again [Intel.RealSense.SDK-WIN10-2.56.3.7838.beta.exe](https://github.com/IntelRealSense/librealsense/releases/download/v2.56.3/Intel.RealSense.SDK-WIN10-2.56.3.7838.beta.exe) 
and 
[Intel.RealSense.Viewer.beta.exe](https://github.com/IntelRealSense/librealsense/releases/download/v2.56.3/Intel.RealSense.Viewer.beta.exe)

in https://github.com/IntelRealSense/librealsense/releases/tag/v2.56.3

[image]

MartyG-RealSense, 2025-02-28:
Hi @yijhenHuang  Do you require further assistance with this case, please?  Thanks!

yijhenHuang, 2025-02-28:
Hi Marty, your previous answer helped me, thank you so much. 
The file was duplicated in another folder haha that’s why i had a hard time finding it.

MartyG-RealSense, 2025-02-28:
It's excellent to hear that your issue is resolved.  Thanks very much for the update!

closed 2025-03-08 by MartyG-RealSense