realsenseai/librealsense #13775: How to get a surface normal image aligned with color and depth images?

Boatsure, 2025-02-19:
I want to get an aligned 3-channel normal map while getting the aligned color and depth images. Each pixel position on this normal map stores the xyz normal direction of the object surface relative to the camera position.

Is there an officially recommended way to achieve this in the current version and time point? Better in python, thank you！

MartyG-RealSense, 2025-02-19:
Hi @Boatsure  A RealSense user at https://github.com/IntelRealSense/librealsense/issues/3321 shared Python code for the method they used to obtain the surface normal of a depth image.

I was not able to find an example of a script that used both surface normals and depth-color alignment though.

Boatsure, 2025-02-20:
@MartyG-RealSense Thanks for your reply！The script you mentioned is a kind of numerical calculation method based entirely on depth image, and its accuracy is worrying. 

The viewer can export .ply file in the 3D mode, which has an option of normals output. May I ask how do the normals calculated here?

MartyG-RealSense, 2025-02-20:
The **save_to_ply()** .ply file export function has a setting called **OPTION_PLY_NORMALS** that exports normals when set to true.  The code for controlling this function is in the librealsense SDK file 'rs_export.hpp'

https://github.com/IntelRealSense/librealsense/blob/e1688cc318457f7dd57abcdbedd3398062db3009/include/librealsense2/hpp/rs_export.hpp

If you do a web browser 'find' operation on this page for the word **normals** then it will highlight the sections of code related to normals calculation.

The easiest way to export a ply with normals included is to use the **Export** option in the 3D pointcloud mode of the RealSense Viewer tool.  You can also use Python code though.  The SDK's **export_ply_example.py** ply export Python example program uses save_to_ply() and sets normals export to true.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py

Boatsure, 2025-02-26:
I get the .ply files from RealSense Viewer saving and Python code exporting with normals included. 

After I imported the .ply into MeshLab, I found that indeed the model showed dense normal rays. 
But how can I get the three-channel normal map at a specified camera view? I know that the MeshLab operation is beyond the scope of this repo, but I would be grateful if you could provide some guidance.

MartyG-RealSense, 2025-02-26:
What kind of results do you get in MeshLab if you select the option **Render > Shaders > Normalmap** to shade the view with a normalmap?

The link below describes how the shader can be customized if the image that its default settings produce are not suitable for oyur needs.

https://stackoverflow.com/questions/45869953/meshlab-normalmap/46067635#46067635

Boatsure, 2025-02-27:
Thanks for your advice, while

Render > Show Normal gives:

[image]

Render > Shaders > Normalmap gives:

[image]

I would find another way and closeing it now.

closed 2025-02-27 by Boatsure