realsenseai/librealsense #13778: missing rgb camera

cpremazzi, 2025-02-21:
| | |
|---|---|
|**librealsense**|2.56.3 RELEASE|
|**OS**|Windows|

Please provide a description of the problem

i have only the stereo module but not have the rgb camera

MartyG-RealSense, 2025-02-21:
Hi @cpremazzi  Which RealSense camera model are you using, please?

cpremazzi, 2025-02-22:
D435

MartyG-RealSense, 2025-02-22:
In the RealSense Viewer tool, is D435 the name of the camera that is listed at the top of the options side-panel?  Or is it a different name such as D430

MartyG-RealSense, 2025-03-08:
Case closed due to no further comments received.

cpremazzi, 2025-03-08:
could you please open again?  is D435
j72-479-100

Carlotta Premazzi Design Studio
Art and Design
www.cpdesignstudio.com
***@***.***

GLITCH&FLOWER Live Visual
//FacebookPage <https://www.facebook.com/GlitchFlower/>
***@***.***

à(s) 17:54):

MartyG-RealSense, 2025-03-09:
If the camera name in the RealSense Viewer is D435 then that means that the RGB sensor in the camera is being detected.  If RGB was not detected then the camera name would be D430.

I tested the 2.56.3  *Intel.RealSense.Viewer.beta.exe** version of the Viewer on Windows with a D435 and did not experience the same problem with the RGB module.  Have you downloaded and run this Viewer program file to test the RGB in it?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.56.3

[image]

closed 2025-03-22 by MartyG-RealSense