realsenseai/librealsense #13779: Python/Windows or just do an ubuntu partition or go back to c++?

arothenberg2, 2025-02-22:
I''ve programmed to the realsense sdk using c++ in visual studio using cmake and linking dlls libraries etc... in VS.
Now I need to set up an environment for python since my collaborator uses python and I'm having a heck of a time in windows. There are several methods I've browsed and none have just worked. 

Before I get deeper into setting up my librealsense/python/windows environment - is this a bad idea? should I take the long way round and make an ubuntu partition? Are either method worth it or is realsense just a lot easier with c++ and no wrapper?

MartyG-RealSense, 2025-02-22:
Hi @arothenberg2  At https://github.com/IntelRealSense/librealsense/issues/6402#issuecomment-629763906 I wrote a guide to performing a **pip install** of the pyrealsense2 wrapper on Windows from packages that may be a useful reference if you have not seen it already.

This Windows installation method supports Python 3.7 to 3.11 at the time of writing this.  If you use a Python 3 version newer than 3.11 then you will need to build the wrapper from source code.

On Windows you can build the librealsense SDK from source code using instructions at the link below, and build the pyrealsense2 wrapper at the same time as librealsense if you set the build flag **BUILD_PYTHON_BINDINGS** to True.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

arothenberg2, 2025-02-22:
No, i did not see your instructions. I'll try this and report back the results and close out if it works.
Thanks!

arothenberg2, 2025-02-22:
OK, thank you. I'll close soon. The sample code from the python document runs. 
is there python rewrite of the Example programs ??- like align-advanced etc... those programs were very useful when getting started with realsense using c++. They are like templates.
Has someone done that?

closed 2025-02-23 by arothenberg2

MartyG-RealSense, 2025-02-23:
Most of the C++ examples do not have official Python equivalents, though it is possible to find scripts created by RealSense users.  A good way to find them is to perform a web search beginning with the term **librealsense python** and then a word relating to what you are searching for, such as `librealsense python alignment`

The list of official Python examples is here:
 
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples

A RealSense user produced a Python conversion of the C++ **rs-measure** example for measuring the distance between two coordinates on an image.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

arothenberg2, 2025-02-23:
Thanks. I suppose a lot of the advantage of your c++ examples is you already link into those hard to use graphics libraries so it makes rendering video in different ways a lot easier.