realsenseai/librealsense #13784: IMU on 435i not working on Jetson with Jetpack 6

wbjorndahl, 2025-02-23:
The 435i camera and depth functionality work great with Jetpack 6 on my Nvidia Jetson Orin Nano Dev kit. This is with FW version 5.16.0.1 and SW version 2.55.1.0. I am not using ROS.

To test the IMU on my Jetson, I run `rs-motion`. The object remains stationary as I move the physical camera around. 

I then connect the 435i to my desktop running Ubuntu and run `rs-motion`. The camera on the screen moves as expected as the physical camera is moved.
 
Does anyone know of a fix?

MartyG-RealSense, 2025-02-24:
Hi @wbjorndahl  RealSense camera models equipped with an IMU, such as D435i, can have problems with JetPack 6 because JP6 removed a function called **hiddraw** that IMU-equipped cameras (known as HID devices) relied upon. 

The range of available solutions includes:

1.  Using JetPack 5.1.3 instead of JP6.  JP5 is designed for Ubuntu 20.04 though and so it may not be possible to downgrade the JP if you are using Ubuntu 22.04 or 24.04.

2.  Building the librealsense SDK from source code with the **libuvc backend** installation method at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The libuvc backend installation method provides a pre-made build script called **libuvc_installation.sh**

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

For some RealSense users, using the libuvc method does not resolve the IMU problem and they have to additionally downgrade their camera firmware driver to an older version.

3.  Installing the RealSense **MIPI driver** to add support for IMU when using a camera with JetPack 6 on a USB cable connection.  This method is complex to set up though.

https://github.com/IntelRealSense/realsense_mipi_platform_driver

********

If you do not need the IMU functions on Jetson and are satisfied with using the IMU only on your Ubuntu desktop then the easiest approach would be to do nothing.

wbjorndahl, 2025-02-24:
Hi @MartyG-RealSense 

Thank you for the quick response. I was able to get RGB, Stereo, and IMU all working by:

1. Building the SDK through the libuvc backend installation method you mentioned
2. Downgrading FW to 5.13.0.55 (following a comment you made here: [https://github.com/IntelRealSense/librealsense/issues/13764#issuecomment-2660851249])

closed 2025-02-24 by wbjorndahl