realsenseai/librealsense #13791: I connected the D435 camera with 3.0 cable, but it shows 2.1

ky047, 2025-02-25:
To check the D435 camera, I'm going to use the USB 3.0 cable
I connected it
But the information says 2.1 cable
-Is there no problem with continuous use of 2.1 cables?
-Tell me why you don't recognize the 3.0 cable
-Does it support USB 2.0?

[image]
[image]

MartyG-RealSense, 2025-02-25:
Hi @ky047  Yes, RealSense cameras do fully support USB2, though the number of supported stream resolution and FPS speed combinations is limited compared to USB3.

Whilst the cameras can use USB 3.0, a **USB 3.1 Gen 1** connection is recommended.  The information below is from the official RealSense Store listing for the D435.

[image]

Akumar201, 2025-02-27:
Hi @ky047 , I encountered the same issue with my D435. In my case, the connection port was loose, and once I tightened it, the camera correctly displayed USB 3.1. Could you let me know which cable you're using? Please try using [this cable](https://newnex.com/realsense-3d-camera-connectivity.php#:~:text=Popular%20models%20like%20the%20RealSense,adopt%20the%20Micro%20B%20socket.). Also please check the port on which you have connected the camera and if it has enough bandwidth.

closed 2025-03-15 by MartyG-RealSense