realsenseai/librealsense #13792: jeston orin nx unable to use three D435 cameras at the same time

zw-12, 2025-02-26:
When I set the image resolution of RGB and depth to 1280×720, I can only use two D435 cameras on the jeston orin nx device, and cannot get images when using three D435 cameras. 
The version of librealsense installed is 2.54.2, and the version of jetpack is 5.1.1.

MartyG-RealSense, 2025-02-26:
Hi @zw-12  Are the three cameras plugged directly into the USB ports of the Jetson or are they attached to a USB hub?

If they are plugged directly into the Jetson then it may be providing insufficient power to the cameras to stream depth and RGB on three cameras at 1280x720, presumably at 30 FPS.  You could test whether reducing the FPS speed to 15 or 6 enables images from all three cameras to be provided.

If the cameras are attached to a USB hub then the maximum allowable data transfer 'bandwidth' may be being exceeded.  A hub can typically handled 5 Gigabits per second (Gbps) of data, and as resolution and FPS is increased then more bandwidth is consumed, bringing the total bandwidth of all attached cameras closer to the 5 Gbps limit.  Again, you could test whether reducing the FPS speed resolves the issue.  If it is a bandwidth limit issue then there are USB hubs available with a higher bandwidth limit of 10 Gbps.

zw-12, 2025-02-27:
> Hi [@zw-12](https://github.com/zw-12) Are the three cameras plugged directly into the USB ports of the Jetson or are they attached to a USB hub?
> 
> If they are plugged directly into the Jetson then it may be providing insufficient power to the cameras to stream depth and RGB on three cameras at 1280x720, presumably at 30 FPS. You could test whether reducing the FPS speed to 15 or 6 enables images from all three cameras to be provided.
> 
> If the cameras are attached to a USB hub then the maximum allowable data transfer 'bandwidth' may be being exceeded. A hub can typically handled 5 Gigabits per second (Gbps) of data, and as resolution and FPS are increased then more bandwidth is consumed, bringing the total bandwidth of all attached cameras closer to the 5 Gbps limit. Again, you could test whether reducing the FPS speed resolves the issue.
> 
> If it is a bandwidth limit issue then there are USB hubs available with a higher bandwidth limit of 10 Gbps. Alternatively, cameras can be spread across more than one hub, each hub having its own 5 Gbps bandwidth allowance.
> 
> If a hub is being used and it is not a mains electricity powered one then insufficient power could still be an issue, as the hub will be drawing power for the attached cameras from the Jetson's power supply in the same way that directly-attached cameras do. If a mains-plug powered hub is used then multiple-camera setups on a Jetson are usually more stable as they are powered independently by the mains electricity instead of drawing from the Jetson.

Hi, @MartyG-RealSense 
Thanks for your response. The three cameras are directly plugged into the USB port of Jeston，and  I have tested reducing the FPS speed to 6, but still can't get any images.

 When the image resolution is set to 640×480, the three cameras can work normally.

Probably not a problem of bandwidth limitation, since when trying to use two D435 cameras with resolution set to 1280x720 and one depth camera of another brand with resolution 2K simultaneously all cameras work fine.

The following is the information of the USB port

[image]

MartyG-RealSense, 2025-02-27:
About five years ago there was an issue with a small number of Jetson owners when using multiple RealSense 400 Series cameras, where one of the cameras in a set would not be detected or would not behave correctly (such as the 4th camera in a set of 5, or the 5th camera in a set of 6).  An example of this with a Jetson NX user is discussed at https://github.com/IntelRealSense/librealsense/issues/10660

In that user's particular case, simply resetting the Jetson resolved their issue.

zw-12, 2025-02-27:
Thanks for your help, I have successfully solved the problem.

MartyG-RealSense, 2025-02-27:
That's great to hear - thanks very much for the update!  :)

closed 2025-03-08 by MartyG-RealSense