realsenseai/librealsense #13793: vscode intelisense with pyrealsense

arothenberg2, 2025-02-26:
Intelisense doesn't work with the pyrealsense in my setup  -  windows vscode or vs 2022 (I played with both) . 
Is there a common fix for the problem? Pyrealsense runs fine. I know this may not be a Realsense issue. I tried various solutions, Maybe I am not providing the right path in the settings.json? Thanks in advance.

MartyG-RealSense, 2025-02-27:
Hi @arothenberg2  Problems with using Intellisense with pyrealsense2 code is a known issue.  Links to discussions about it and a possible solution can be found at https://github.com/IntelRealSense/librealsense/issues/11789#issuecomment-1542273245

arothenberg2, 2025-02-28:
The soution seems to be(?) scrapping the python docs and making a stub file from that - unless I'm reading it wrong. That's seems  like a lot of of work.
I'll just not use autocomplete for the pyrealsense.  It's disappointing but I think I can work around it.

MartyG-RealSense, 2025-03-01:
Okay, thanks very much for the update.

closed 2025-03-03 by arothenberg2