realsenseai/librealsense #13794: d415 can it capture the area i need

arothenberg2, 2025-02-26:
I am working with an older d435. I had 2 but one has started to fail - it shows noise  except in a cross pattern in the middle of the frame and it is out of warranty(sort of frustrating since it was well maintained) - so I am thinking of buying another camera and  as I read the d415 I realized it may be a better fit.
1. Is the d415 more 'accurate' than the d435. Also, Is it as fast - are they both 30fps? 
2. can it capture a  2 meter tall, 1.5 meter wide object in motion at between 1 and 2.5 meters directly in front of the camera ?

MartyG-RealSense, 2025-02-27:
Hi @arothenberg2  Thanks very much for your questions.

1.  D415 has around 2x less noise and 2x less error over distance than D435.  In terms of speed, both support 30 FPS.  D435 has a fast 'global' shutter on its depth sensor though, meaning that it can capture high-speed motion (the camera moving or an object in front of it moving) much better than the D415, whose image can lag when experiencing high speed. 
 
Both D415 and D435 have a slow 'rolling' shutter on their RGB sensor, so RGB performance should be very similar.  D415 also has the advantage of the depth and RGB sensors being mounted on the same sensor board, meaning that it is better able to synchronize between the depth and RGB frames.

2.  D415 has a smaller horizontal and vertical field of view size than D435, so the camera would have to be moved further away to fit a large object in its view.

Also bear in mind that a D415 has a larger minimum depth sensing distance, meaning that when measuring depth, an object is recommended to be located at a minimum of around 50 cm / 0.5 meters away from the camera, compared to 30 cm / 0.3 m for D435.

Overall, another D435 is likely to meet your needs instead of a D415.

arothenberg2, 2025-02-28:
1. Thank you. I need the capture high speed provided by the global shutter. Though less noise and more accuracy is nice. Oh well.
So d435 is the one. Intel doesn't repair, right?

closed 2025-02-28 by arothenberg2

MartyG-RealSense, 2025-02-28:
No, Intel does not offer a camera repair service, only return to the retailer that it was purchased from for refund or replacement during its warranty period.  

It is possible to buy the two circuit boards inside the D435 camera separately (the D430 Depth Module and the Vision Processor D4 V1) and open up the camera and attempt a repair yourself by replacing one or both boards if the camera is outside of its warranty.  The D4 V1 board costs around 45 to 50 dollars / euros and the depth module is the more expensive board of the two.    

It may be worth attempting resetting your D435 with the noise problem to its factory-new default calibration in the RealSense Viewer using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 to see if it improves the problem before you purchase another camera.