realsenseai/librealsense #13796: Option to build RealSense SDK without realsense-file?

RobbertProost, 2025-02-27:
I am currently using a statically build version of the realsense SDK and are encountering an issue with duplicate LZ4 symbols. If I understand correctly the realsense SDK contains the LZ4 library as part of the realsense-file target.
For my use case I don't need the realsense-file functionallity. 

So I was wondering, can I build the realsense SDK without realsense-file?

MartyG-RealSense, 2025-02-27:
Hi @RobbertProost  You could try building the SDK with CMake with symbols stripped out of the build using instructions at https://github.com/IntelRealSense/librealsense/issues/7982#issuecomment-743991477 to see whether removing the symbols resolves your issue.

Alternatively, you could try editing the **config-cmake** file for realsense-file in the SDK source code's `third-party` folder and then build the SDK from that modified source code.

https://github.com/IntelRealSense/librealsense/blob/master/third-party/realsense-file/config.cmake

MartyG-RealSense, 2025-03-08:
Hi @RobbertProost   Thanks very much for your PR.  I note that it was successfully merged into the SDK 4 days ago at the time of writing this.

Do you require further assistance with this case, please?  Thanks!

MartyG-RealSense, 2025-03-15:
Case closed due to PR merged and no further comments received.

closed 2025-03-15 by MartyG-RealSense