realsenseai/librealsense #13798: Intermittent Depth Data Issue with D457 Cameras

sinuku, 2025-02-28:
Hello,

I am currently using two D457 cameras to capture color and depth data simultaneously over USB, not GMSL.

While developing with the librealsense SDK, I noticed an intermittent issue where depth data is not received properly. Occasionally, the depth stream stops, and waving a hand in front of the camera seems to resolve the issue. This problem also occurs in RealSense Viewer and happens sporadically.

Since I am unable to attach a video, I have provided a link to the footage below.
https://youtu.be/lRLZ3exhQYM

I am using RealSense Viewer v2.56.2 and librealsense SDK v2.56.3.

The camera settings are as follows, and both cameras use the same configuration:
RGB: 848x480
Depth: 848x480
FPS: 30

Could you please investigate this issue?

Thank you.

sinuku, 2025-02-28:
Since the issue is intermittent and requires further testing, it appears that setting both the RGB and Depth resolutions to 1280x720 prevents the problem from occurring so far.

MartyG-RealSense, 2025-02-28:
Hi @sinuku  As the depth stream recovers if a hand is placed in front of it, it is possible that something has gone wrong with the exposure.  If the camera is covered by a hand and auto-exposure is enabled then this blocking of the scene that the camera is observing allows the auto-exposure to automatically correct the image.

Immediately before the depth image breaks up, there is a white flash on the infrared image of that camera (shown in the picture below).  This could be further evidence that some kind of disruptive exposure-related event has occurred.

[image]

sinuku, 2025-02-28:
> Hi [@sinuku](https://github.com/sinuku) As the depth stream recovers if a hand is placed in front of it, it is possible that something has gone wrong with the exposure. If the camera is covered by a hand and auto-exposure is enabled then this blocking of the scene that the camera is observing allows the auto-exposure to automatically correct the image.
> 
> Immediately before the depth image breaks up, there is a white flash on the infrared image of that camera (shown in the picture below). This could be further evidence that some kind of disruptive exposure-related event has occurred.
> 
> It is not clear why setting the resolution to 1280x720 instead of 848x480 seems to prevent it from occurring.
> 
> [image]

After continuing development with 1280x720, the issue started appearing after changing the resolution to 848x480.

It occurs approximately once every 4 to 5 times when starting the program (starting the pipeline).
Is there no specific solution for this issue? Has any other user experienced a similar problem before?
Is there a way to automatically enable the auto-exposure function when starting the pipeline?

MartyG-RealSense, 2025-02-28:
If you want to ensure that auto-exposure is enabled then you can do so with program code.

**C++** - https://github.com/IntelRealSense/librealsense/issues/3141#issuecomment-457361895

Set **RS2_OPTION_ENABLE_AUTO_EXPOSURE** to '1' to enable auto-exposure and '0' to disable it.

**Python** - https://github.com/IntelRealSense/librealsense/issues/3558

Set **RS.OPTION_ENABLE_AUTO_EXPOSURE** to '1' to enable auto-exposure and '0' to disable it.

***********

You could try disabling auto-exposure in the Viewer's options side-panel to confirm whether or not it is being caused by the exposure changing due to something in the room.  If the camera will be used in a location with a constant level of artificial lighting like this room then you may not need auto-exposure to be enabled anyway.

sinuku, 2025-03-04:
@MartyG-RealSense 
As the depth stream recovers if a hand is placed in front of it, it is possible that something has gone wrong with the exposure. If the camera is covered by a hand and auto-exposure is enabled then this blocking of the scene that the camera is observing allows the auto-exposure to automatically correct the image.

Immediately before the depth image breaks up, there is a white flash on the infrared image of that camera (shown in the picture below). This could be further evidence that some kind of disruptive exposure-related event has occurred.

It is not clear why setting the resolution to 1280x720 instead of 848x480 seems to prevent it from occurring.

sinuku, 2025-03-04:
@MartyG-RealSense  In your previous response, you mentioned that the white flash in the RGB image is evidence of some kind of disruptive exposure-related event. However, this phenomenon always occurs even during normal operation.

Additionally, based on tests using RealSense Viewer, the issue only appears when Auto Exposure is enabled for both the RGB and Stereo modules.
(If only one of them is enabled, there is no issue.)

However, it is still unclear why the results differ depending on the resolution.

MartyG-RealSense, 2025-03-04:
The depth and RGB exposures do not have any interaction with each other.  Changing the depth exposure affects the exposure of the depth and infrared images but not RGB. Likewise, changing the RGB exposure does not affect the exposure of the depth or infrared images.  So have both exposure types enabled should not usually cause a problem.

If the problem does not occur if only one exposure type is enabled, then the best choice may be to disable the RGB exposure and if the RGB image darkens then compensate by increasing the value of the **Brightness** RGB option.

MartyG-RealSense, 2025-03-15:
Hi @sinuku  Do you require further assistance with this case, please?  Thanks!

sinuku, 2025-03-17:
@MartyG-RealSense I am currently developing with a different resolution as a temporary workaround.

Although an immediate fix is not required, this issue should not occur at any resolution in the future.

I will add a comment if I find a solution.

Thanks for your help.

MartyG-RealSense, 2025-03-17:
Thanks very much @sinuku for the update.  I will close the issue, as you can still post a comment on the discussion in future if you find a solution.  Thanks again!

closed 2025-03-17 by MartyG-RealSense