realsenseai/librealsense #13799: Error!  The /var/lib/dkms/librealsense2-dkms/1.3.28/5.4.0-208-generic/x86_64/dkms.conf for module librealsense2-dkms includes a BUILD_EXCLUSIVE directive which does not match this kernel/arch.  This indicates that it should not be built.

monajalal, 2025-02-28:
I have a docker I need to install `librealsense2.-dkms` on it. How should I fix this?

```
(my) root@DOS:/# echo "deb https://librealsense.intel.com/Debian/apt-repo focal main" > /etc/apt/sources.list.d/librealsense.list
(my) root@DOS:/# vi /etc/apt/sources.list.d/librealsense.list
(my) root@DOS:/# apt-get update
Hit:1 http://archive.ubuntu.com/ubuntu focal InRelease
Hit:2 http://security.ubuntu.com/ubuntu focal-security InRelease                                                            
Hit:3 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu2004/x86_64  InRelease                                 
Hit:4 http://archive.ubuntu.com/ubuntu focal-updates InRelease                                                              
Hit:5 http://archive.ubuntu.com/ubuntu focal-backports InRelease
Get:6 https://librealsense.intel.com/Debian/apt-repo focal InRelease [3249 B]
Get:7 https://librealsense.intel.com/Debian/apt-repo focal/main amd64 Packages [39.7 kB]
Fetched 42.9 kB in 1s (36.9 kB/s)   
Reading package lists... Done
(my) root@DOS:/# apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
librealsense2-dkms is already the newest version (1.3.28-0ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 243 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] y
Setting up librealsense2-dkms (1.3.28-0ubuntu1) ...
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76.)
debconf: falling back to frontend: Readline
Removing old librealsense2-dkms-1.3.28 DKMS files...

------------------------------
Deleting module version: 1.3.28
completely from the DKMS tree.
------------------------------
Done.
Loading new librealsense2-dkms-1.3.28 DKMS files...
It is likely that 6.8.0-52-generic belongs to a chroot's host
Building for 5.4.0-208-generic
Building initial module for 5.4.0-208-generic
Error!  The /var/lib/dkms/librealsense2-dkms/1.3.28/5.4.0-208-generic/x86_64/dkms.conf for module librealsense2-dkms includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
Skipped.
\nLoading the modified modules into kernel...dpkg: error processing package librealsense2-dkms (--configure):
 installed librealsense2-dkms package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 librealsense2-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
(my) root@DOS:/# 

```

here is dockerfile:

```
FROM nvidia/cudagl:11.3.0-devel-ubuntu20.04

ENV TZ=US/Pacific
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone

RUN apt-get update --fix-missing && \
    apt-get install -y libgtk2.0-dev && \
    apt-get install -y wget bzip2 ca-certificates curl git vim tmux g++ gcc build-essential cmake checkinstall gfortran libjpeg8-dev libtiff5-dev pkg-config yasm libavcodec-dev libavformat-dev libswscale-dev libdc1394-22-dev libxine2-dev libv4l-dev qt5-default libgtk2.0-dev libtbb-dev libatlas-base-dev libfaac-dev libmp3lame-dev libtheora-dev libvorbis-dev libxvidcore-dev libopencore-amrnb-dev libopencore-amrwb-dev x264 v4l-utils libprotobuf-dev protobuf-compiler libgoogle-glog-dev libgflags-dev libgphoto2-dev libhdf5-dev doxygen libflann-dev libboost-all-dev proj-data libproj-dev libyaml-cpp-dev cmake-curses-gui libzmq3-dev freeglut3-dev

RUN cd / && git clone https://github.com/pybind/pybind11 &&\
    cd pybind11 && git checkout v2.10.0 &&\
    mkdir build && cd build && cmake .. -DCMAKE_BUILD_TYPE=Release -DPYBIND11_INSTALL=ON -DPYBIND11_TEST=OFF &&\
    make -j6 && make install

RUN cd / && wget https://gitlab.com/libeigen/eigen/-/archive/3.4.0/eigen-3.4.0.tar.gz &&\
    tar xvzf ./eigen-3.4.0.tar.gz &&\
    cd eigen-3.4.0 &&\
    mkdir build &&\
    cd build &&\
    cmake .. &&\
    make install

SHELL ["/bin/bash", "--login", "-c"]

RUN cd / && wget --quiet https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh -O /miniconda.sh && \
    /bin/bash /miniconda.sh -b -p /opt/conda &&\
    ln -s /opt/conda/etc/profile.d/conda.sh /etc/profile.d/conda.sh &&\
    echo ". /opt/conda/etc/profile.d/conda.sh" >> ~/.bashrc &&\
    /bin/bash -c "source ~/.bashrc" && \
    /opt/conda/bin/conda update -n base -c defaults conda -y &&\
    /opt/conda/bin/conda create -n my python=3.8

ENV PATH $PATH:/opt/conda/envs/my/bin

RUN conda init bash &&\
    echo "conda activate my" >> ~/.bashrc &&\
    conda activate my &&\
    pip install torch==2.0.0+cu118 torchvision==0.15.1+cu118 torchaudio==2.0.1 --index-url https://download.pytorch.org/whl/cu118 &&\
    pip install "git+https://github.com/facebookresearch/pytorch3d.git@stable" &&\
    pip install scipy joblib scikit-learn ruamel.yaml trimesh pyyaml opencv-python imageio open3d transformations warp-lang einops kornia pyrender

RUN cd / && git clone --recursive https://github.com/NVIDIAGameWorks/kaolin
RUN conda activate my && cd /kaolin &&\
    FORCE_CUDA=1 python setup.py develop

RUN cd / && git clone https://github.com/NVlabs/nvdiffrast &&\
    conda activate my && cd /nvdiffrast && pip install .

ENV OPENCV_IO_ENABLE_OPENEXR=1

RUN conda activate my &&\
    pip install scikit-image meshcat webdataset omegaconf pypng roma seaborn opencv-contrib-python openpyxl wandb imgaug Ninja xlsxwriter timm albumentations xatlas rtree nodejs jupyterlab objaverse g4f ultralytics==8.0.120 pycocotools videoio numba &&\
    conda install -y -c anaconda h5py

ENV SHELL=/bin/bash
RUN ln -sf /bin/bash /bin/sh
```

I am doing changes inside built docker (and will commit to it -- not planning to rebuild)

MartyG-RealSense, 2025-02-28:
Hi @monajalal  Can you confirm whether you are using an Nvidia Jetson single-board computer please, as your Dockerfile makes references to Nvidia and CUDA.

If you do have a Jetson then the `librealsense2-dkms` package should not be used as it is installed on PC computers.  Jetson has its own pair of dedicated librealsense DKMS packages, **librealsense2-utils** and **librealsense2-dev**

https://github.com/intelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

monajalal, 2025-02-28:
@MartyG-RealSense I am using my laptop not a jetson.

I am running the FoundationPose docker (Ubuntu 20.04) inside my host which runs 22.04.

MartyG-RealSense, 2025-02-28:
My knowledge of Docker installation is admittedly limited.  However, a RealSense user who experienced a similar problem with librealsense2-dkms when installing it in a Docker container on a PC with Ubuntu 22.04 posted the solution that worked for them at https://github.com/IntelRealSense/librealsense/issues/13536#issuecomment-2495419513

closed 2025-03-10 by monajalal