realsenseai/librealsense #13802: Compilation Error: "No rule to make target" during Librealsense 2.56.3 Installation on AGX Orin with Ubuntu 22.04

happydrones, 2025-02-28:
I am attempting to install Librealsense version 2.56.3 on my NVIDIA AGX Orin device running Ubuntu 22.04 Jammy Jellyfish. The system specifications are as follows:

Operating System: Ubuntu 22.04 Jammy Jellyfish
Kernel Version: 5.15.148-tegra
Python Version: 3.10.12
CUDA Version: 12.6.68
Jetpack Version: 6.2 [L4T 36.4.3]
The objective is to utilize Python to read video streams from a D435 camera.

Installation Steps:

Clone the Librealsense repository:
`git clone https://github.com/IntelRealSense/librealsense.git`
Navigate to the cloned directory and create a build directory:
```
cd librealsense
mkdir build
cd build
```
Run CMake with the specified options:
`cmake .. -DBUILD_EXAMPLES=true -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=/usr/bin/python3.10 -DBUILD_DOCUMENTATION=false`
Compile the project using make:
Error Encountered:
During the make process, the following error occurs:
```
make[5]: *** No rule to make target '/home/oscar/librealsense-2.56.md/build/libcurl/src/libcurl-build/docs/libcurl/libcurl-symbols.md', needed by 'docs/libcurl/curl_easy_cleanup.3'.  Stop.
make[4]: *** [CMakeFiles/Makefile2:231: docs/libcurl/CMakeFiles/curl-man.dir/all] Error 2
make[3]: *** [Makefile:136: all] Error 2
make[2]: *** [CMakeFiles/libcurl.dir/build.make:87: libcurl/src/libcurl-stamp/libcurl-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:895: CMakeFiles/libcurl.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```
Additional Information:
Request for Assistance:
I seek guidance on resolving this compilation error to successfully install Librealsense 2.56.3 and utilize Python for accessing the D435 camera's video stream.

happydrones, 2025-02-28:
> Hi @happydrones  Another RealSense user recently experienced this issue with 2.56.3 and libcurl at https://github.com/IntelRealSense/librealsense/issues/13767 and I suggested trying using the CMake build flag `-DCHECK_FOR_UPDATES=false` as another user had reported that it enabled the libcurl issue to be bypassed.

Oh really, I am truly grateful for your response, as I have been troubled by this issue for a whole day. I really appreciate it.

happydrones, 2025-02-28:
> Hi @happydrones  Another RealSense user recently experienced this issue with 2.56.3 and libcurl at https://github.com/IntelRealSense/librealsense/issues/13767 and I suggested trying using the CMake [...]

By the way, if I want to achieve my objective, is it correct to first install the librealsense2 SDK and then install pyrealsense2? Looking forward to your reply.

MartyG-RealSense, 2025-02-28:
If you are including the Python flags `-DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=/usr/bin/python3.10` in your CMake build instruction, as you have done, then the librealsense SDK and the pyrealsense2 wrapper will be built together at the same time and so you do not need to install the wrapper once the installation process is complete as pyrealsense2 has already been built.

happydrones, 2025-02-28:
> If you are including the Python flags `-DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=/usr/bin/python3.10` in your CMake build instruction, as you have done, then the librealsense SDK and the [...]

Thank you so much, I really appreciate it!

closed 2025-03-01 by happydrones

checolag, 2025-03-05:
sorry @MartyG-RealSense I don't think it's true, i'm trying to follow your instructions but I don't see pyrealsense2 installed automatically in my python environment. my problem is that, after installing librealsense 2.56.3, i cannot install pyrealsense2 with the same version.

MartyG-RealSense, 2025-03-05:
@checolag  If you built the librealsense SDK from a source code folder with CMake and included -DBUILD_PYTHON_BINDINGS=bool:true  in the CMake build instruction then the pyrealsense2 wrapper should have been built at the same time that librealsense installed.  After the pyrealsense2 wrapper has been successfulyl built, its files will typically be located in the directory of the librealsense source code folder shown below.

`librealsense > build > wrappers > python`

You should not need to use the **-DPYTHON_EXECUTABLE** flag unless you have more than one version of Python installed on your computer.

It is not possible to install pyrealsense2 from pip package with **pip install pyrealsense2** on the librealsense beta versions (2.56.1, 2,56.2 and 2.56.3).  Only source code installation of pyrealsense2 is an option.

checolag, 2025-03-05:
@MartyG-RealSense thank you for your answer! I thought we were talking about pyrealsense2 too. How can I install it by using source code installation? I didn't understand.
Thank you in advance.

MartyG-RealSense, 2025-03-05:
@checolag  You can build librealsense and pyrealsense2 from source together at the same time with the **libuvc backend** procedure at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

The procedure provides a pre-made build script to install librealsense called 'libuvc_installation.sh'.  

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

Edit line 46 of the script to add the **-DBUILD_PYTHON_BINDINGS=bool:true** flag to that line, and then run the script to build both librealsense and pyrealsense2 from source code.

********

If you have already installed librealsense from source and do not want to install it again you can install pyrealsense2 on its own by going to the `librealsense > build `folder and use the CMake command below.

`cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true`

checolag, 2025-03-06:
I repeat the procedure I followed so that's clearer. I have a virtual environment in which i need to process data from depth camera d435i. I have ubuntu 22.04 with kernel 6.8 and my problem is that i cannot extract accel and gyro data in python (and also in realsense viewer.
What i did was:
``` 
git clone https://github.com/IntelRealSense/librealsense.git
git checkout v2.56.3
./scripts/setup_udev_rules.sh
mkdir build && cd build
cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_PYTHON_BINDINGS=true -DPYTHON_EXECUTABLE=$(which python3)
sudo make uninstall && make clean && make -j19 && sudo make install
```

now the motion module on realsense-viewer works but still cannot manage data on python, where the pyrealsense2 version in 2.55.1. I was thinking that i had to update to the same version of realsense-viewer but i cannot update the python library.
How can I solve, please?

checolag, 2025-03-06:
I solved!! It wasn't a problem of pyrealsense2 library, but of how i was calling module API. 
Previously i was coding:
```
frame = p.wait_for_frames()
frame.as_motion_frame()
```
and i was always getting NULL values.

Instead the correct way is:
```
frame = p.wait_for_frames()
accel_frame = frame.first_or_default(rs.stream.accel)
gyro_frame = frame.first_or_default(rs.stream.gyro)
```
and then transforming in motion frame
`gyro_frame.as_motion_frame().get_motion_data()`

MartyG-RealSense, 2025-03-06:
Great to hear that you achieved a solution - thanks very much for the update and for sharing the details of your solution!