realsenseai/librealsense #13804: Deprojection pixel to point

Cody-Vu, 2025-02-28:
Hello everyone,

I got handed a object detection system with a algorithm using the D435if. Basically I receive a pixel point coordinate from the rgb color frame. I need to convert the pixel point into 3D point coordinates. In my code I execute `rs.rs2_deproject_pixel_to_point()` for that. I assume the 3D coordinate is relative to the origin of the RGB module, so the left camera, like in the picture. 
[image]

The problem is that when i measure the distance in reality there is a deviation of about 1cm. Now I try to figure out where this comes from. I tried to enclose the possible mistakes. The pixel coordinates are correct. I verified that by using the realsense viewer. In the program I hovered over the object in 2D pixel coordinates in the rgb frame. They match the values I receive from the object detection code. So the mistake must lie in the transformation of the pixel point coordinates to 3D point coordinates. In `rs.rs2_deproject_pixel_to_point()` I hand over the intrinsics of the color frame. When I read them out it printed me for the principal points: ppx = 311, ppy = 250 with a used resolution of 640x480. Doesn't that mean the actual origin is about 10 pixels off in x and y axis of the assumed origin of the rgb module (17.5+15 mm from the centerline, see picture below). Therefore do I have to assume the outputed 3D point coordinates are relative to ppx and ppy? This would explain the deviation but it would be great if someone could confirm.

[image]

Also I don't quiet understand, when I hover over the object in 3D (realsense viewer, where depth frame and color frame is active, it shows me the 3D coordinates which match with what I measure in reality. But this one I think takes the point coordinates from the depth frame (even tho both are active) where the origin is in the left imager? Correct me if I'm wrong.

For my task it must be accurate. A few millimeters offset wouldn't be dramatic but 1cm is a problem. So I try to understand where the deviation comes from so I can fix it. I appreciate any help and thank you very much in advance!

MartyG-RealSense, 2025-03-01:
Hi @Cody-Vu  Does your Python program perform depth to color alignment?  Usually when a program uses both depth and color frames and calls `rs.rs2_deproject_pixel_to_point()` to convert 2d pixels to 3D points, depth to color alignment is performed first in the script with **align_to** before using rs.rs2_deproject_pixel_to_point() secondly afterwards later in the script.  https://github.com/IntelRealSense/librealsense/issues/10875 has a Python script that demonstrates this.  https://github.com/IntelRealSense/librealsense/issues/11031#issuecomment-1352879033 - which contains three Python scripts that demonstrate different methods for finding the 3D point - may also be a helpful reference.

 Alignment is necessary when using both depth and color and trying to obtain 3D coordinates because the physical position of the RGB sensor on the front of the camera is horizontally offset from the position of the left infrared sensor (the center-line of which is the origin of depth).  When depth to color alignment is performed, the origin of depth changes from the left IR sensor to the center-line of the RGB sensor.

In regard to seemingly incorrect coordinates in the RealSense Viewer in 3D mode, as https://github.com/IntelRealSense/librealsense/issues/11031#issuecomment-1352879033 mentions, the center of projection (ppx and ppy) is not necessarily the center of the image.

Cody-Vu, 2025-03-01:
Hello @MartyG-RealSense, thank you for taking your time. Yes I do run depth to color alignment with the following commands:

`align = rs.align(rs.stream.color)`
`frames = pipeline.wait_for_frames()`
`aligned frames = align.process(frames)`
`depth_frame = aligned_frames.get_depth_frame()`
`color_frame = aligned_frames.get_color_frame()`

After that I run `rs.rs2_deproject_pixel_to_point()` with the intrinsics of the color frame (rather both frames will have the same intrinsics now after the alignment happened, if I understand it correctly).

I think I explained it incomprehensible. In the Realsense viewer in 3D mode, I measure correct values for the 3D coordinates. They match with the ones I measure in reality and I measure it relative to the origin of the left IR. That's why I don't understand that I get different values from my code even tho both the 2D pixel coordinates from the Realsense viewer and my code are the same (for the same object for example). But when I switch to 3D mode in realsense viewer it shows me correct values relative to the left IR. My code in comparison output coordinates relative to the RGB sensor with a slightly offset (about 1cm) after the deprojection from pixel to point. Is there a way I get the same values as the realsense viewer, because they look pretty accurate. Or do you know how the calculation in realsense viewer work internally? If not I have to keep working with the deprojection of pixel to point.

I don't quit understand what is the exact origin (0,0) of the 3D coordinates I receive from my code. For my task the camera will be mounted on a robotic arm. That's why I need to know where the exact origin is. From what I assume now the origin lies in the RGB sensor, but do I have to take ppx and ppy into account? And is that the reason of the offset that I measure? With a difference of 10 pixels, x and y coordinates will have about a 0,8 cm flaw at a depth of 50 cm, from what I caculated. I mean that would explain the offset but it would be great if you could confirm.

MartyG-RealSense, 2025-03-01:
Thanks very much for the additional information.

Yes, by using color intrinsics after depth-to-color alignment, depth and color will have shared intrinsics.

The 3D pointcloud mode of the RealSense Viewer uses a different method from `align_to` to map depth and RGB together into a 'textured pointcloud'.  That method is based on performing the instructions `map_to` and then `pc.calculate`, similar to how the RealSense SDK's **rs-pointcloud** C++ example program at the link below works.  

https://github.com/IntelRealSense/librealsense/blob/master/examples/pointcloud/rs-pointcloud.cpp

So alignment with align_to and alignment in the RealSense Viewer's 3D mode is not an exact like-for-like comparison, though the pointcloud that is output is similar for both methods.

The map_to / pc.calculate approach can be slightly more accurate than using align_to and then rs2_deproject_pixel_to_point because it avoids small distortions in the image that might occur as a consequence of alignment with align_to before performing deprojection.

The chances of an incorrect measurement being output when obtaining 3D coordinates with Python using deprojection are increased if the instruction **get_distance()** is used in the script, as coordinates may be accurate at the center of the image and increasingly drift when checking coordinates towards the outer edges of the image.  Using color intrinsics helps to avoid this, but inaccuracies in coordinates at the horizontal and vertical sides of the image may not be completely avoided.

If you have performed depth-color alignment with align_to then the 0,0 origin of the 3D coordinates will be the center-line of the RGB sensor.  When using align_to, the RealSense SDK takes care of the calculations involved in the alignment automatically using a software component called the **align processing block**, so you do not need to calculate the alignment manually yourself.

Cody-Vu, 2025-03-01:
@MartyG-RealSense, Thank you very much for taking time and giving a detailed explanation. I think I have a good overview and understanding of how this works now. With the depth-color alignment I have to expect some inaccuracies. From the output. the 3D coordinates are relative to the RGB sensor and  I do not have to take ppx and ppy into account, because that is already considered by the interal calculation of the deprojection. 

So my only option to improve it, would be the map_to/pc.caculate approach, but the expected output would be similar, maybe with slightly more accurate values. Thanks again, I hope I understood it correct!

MartyG-RealSense, 2025-03-01:
You are correct.  map_to / pc_calculate can be slightly more accurate.  If you already have a working program that you are satisfied with then it would likely not be worthwhile writing an entirely new script to achieve a minor improvement.

closed 2025-03-01 by Cody-Vu