realsenseai/librealsense #13805: Depth streams were distorted when connected the sync cable

Ayanami0-39, 2025-03-01:
I connected two RealSense D415 cameras as described in the whitepaper. However, the depth frames started to be distorted once I connected them. But they did succeed in syncing together according to the logging info.

Correct depth frames:
[image]

Erroneous depth frames:
[image]

Logging info after synchronization:
[image]

MartyG-RealSense, 2025-03-01:
Hi @Ayanami0-39  It is not clear why the level of detail on the depth image would drop when the cameras are connected together in a sync arrangement.  If these images are displayed using a program that you have written yourself though then you could attempt to compensate for it though by increasing the **Laser Power** for each camera to its maximum value of '360'.

When checking the timestamps of the cameras after sync has begun, you should also check whether the values are drifting apart over time.  If they stay identical constantly then it actually means that sync is **not** working.  This is discussed at the section of the multiple camera white-paper guide linked to below, in the paragraph that begins with **Now to the somewhat counter intuitive aspect of time stamps**.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#3-multi-camera-programming

You could also check the wiring of your sync cables using the guide at the links below.

https://vcl3d.github.io/VolumetricCapture/docs/hardware/rs2_hardware/

https://support.intelrealsense.com/hc/en-us/community/posts/360050894574/comments/360013337454

Ayanami0-39, 2025-03-01:
Thank you for your swift reply👍. I'll check the white-paper again and see whether the timestamps are drifting apart over time.😀

Ayanami0-39, 2025-03-01:
Good news! After increasing the Laser Power to '360', the distortion problem seems to disappear (although I still observed slight distortion when setting the Laser Power from 150 to '240'). Additionally, I did observe timestamp drifting, as shown below.

[image]

MartyG-RealSense, 2025-03-01:
That's great to hear that you were successful with the depth detail and the timestamping, @Ayanami0-39 - thanks very much for the update!

closed 2025-03-01 by Ayanami0-39