realsenseai/librealsense #13806: no IMU data in my docker container

Frank-Schiro, 2025-03-01:
Hello, I can't get IMU data on my realsense d435i camera. realsense-viewer tells me no frames received for IMU in 2D mode. 

The software also tells me I have version 2.55.1.0 and it recommends 2.55.1.6486, but I can't find the newer version even after building from scratch using the repo I still get 2.55.1.0.

Here is an example dockerfile and docker-compose which boot up everything but no IMU data:

Dockerfile:
```yml
FROM nvidia/cuda:12.3.1-devel-ubuntu22.04

ENV DEBIAN_FRONTEND=noninteractive

# =============== #
# D435i
# =============== #

# Install basic dependencies
RUN apt-get update && \
    apt-get install -y \
    software-properties-common \
    wget \
    curl \
    lsb-release \
    gpg \
    udev \
    && rm -rf /var/lib/apt/lists/*

# Step 1: Add the key first
RUN mkdir -p /etc/apt/keyrings && \
    curl -sSf https://librealsense.intel.com/Debian/librealsense.pgp | tee /etc/apt/keyrings/librealsense.pgp > /dev/null

# Step 2: Then configure the repository
RUN echo "deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo jammy main" | \
    tee /etc/apt/sources.list.d/librealsense.list && \
    apt-get update && \
    apt-get install -y librealsense2-utils librealsense2

# Step 3 - Install developer and debug packages
RUN apt-get install -y \
    librealsense2-dev \
    librealsense2-dbg \
    && rm -rf /var/lib/apt/lists/*

# =============== #
# ROS
# =============== #

RUN apt-get update && \
    apt-get install -y \
    python3-pip \
    python3-dev \
    && rm -rf /var/lib/apt/lists/*

# Add ROS2 apt repository
RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg && \
    echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

# Install ROS2 Humble
RUN apt-get update && \
    apt-get install -y \
    ros-humble-desktop \
    ros-dev-tools \
    && rm -rf /var/lib/apt/lists/*

RUN apt-get update && apt-get install -y ros-humble-diagnostic-updater

# Install RealSense ROS2 wrapper directly from apt repositories
RUN apt-get update && \
    apt-get install -y \
    ros-humble-realsense2-* \
    && rm -rf /var/lib/apt/lists/*

RUN echo "source /opt/ros/humble/setup.bash" >> ~/.bashrc

# Set up display
ENV DISPLAY=:0

RUN mkdir /workspace

# https://github.com/IntelRealSense/realsense-ros
# Launch full realsense suite
CMD ["bash", "-c", "\
    source /opt/ros/humble/setup.bash && \
    ros2 launch realsense2_camera rs_launch.py \
    pointcloud.enable:=true \
    pointcloud.ordered_pc:=true \
    pointcloud.allow_no_texture_points:=true \
    align_depth.enable:=true \
    enable_sync:=true \
    enable_color:=true \
    enable_depth:=true \
    # enable_infra1:=true \
    # enable_infra2:=true \
    enable_gyro:=true \
    enable_accel:=true \
    enable_rgbd:=true \
    unite_imu_method:=1 \
    spatial_filter.enable:=true \
    temporal_filter.enable:=true \
    # 2 means interpolate (0=none, 1=near pixel, 2=interpolate)
    hole_filling_filter.holes_fill:=2 \
    hole_filling_filter.enable:=true \
    depth_module.depth_profile:=848x480x30 \
    rgb_camera.color_profile:=848x480x30 \
    # visual odometry 
    publish_odom_tf:=true \
    odom_frame_id:='odom' \
    "]

```

docker-compose.yml
```yml
realsense:
    build:
      context: ./realsense_d435i_node
    privileged: true
    ipc: host # for dds shared memory transport
    environment:
      - NVIDIA_VISIBLE_DEVICES=all
      - NVIDIA_DRIVER_CAPABILITIES=all
      - ROS_DOMAIN_ID=42
    volumes:
      - /tmp/.X11-unix:/tmp/.X11-unix
      - /dev:/dev
      - /dev/shm:/dev/shm 
    devices:
      - /dev/video0:/dev/video0
      - /dev/video1:/dev/video1
      - /dev/video2:/dev/video2
      - /dev/bus/usb:/dev/bus/usb
    network_mode: host
    runtime: nvidia
    deploy:
      resources:
        reservations:
          devices:
            - driver: nvidia
              count: 1
              capabilities: [ gpu ]
```

Frank-Schiro, 2025-03-02:
I switched to foxy and FROM ubuntu:20.04 and the IMU works SOMETIMES now. Basically it seems like it works if I unplug and replug the camera on startup but not if I turn the docker container on and off without power cycling the camera usb. 
Does the IMU just not work with 22.04 and humble?

MartyG-RealSense, 2025-03-02:
Hi @Frank-Schiro  Yes, the IMU can work with 22.04 and Humble.

Can you confirm whether you are using an Nvidia Jetson single-board computer, or a PC desktop or laptop equipped with an Nvidia graphics GPU?

It looks as though your Dockerfile is based on the **distribution_linux.md** installation instructions, which are meant for use on PC desktop / laptop computers.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

Nvidia Jetson boards have a different set of package installation instructions.

https://github.com/intelRealSense/librealsense/blob/master/doc/installation_jetson.md#4-install-with-debian-packages

Frank-Schiro, 2025-03-03:
I'm on an ubuntu 24.04 LTS with Nvidia 4090. 

When using Humble, the IMU would not broadcast. When using When using Foxy, The IMU broadcasts, but for some reason the foxy change caused the color resolution to switch to 720 x 1280 even though its set explicitly to 848x480x30. 

```bash
ros2 echo /camera/color/info
---
header:
  stamp:
    sec: 1741003685
    nanosec: 650285568
  frame_id: camera_color_optical_frame
height: 720
width: 1280
...

ros2 echo /camera/depth/info
---
header:
  stamp:
    sec: 1741003758
    nanosec: 34195200
  frame_id: camera_depth_optical_frame
height: 480
width: 848
```

Above, you can see width and height are different, even though I set them explicitly and set align_depth.enable to true. 

From my docker compose:
```yml
CMD ["bash", "-c", "\
    source /opt/ros/foxy/setup.bash && \
    ros2 launch realsense2_camera rs_launch.py \
    depth_module.depth_profile:=848x480x30 \
    rgb_camera.color_profile:=848x480x30 \
    align_depth.enable:=true \
    pointcloud.enable:=true \
    pointcloud.ordered_pc:=true \
    pointcloud.allow_no_texture_points:=true \
    enable_sync:=true \
    enable_color:=true \
    enable_depth:=true \
    enable_gyro:=true \
    enable_accel:=true \
    enable_rgbd:=true \
    unite_imu_method:=1 \
    spatial_filter.enable:=true \
    temporal_filter.enable:=true \
    hole_filling_filter.holes_fill:=2 \
    hole_filling_filter.enable:=true \
    publish_odom_tf:=true \
    odom_frame_id:='odom' \
    "]
```

To switch to foxy, I switched first line of dockerfile to this: `FROM ubuntu:20.04`

Heres the updated dockerfile:

```yml
# Use Ubuntu 20.04 as the base image (includes a compatible kernel by default)
FROM ubuntu:20.04

# Set environment variables to avoid interactive prompts during package installation
ENV DEBIAN_FRONTEND=noninteractive

# Install necessary packages
RUN apt-get update && \
    apt-get install -y \
    curl \
    lsb-release \
    apt-transport-https \
    software-properties-common \
    gnupg2 \
    && rm -rf /var/lib/apt/lists/*

# Add the Librealsense repository
RUN mkdir -p /etc/apt/keyrings && \
    curl -sSf https://librealsense.intel.com/Debian/librealsense.pgp | tee /etc/apt/keyrings/librealsense.pgp > /dev/null && \
    echo "deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo `lsb_release -cs` main" | \
    tee /etc/apt/sources.list.d/librealsense.list && \
    apt-get update

# Install Librealsense packages
RUN apt-get update && \
    apt-get install -y \
    librealsense2-dkms \
    librealsense2-utils \
    librealsense2-dev \
    librealsense2-dbg \
    && rm -rf /var/lib/apt/lists/*

# Upgrade all installed packages, including Librealsense
RUN apt-get update && \
    apt-get upgrade -y && \
    rm -rf /var/lib/apt/lists/*

# =============== #
# ROS
# =============== #

RUN apt-get update && \
    apt-get install -y \
    python3-pip \
    python3-dev \
    && rm -rf /var/lib/apt/lists/*

# Add ROS2 apt repository
RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg && \
    echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

# Install ROS2 Humble
RUN apt-get update && \
    apt-get install -y \
    ros-foxy-desktop \
    ros-dev-tools \
    && rm -rf /var/lib/apt/lists/*

# Install RealSense ROS2 wrapper directly from apt repositories
RUN apt-get update && \
    apt-get install -y \
    ros-foxy-realsense2-* \
    && rm -rf /var/lib/apt/lists/*

RUN echo "source /opt/ros/foxy/setup.bash" >> ~/.bashrc

# Set up display
ENV DISPLAY=:0

RUN mkdir /workspace

# https://github.com/IntelRealSense/realsense-ros
# Launch full realsense suite
CMD ["bash", "-c", "\
    source /opt/ros/foxy/setup.bash && \
    ros2 launch realsense2_camera rs_launch.py \
    initial_reset:=true \
    pointcloud.enable:=true \
    pointcloud.ordered_pc:=true \
    pointcloud.allow_no_texture_points:=true \
    align_depth.enable:=true \
    enable_sync:=true \
    enable_color:=true \
    enable_depth:=true \
    # disabled to minimize resource consumption
    # enable_infra1:=true \
    # enable_infra2:=true \
    enable_gyro:=true \
    enable_accel:=true \
    enable_rgbd:=true \
    unite_imu_method:=2 \
    spatial_filter.enable:=true \
    temporal_filter.enable:=true \
    # 2 means interpolate (0=none, 1=near pixel, 2=interpolate)
    hole_filling_filter.holes_fill:=2 \
    hole_filling_filter.enable:=true \
    depth_module.depth_profile:=848x480x30 \
    rgb_camera.color_profile:=848x480x30 \
    # visual odometry 
    publish_odom_tf:=true \
    odom_frame_id:='odom' \
    "]
```

MartyG-RealSense, 2025-03-03:
Thanks very much for the confirmation that you are using a PC instead of Jetson.

If you have installed the **librealsense2-utils** package then you should have the librealsense example IMU test program **rs-motion** installed on your computer.  If you run rs-motion, does the on-screen graphic react to movement of the camera? 

[image]

In realsense-viewer, is IMU data able to be displayed if you only enable the Motion Module on its own?

Frank-Schiro, 2025-03-04:
In the rs-motion program I see motion when moving the device.  
In realsense-viewer, I do not see IMU data even when using IMU only. See attached image. There is also a message about updating in the image top right corner, but I have not been able to figure that out. 

[image]

I ran the tests in this docker container:

docker-compose.yml
```yml
# To run
# 1. docker-compose up -d realsense2
# 2. docker-compose exec realsense2 /bin/bash

  realsense2:
    build:
      context: ./realsense2
    privileged: true
    ipc: host # for dds shared memory transport
    environment:
      - DISPLAY=${DISPLAY}
      - NVIDIA_VISIBLE_DEVICES=all
      - NVIDIA_DRIVER_CAPABILITIES=all
      - ROS_DOMAIN_ID=${ROS_DOMAIN_ID}
    volumes:
      - /tmp/.X11-unix:/tmp/.X11-unix
      - /dev:/dev
      - /dev/shm:/dev/shm # Trying to fix this: [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7413: open_and_lock_file failed -> Function open_port_internal # for dds shared memory transport
      - /home/frank/dev/projects/AI/vision/d435i_docker_ros:/workspace
    devices:
      - /dev/video0:/dev/video0
      - /dev/video1:/dev/video1
      - /dev/video2:/dev/video2
      - /dev/bus/usb:/dev/bus/usb
    network_mode: host
    runtime: nvidia
    deploy:
      resources:
        reservations:
          devices:
            - driver: nvidia
              count: 1
              capabilities: [ gpu ]
    entrypoint: /bin/bash
    stdin_open: true
    tty: true
```

Dockerfile:

```yml
FROM nvidia/cuda:12.3.1-devel-ubuntu22.04

ENV DEBIAN_FRONTEND=noninteractive

# =============== #
# D435i
# =============== #

# Install basic dependencies
RUN apt-get update && \
    apt-get install -y \
    software-properties-common \
    wget \
    curl \
    lsb-release \
    gpg \
    udev \
    && rm -rf /var/lib/apt/lists/*

# Step 1: Add the key first
RUN mkdir -p /etc/apt/keyrings && \
    curl -sSf https://librealsense.intel.com/Debian/librealsense.pgp | tee /etc/apt/keyrings/librealsense.pgp > /dev/null

# Step 2: Then configure the repository
RUN echo "deb [signed-by=/etc/apt/keyrings/librealsense.pgp] https://librealsense.intel.com/Debian/apt-repo jammy main" | \
    tee /etc/apt/sources.list.d/librealsense.list && \
    apt-get update && \
    apt-get install -y librealsense2-utils librealsense2

# Step 3 - Install developer and debug packages
RUN apt-get install -y \
    librealsense2-dev \
    librealsense2-dbg \
    && rm -rf /var/lib/apt/lists/*

ENTRYPOINT ["realsense-viewer"]
```

closed 2025-03-22 by MartyG-RealSense