realsenseai/librealsense #13812: Unable to access Intel RealSense D435i without sudo despite udev rules

happydrones, 2025-03-03:
## **Description**
### env
Operating System: Ubuntu 22.04 Jammy Jellyfish
Kernel Version: 5.15.148-tegra
Python Version: 3.10.12
CUDA Version: 12.6.68
Jetpack Version: 6.2 [L4T 36.4.3]
librealsense sdk:2.56.3

First, I will use the following installation steps：
Clone the Librealsense repository:
`git clone https://github.com/IntelRealSense/librealsense.git`
Navigate to the cloned directory and create a build directory:

``` 
cd librealsense
mkdir build
cd build
cmake .. -DBUILD_EXAMPLES=true \
-DBUILD_PYTHON_BINDINGS=bool:true\
-DPYTHON_EXECUTABLE=/usr/bin/python3.10 \
-DBUILD_WITH_CUDA=true \
-DCUDA_ARCHITECTURES=87 \
-DCHECK_FOR_UPDATES=false
sudo make 
sudo make install
```
then I can get the video steam run realsense-viewer without sudo privilege 

### Problem

 But I am unable to access my Intel RealSense D435i device from Python scripts without using sudo,  The device is only recognized when I run the script with elevated privileges (e.g., sudo python script.py). Without sudo, the device is not detected.

### Details

When I run ` lsusb`, the device is correctly detected by the system:
` Bus 002 Device 003: ID 8086:0b3a Intel Corp. Intel(R) RealSense(TM) Depth Camera 435i`
My user (oscar) is part of the plugdev group, as confirmed by:
```
groups
oscar adm cdrom sudo audio dip video plugdev render i2c lpadmin gdm sambashare ollama weston-launch gpio jtop 
```
The udev rules file /etc/udev/rules.d/99-realsense-libusb.rules contains the following rule for my device:
` SUBSYSTEMS=="usb", ATTRS{idVendor}=="8086", ATTRS{idProduct}=="0b3a", MODE:="0666", GROUP:="plugdev" `
### Steps Taken
I’ve tried the following troubleshooting steps without success:
```
#Reloaded the udev rules:
sudo udevadm control --reload-rules && sudo udevadm trigger
```
Unplugged and replugged the device to ensure the rules take effect.
Checked the device file permission 
Tested with a simple Python script using the pyrealsense2 library:
```python
import pyrealsense2 as rs
ctx = rs.context()
devices = ctx.query_devices()
print(f"Number of devices: {len(devices)}")
With sudo: Outputs Number of devices: 1 (device detected).
Without sudo: Outputs Number of devices: 0 (device not detected).
```
Request for Help
Despite following the standard steps to configure udev rules for RealSense devices, I still cannot access the D435i without sudo. Could someone please help me identify what might be missing or incorrectly configured? Are there additional steps or checks I should perform to resolve this issue? Any suggestions would be greatly appreciated!

happydrones, 2025-03-04:
Thank you for your reply. I have tried your guidance, but nothing has changed.

MartyG-RealSense, 2025-03-04:
Another user at https://github.com/IntelRealSense/librealsense/issues/10352 had the same problem with sudo and managed to resolve it.  They said it had been an 'admin problem'.  You could leave a comment at that case to see if they can share the details of what worked for them.

happydrones, 2025-03-08:
Hello, I left a comment under another issue(#10352), but no one has responded. Please assist me with this, and the issue that person encountered is somewhat different from mine as well.

MartyG-RealSense, 2025-03-08:
Another Jetson user at https://github.com/IntelRealSense/librealsense/issues/13821 has created a new issue regarding their camera only working correctly with sudo.  It may be worth following that discussion too.

happydrones, 2025-03-11:
>Another Jetson user at https://github.com/IntelRealSense/librealsense/issues/13821 has created a new issue regarding their camera >only working correctly with sudo. It may be worth following that discussion too.

Hello developer, I also followed that issue, but it still doesn't solve my problem because my realsense-viewer works fine, but only my own Python program can't execute. Please help me.

MartyG-RealSense, 2025-03-11:
I do not have any further solutions to suggest, unfortunately.

As your Python programs works when launching it with sudo, is it possible for you to keep using sudo?

If your script needs to work without sudo, is there any change if you repeat the udev load procedure and additionally input an **export LD_LIBRARY_PATH** command?

```
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && sudo udevadm trigger

export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH
```

MartyG-RealSense, 2025-03-22:
Hi @happydrones  Have you been able to make progress with this issue in the last couple of weeks, please?

happydrones, 2025-03-24:
> Hi [@happydrones](https://github.com/happydrones) Have you been able to make progress with this issue in the last couple of weeks, please?

yes  I have solved this problem by remaking after changing the path of python interpreter

but I still have  a question

Description:

I have encountered an issue while using a conda virtual environment on Linux. My device only supports installing the RealSense library from source. Previously, I built the entire package with Python bindings by running the following commands:

```
cd librealsense
mkdir build
cd build
cmake .. -DBUILD_EXAMPLES=true \
         -DBUILD_PYTHON_BINDINGS=bool:true \
         -DPYTHON_EXECUTABLE=/usr/bin/python3.10 \
         -DBUILD_WITH_CUDA=true \
         -DCUDA_ARCHITECTURES=87 \
         -DCHECK_FOR_UPDATES=false
sudo make 
sudo make install
```

This process installs not only the Python wrapper (pyrealsense2) but also other components that I don't need in my conda environment.

My Goal:

I would like to build and install only the Python wrapper (pyrealsense2) in my conda virtual environment without affecting or installing the other components of the library. One idea I considered is modifying the PYTHON_EXECUTABLE path in the CMake configuration to point to the Python interpreter inside the virtual environment. However, this still results in the installation of the full suite of components.

Questions:

Is there a way to compile and install only the Python bindings (wrapper) for pyrealsense2 specifically for a conda environment?

Can the build process be modified to isolate the Python wrapper installation, avoiding the installation of unnecessary components?

Any guidance or suggestions on how to achieve this would be greatly appreciated.

MartyG-RealSense, 2025-03-24:
https://github.com/IntelRealSense/librealsense/issues/4585#issuecomment-889002599 provides advice for installing only pyrealsense on Conda via the **pip install pyrealsense2** method of installing from a .whl package instead of compiling pyrealsense2 from source code.

The pip install pyrealsense2 method supports Python 3.10 when installing the wrapper on Arm-based devices such as Jetson.

closed 2025-04-10 by MartyG-RealSense