realsenseai/librealsense #13813: RealSense D450 and D4 Vision Processor NDAA compliance

GirishiF, 2025-03-04:
Required Info	
Camera Model	{D450} 
Firmware Version	(Open RealSense Viewer --> Click info)
Operating System & Version	{Win (8.1/10) / Linux (Ubuntu 14/16/17)
Kernel Version (Linux Only)	(e.g. 4.14.13)
Platform	PC/Raspberry Pi/ NVIDIA Jetson / Qualcomm/etc..
SDK Version	{legacy / 2.. }
Language	{C/C#/labview/nodejs/opencv/pcl/python/unity }
Segment	{Robot/Smartphone/VR/AR/others }

Issue Description
<Describe your issue / question / feature request / etc..>
Hello @MartyG-RealSense  We would like to know whether the Devices MPN: 82635DSD450 (REALSENSE DEPTH MODULE D450) & MPN: 82635DSASICBDIF (BOARD VISION PRO D4 IMU FOR D450) are NDAA compliant. 
D455 Depth Camera Requirement.pdf

MartyG-RealSense, 2025-03-04:
Hi @GirishiF  The link below has an archive of the regulatory certificates for RealSense products. 

https://www.intelrealsense.com/regulatory-information/

Formal NDAA compliance is not included in the testing regime but I am not aware of any aspect of the cameras that would not be in compliance with NDAA as described at the link below.

https://getsafeandsound.com/ndaa-compliant/

MartyG-RealSense, 2025-03-15:
Hi @GirishiF  Do you require further assistance with this case, please?  Thanks!

GirishiF, 2025-03-15:
Hi @MartyG-RealSense, Thanks for your prompt response and follow up! The question in the doubt for NDAA compliance arises as D4 vision processors 82635DSASICBDIF (BOARD VISION PRO D4 IMU FOR D450) mentions it on its silk screen to be "Made in China" Can you help on this? 

[image]

MartyG-RealSense, 2025-03-15:
The image of the D4 V3 board in the official RealSense store also has Made in China written on it.

[image]

My understanding from the NDAA information that I linked to earlier though is that it relates to products containing components from a very small number of specifically listed manufacturers rather than a blanket ban.

Also, the 'Made In' label often means that the stated country is the one where final assembly of a product takes place, not where its individual components originated from.  For example, the L515, D415 and D415 camera models have had 'Made in Thailand' on the sticker on their outer casing.

GirishiF, 2025-03-17:
Can you suggest us any sources for the vision board which can have a silk screen print as "Made in Thailand" or "Made in Vietnam" instead of made in China? I believe it has to do with HS code/HTSUS.

MartyG-RealSense, 2025-03-17:
It can be difficult to research components based on images because retailers who stock them tend to use sample images to represent the item rather than an image that exactly matches the item in stock.

However, the Intel approved parts supplier Rutronik24 does list the country of origin as Thailand on all of their RealSense D4 board listings. 

**D4 V1**
This board is used with D415 and D430 depth modules and does not have an IMU.

https://www.rutronik24.co.uk/product/intel/952019/10164045.html

**D4 V3**
This board is used with the D430 and D450 depth modules and has an IMU

https://www.rutronik24.co.uk/product/intel/99a2nx/14278151.html

**D4 V4**
This board is used only with the D401 depth module and does not have an IMU.

https://www.rutronik24.co.uk/product/intel/99agar/17740098.html

**D4 V5**
This board is used with the D450 depth module to make a D457 camera with fast GMSL / Fakra connector.

https://www.rutronik24.co.uk/product/intel/99c2m4/20357915.html

MartyG-RealSense, 2025-03-17:
If you are using the D450 depth module then I would recommend pairing it with the D4 V5 board if possible to do so.  This is because there are two versions of D450 (the '00' and '01' model), and if D4 V3 and the D450 module are purchased separately then there is the risk that you could be supplied with the older '00' model.  This can cause an 'Invalid SKU Error' when joined to the D450 that can only be fixed by sending the D4 board to Intel for a firmware update.

The D4 V5 board does not have this issue.  However, because it is set to use a GMSL / Fakra cable connection by default, it would mean that if you plan to use the board with a USB cable then someone would manually have to move a physical switch on the board to the 'USB' position, which may not be suitable if you will be mass-producing a product using the RealSense boards.

GirishiF, 2025-04-11:
Hi @MartyG-RealSense. Thanks for your consistent support. We can proceed with this issue as resolved and close it. Once again thank you!

MartyG-RealSense, 2025-04-11:
You are very welcome, @GirishiF - I'm pleased to be able to help.  As you suggest, I will close this case.  Thanks again!

closed 2025-04-11 by MartyG-RealSense