realsenseai/librealsense #13816: D435i on rk3588

agakisama1234, 2025-03-06:
| | |
|---|---|
|**librealsense**|2.55.1 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435I|
|**Serial Number**|233622072850|
|**Firmware Version**|5.16.0.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|229123060702|
|**Firmware Update Id**|229123060702|
|**Dfu Device Path**||

I want to build D435i ROS2 environment  on RK3588. Kernel version is Ubuntu 5.10.160.
I've done so far: 
1.Pull source code from github for compilation according to Linux Guide(can't apply kernel patch), and i finally find kernel conflict,camera can't work.
2.i try to find some other way: libuvc_installation.sh to bypass kernel conflict which learned previous github issue.
And problem is that whenever i run realsense-viewer(can open camera and see rgb image)  ,my system space is constantly growing until it crashes.
Meanwhile under ros2, run camera node find power state failure.

MartyG-RealSense, 2025-03-06:
Hi @agakisama1234  Using the libuvc_installation.sh installation method to bypass the kernel is the correct method to use on a Rockchip (RK) device.

When you launch realsense-viewer with the camera plugged in, is the camera automatically listed in the options side-panel of realsense-viewer or do you have to use the Add Source button at the top of the side-panel to access the RGB?

agakisama1234, 2025-03-06:
I flashed the system separately . One system, one try.

MartyG-RealSense, 2025-03-06:
Does realsense-viewer have the same system space issue if you launch it in sudo admin powers mode with **sudo realsense-viewer**

agakisama1234, 2025-03-06:
i will try tomorrow

MartyG-RealSense, 2025-03-06:
Thank you.  I look forward to your next report.

agakisama1234, 2025-03-07:
Hi, Marty. I reflash a system and build enviroment according to 
[](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md)
and then connect RealSense device(D435i) with USB3.0, run rs-enumerate-devices.
so far, it seems like ok.
but when I run **sudo realsense-viewer** ,  the interfacing click is very stuck,not smooth like last time.Meanwhile, my storage space 
keeps going up until it fills up to 64g. When I close the realsense-viewer, growing stop.
In other words,whenever I open and keep running realsense-viewer,it caues the system to freeze and the sapce  to explode.

I crossing my fingers that hope you will  help me out.

[image]
[image]

agakisama1234, 2025-03-07:
[image]

agakisama1234, 2025-03-07:
Is cmake parament need change?

MartyG-RealSense, 2025-03-07:
If you are using realsense-viewer then the parameters below should be added to the CMake build instruction on line 46 of the **libuvc_installation.sh** script.

`-DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

Does it make a difference if you delete the configuration file of realsense-viewer from your Ubuntu computer using the command below?  Doing so will make realsense-viewer use its default settings when it is launched.

`rm ~/.realsense-config.json`

Does the system space still grow if you enable the depth stream only instead of the RGB stream?

agakisama1234, 2025-03-10:
[image]
I follow Installation on Ubuntu step 2 and step 3 on ros2(foxy) and run ros2 node.Device can't find anymore.

agakisama1234, 2025-03-10:
[image]

agakisama1234, 2025-03-10:
[image]
I follow installation on ubuntu step 2 step3 and run camera node:

MartyG-RealSense, 2025-03-10:
I have not come across a similar problem to yours in a past case so there is not a past precedent to refer to.  However, I would always recommend enabling the depth stream first and RGB stream second when using the realsense-viewer tool, if only because the 3D pointcloud mode of realsense-viewer seems to work best when the streams are enabled in that order.

The RealSense ROS wrapper has differences from librealsense tools such as realsense-viewer in how it operates (as it runs on top of librealsense), so there is a good chance that the system space growth problem will not occur in the wrapper.

agakisama1234, 2025-03-10:
Thank you for your reply.One thing I want to confirm is whether the latest SDK package and ROS2 wrapper on GitHub can meet my current needs?

agakisama1234, 2025-03-10:
I will try again on ros2_ws(foxy) directly.

MartyG-RealSense, 2025-03-10:
The RealSense ROS2 wrapper no longer supports Foxy unfortunately as Foxy support was removed in version 4.54.1.  The wrapper currently supports Humble and Iron, and on Ubuntu 24.04 supports Rolling and Jazzy.

agakisama1234, 2025-03-11:
[image]

[image]
If the image kernel of this board has been adapted to the uvc universal driver D4M, will there be a conflict?

agakisama1234, 2025-03-11:
> The RealSense ROS2 wrapper no longer supports Foxy unfortunately as Foxy support was removed in version 4.54.1. The wrapper currently supports Humble and Iron, and on Ubuntu 24.04 supports Rolling and Jazzy.

Can you recommend a relatively stable SDK package that can run stably under ROS2 Ubuntu20.04？

agakisama1234, 2025-03-13:
Thanks so much for the technical support, I've seen the topic of the camera under rviz2 and I'll close this issue.

closed 2025-03-13 by agakisama1234

MartyG-RealSense, 2025-03-13:
You are very welcome.  I'm pleased that I could help.  Thanks very much for the update!